margelo/react-native-vision-camera #2243: 🐛 Application closes by itself after scanning a barcode.

Rubensem93, 2023-12-01:
### What's happening?

Hello, greetings to the great team you are, I have a problem using the library. It turns out that I implement it as indicated but I have problems after scanning a barcode the application simply quits completely. It can work fine for a few minutes scanning calmly and suddenly as if nothing comes out. Sorry if I'm not a programming veteran, I'm learning to use the library :D greetings and many thanks to the answers.

### Reproduceable Code

```tsx
import React, {useCallback, useState} from 'react';
import {useSelector} from 'react-redux';
import {RootState} from '../assets/store';
import FontAwesome from 'react-native-vector-icons/FontAwesome';
import {TextInput, Button, Modal} from 'react-native-paper';
import {Picker} from '@react-native-picker/picker';
import {View, TouchableOpacity, StyleSheet, Text} from 'react-native';
import {
  Camera,
  useCodeScanner,
  useCameraDevice,
} from 'react-native-vision-camera';

const EntregasOrdenes = () => {
  const [selectedOption, setSelectedOption] = useState<string | null>(null);
  const user = useSelector((state: RootState) => state.auth.user);
  const isMasivaEnabled = user?.AGENTE_ASINGADO === 'BODEGA SANTIAGO';
  const [OD, setOD] = useState('');
  const [odsList, setOdsList] = useState<string[]>([]);
  const device = useCameraDevice('back');
  const [showCameraModal, setShowCameraModal] = useState(false);
  const [lockScanner, setLockScanner] = useState(false);

  const updateScanningState = (scannedCode: string) => {
    setOD(scannedCode);
    setOdsList(prevList => [...prevList, scannedCode]);
  };

  const codeScanner = useCodeScanner({
    codeTypes: [
      'qr',
      'ean-13',
      'code-128',
      'code-39',
      'code-93',
      'codabar',
      'ean-8',
    ],
    onCodeScanned: codes => {
      if (!lockScanner && codes.length > 0 && codes[0].value) {
        setLockScanner(true);
        try {
          const scannedCode = codes[0].value;
          updateScanningState(scannedCode);
          console.log(scannedCode);
        } catch (error) {
          console.error('Error al procesar el código escaneado:', error);
        } finally {
          setShowCameraModal(false);
          setTimeout(() => {
            setLockScanner(false);
          }, 3000);
        }
      }
    },
  });

  const onCloseCamera = useCallback(() => {
    setShowCameraModal(false);
  }, []);

  const onCameraError = useCallback(
    (error: any) => {
[... 237 lines cut ...]
### Camera Device

```json
camara doble ancho
```

### Device

redmi note 9s 4gb ram miui 14.0.4

### VisionCamera Version

    "react-native-vision-camera": "^3.6.12",

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

tahaipek, 2023-12-06:
I downgraded the package version to 3.6.6 but it didn't work. 
How can i solve this issue?

tahaipek, 2023-12-06:
I found a temporary solution.

Don't use any function that affects any interface inside the "onCodeScanned" function. Like "useState, useCallback, setXYZ, runOnJS".

When I ran the process without using these functions, I noticed that it didn't give an error when reading the QR code. For now, the problem seems to be solved, the code quality of course decreases a bit.

Can anyone else try this workaround?

```
const codeScanner = useCodeScanner({
    codeTypes: [
      'qr',
      'ean-13',
      'code-128',
      'code-39',
      'code-93',
      'codabar',
      'ean-8',
    ],
    onCodeScanned: codes => {
    ///// This function
    },
  });
```

iliapnmrv, 2023-12-08:
@tahaipek can you please share more data? how can you
> Don't use any function that affects any interface inside the "onCodeScanned" function.

Isn't it the main purpose of `onCodeScanned` method?

tahaipek, 2023-12-08:
Of course this is the task of the "onCodeScanned" function. But here there is an error when the QR code is read. I am looking for a permanent solution for this. For now, I have reached a temporary solution, even if it is bad.

For now, I completed the development by removing the interface interactions. When the QR code is scanned, I showed a message on the screen with "Alert" and redirected the page.

In the meantime, I realized that when the QR code is read, the application crashes as soon as I perform an operation that affects the UI in ReactNative. So I experimented by bypassing the elements that affect the UI one by one. I realized that there was no problem in using the native components of the phone, so I used "Alert". 

I tried this crash issue on Samsung S23, S23 Ultra and android emulator and so far no problem.

If anyone tries differently, we can address the error more clearly. Since my knowledge of native application development is limited, I think the issue is caused by data sharing between threads, I am not sure and I do not want to mislead.

```

export default function CodeScannerPage({ navigation }) {
  const [qrData, setQrData] = useState('-');

  // Use a simple default back camera
  const device = useCameraDevice('back');

  const onCodeScanned = useCallback((codes) => {
    console.log(`Scanned ${codes.length} codes:`, codes);

    setQrData('Scanning QR Data...');

    const value = codes[0]?.value;
    if (value == null) {
      setQrData('Invalid QR Data or something...');
      return;
    }
    setQrData(value);

    // Some QR Scenarios...
  }, []);

  // Initialize the Code Scanner to scan QR codes and Barcodes
  const codeScanner = useCodeScanner({
    codeTypes: ['qr', 'ean-13'],
    onCodeScanned: onCodeScanned,
  });

  return (
    <View>
      <Camera
        style={StyleSheet.absoluteFill}
        device={device}
        isActive={true}
        codeScanner={codeScanner}
      />

      <View>
        <Text>{qrData}</Text>
      </View>
    </View>
  );
}
```

Temporary solution

```

export default function CodeScannerPage({ navigation }) {
  // Use a simple default back camera
  const device = useCameraDevice('back');
  const codeScanner = useCodeScanner({
    codeTypes: ['qr', 'ean-13'],
    onCodeScanned: (codes) => {
      console.log(`Scanned ${codes.length} codes:`, codes);

      const value = codes[0]?.value;
      if (value == null) {
        Alert.alert('Invalid QR Data or something...');
        return;
      }

      Alert.alert('QR Data:', value);

      // Some QR Scenarios...
    },
  });

  return (
    <View>
      <Camera
        style={StyleSheet.absoluteFill}
        device={device}
        isActive={true}
        codeScanner={codeScanner}
      />
    </View>
  );
}

```

Rodricity, 2023-12-20:
I've been struggling with a similar issue for the last two days, cant get the source problem, but seems like navigating away from the camera while it hasnt properly stop triggers this type of crash.

Seems like another way arround is to 'wait' for propper stop of the camera, so what I did was to use the code read to alter the 'isActive' property and then on the 'onStopped' prop of Camera instance fire the navigation/detection to another component.

Here is an example.

`
type Props = { onDetection?: (code: Code) => void };

export function CameraScanner(props: Props) {
	const { onDetection } = props;
	const [QR, setQR] = useState<Code>();

	const isFocused = useIsFocused();
	const appState = useAppState();
	const isActive = appState === "active" && isFocused && QR === undefined;

	// To clear the QR everytime the navigation switchs to this component.
	useEffect(() => {
		if (isFocused && QR) setQR(undefined);
	}, [isFocused]);

	const QRDetector = useCodeScanner({ codeTypes: ["qr"], onCodeScanned: codes => QR === undefined && setQR(codes[0]) });

	const device = useCameraDevice("back");
	if (!device) return <Text>No hemos logrado obtener un dispositivo de cámara</Text>;

	return (
		<Camera
			onStopped={() => QR && onDetection && onDetection(QR)}
			key={device.id}
			device={device}
			video={false}
			audio={false}
			codeScanner={QRDetector}
			style={StyleSheet.absoluteFill}
			isActive={isActive}
		/>
	);
}

`

klabusta, 2024-01-05:
Same here Xiaomi Redmi Note 10, 3.6.16

mrousavy, 2024-01-05:
Hey! Thanks for reporting this issue.

I've been working the past days on making sure the Camera lifecycle is safely handled and released when needed to make sure this crash doesn't occur anymore.

I just created a PR for this - can you please this to see if that fixes the issue for you? https://github.com/mrousavy/react-native-vision-camera/pull/2339

If this fixes your issue, please consider [💖 **sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy/) to support me / thank me for building VisionCamera and continuously improving it. 

If this does not fix your issue, please clone the repo, check out the branch `fix/blackscreen` (the PR above), run the Example app, reproduce your issue there and share the `adb logcat` logs with me (upload to pastebin or gist) so I can investigate this further.

Thank you! 🙏

closed 2024-01-08 by mrousavy

klabusta, 2024-01-11:
Still crashing with 3.7.0, initialization works much better though! My adb logcat looks pretty similar to the one from above :(

klabusta, 2024-01-11:
Similar to https://github.com/mrousavy/react-native-vision-camera/issues/2341#issuecomment-1886916320 I now know how I can prevent it from closing/crashing. I render the Camera component conditionally. The reason is the following:

I have a bottom tab bar and I show the screen with the camera on one tab. But I also show the camera on another tab, when opening a modal (on this modal). To not have two Camera components open at the same time, I had to hide either. @mrousavy I hope that makes sense. This behaviour worked fine with v2.