margelo/react-native-vision-camera #2245: 🐛 - Error while updating property 'cameraId' of a view managed by: CameraView

cauediademi, 2023-12-02:
### What's happening?

Just after updating my app to react-native-vision-camera version 3.6.11, it started crashing every time I make a call to the vision-camera Camera component. However, this bug only seems to occur on Galaxy S20 and Note 10 devices.

See the bug reported by one of my users:
[image]
**Alert translation:**
_An error has occurred with DoctorID
DoctorID has closed because this app has a bug.
Try to update the app after the developer has provided a fix for this error._

### Reproduceable Code

```tsx
/* eslint-disable react-hooks/exhaustive-deps */
import Bugsnag from '@bugsnag/react-native';
import Geolocation from '@react-native-community/geolocation';
import moment from 'moment';
import React, { useEffect, useState } from 'react';
import {
  Platform,
  StyleSheet,
  Text,
  TouchableOpacity,
  View,
} from 'react-native';
import { RNHoleView } from 'react-native-hole-view';
import { PERMISSIONS, request } from 'react-native-permissions';
import { Actions } from 'react-native-router-flux';
import {
  Camera,
  useCameraDevices,
  useCodeScanner,
} from 'react-native-vision-camera';
import { Worklets } from 'react-native-worklets-core';
import { connect } from 'react-redux';
import { heightToDp, widthToDp } from 'rn-responsive-screen';
import Alerta from '../components/common/Alerta';
import { sincronizarChecksRepresados } from '../redux/actions/CalendarioActions';
import { enviarCheckQRCode } from '../redux/actions/CheckActions';
import { setLoadingSpinner } from '../redux/actions/SistemaConfigActions';
import Api, * as util from '../redux/util/api';
import urls from '../redux/util/urls';
import { store } from '../store';
import theme from '../styles/DoctorID.style';
import translate from '../utils/language.utils';

function QRCodeScreen() {
  const [latitude, setLatitude] = React.useState(null);
  const [longitude, setLongitude] = React.useState(null);
  const [tentativa, setTentativa] = React.useState(0);
  const [podeAcessarGPS, setPodeAcessarGPS] = React.useState(false);
  const maxTentativas = 20;

  // TENTAR PEGAR LOCALIZAÇÃO ATÉ maxTentativas OU DESCOBRIR PERMISSÃO NEGADA
  useEffect(() => {
    console.log('useEffect', tentativa);
    store.dispatch(setLoadingSpinner(true));
    podeAcessarLocalizacao();
  }, [tentativa]);

  const podeAcessarLocalizacao = () => {
    if (podeAcessarGPS) {
      // JÁ PERMITIU ACESSO À LOCALIZAÇÃO:
      // Não precisa verificar permissão novamente e tenta pegar localização.
      acessarLocalizacao();
    } else {
      request(
        Platform.select({
          android: PERMISSIONS.ANDROID.ACCESS_FINE_LOCATION,
[... 493 lines cut ...]
  "hasTorch": false,
  "position": "back",
  "id": "4"
}
```

### Device

Galaxy Note10+ (SM-N975F) & Galaxy S20+ (SM-G985F)

### VisionCamera Version

3.6.11

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

cauediademi, 2023-12-16:
Think was fixed in 3.6.15. Uploading my app to production right now.

mrousavy, 2024-01-11:
Hey! I think this is fixed already :)

closed 2024-01-11 by mrousavy