margelo/react-native-vision-camera #2246: iPhone 15 pro can't focus close up documents when using the scan utility.🐛

akichim21, 2023-12-03:
### What's happening?

Since purchased the new iPhone 15 pro, it seems that the iOS app can't automatically switch to the macro lens (the ultra wide lens), so it can't focus small documents (like tickets or barcodes).

This same error occurs in other third party apps.

Apple says is the "developer fault" because they are making the wrong call to the camera function, and so it only uses the main camera, that in the 14 and 15 pro models can't close focus.

### Reproduceable Code

```tsx
const device = useCameraDevice('back');
  const codeScanner = useCodeScanner({
    codeTypes: ['qr', 'ean-13'],
    onCodeScanned: onCodeScanned,
  });

  const _renderCamera = () => {
    if (device == null || !hasPermission || !isFocused) {
      return null;
    }

    return (
      <Camera
        style={styles.qrCamera}
        device={device}
        isActive={true}
        codeScanner={codeScanner}
        enableZoomGesture={true}
        torch={torchStatus}
      />
    );
  };
```

### Relevant log output

```shell
None
```

### Camera Device

```json
None
```

### Device

iPhone15 pro

### VisionCamera Version

3.6.12

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-01-15:
Yes - I can expose `minimumFocusDistance` on the device.

mrousavy, 2024-01-15:
Created a PR: https://github.com/mrousavy/react-native-vision-camera/pull/2391

mlazari, 2024-03-15:
For anyone else looking for a solution, try using this:
```
const device = useCameraDevice('back', {
  physicalDevices: [
    'ultra-wide-angle-camera',
    'wide-angle-camera',
    'telephoto-camera'
  ]
})
```
instead of
```
const device = useCameraDevice('back');
```
This fixed the issue for us.
Documentation: https://react-native-vision-camera.com/docs/guides/devices#selecting-multi-cams

seba9999, 2024-10-31:
@mrousavy Can we use this ``minimumFocusDistance`` in the [`getcameradevice`](https://react-native-vision-camera.com/docs/api/#getcameradevice)'s filters ?

Something similare to the format filters like minFPS ? Would be great to fix the iphone 14 pro ( and more ) problems with thoses new minim focus distances 

Discussed [HERE](https://www.reddit.com/r/iphone/comments/175r2yk/comment/k6e5p6b/?utm_source=share&utm_medium=web3x&utm_name=web3xcss&utm_term=1&utm_content=share_button) and [THERE](https://discussions.apple.com/thread/254228070?answerId=258204420022&sortBy=rank#258204420022)