margelo/react-native-vision-camera #2247: ❓ The screen size is different before and after shooting (Android)

PYSoYa, 2023-12-04:
### Question

  const device = useCameraDevice('back', {
    physicalDevices: ['wide-angle-camera'],
  });

  const cameraRef = useRef(null);

  const format = useCameraFormat(device, [
    {
      photoAspectRatio: Platform.OS === 'android' ? 1 : undefined,
    },
    {
      photoResolution: {
        width: 1080,
        height: 1080,
      },
    },
  ]);

        <Camera
          device={device}
          ref={cameraRef}
          isActive={true}
          photo={true}
          video={false}
          format={format}
          style={{
            width: width,
            height: width,
          }}
        />
[image]
[image]

          

### VisionCamera Version

3.6.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

Felipe-OT, 2023-12-05:
im having the same problem :/

Felipe-OT, 2023-12-12:
Worked. Thanks!

closed 2024-01-15 by mrousavy