margelo/react-native-vision-camera #2256: ❓ Waiting for the barcode module to be downloaded.

erwanlpfr, 2023-12-05:
### Question

Hello,

On my app, I start the app with a Camera View directly after the splash screen.
The CameraView is set to get the Qrcode directly.
On my devices it works even after a fresh install but according to Google Console, when they test I get always 2 crashes on 7.

It was a bit tricky to understand why without logs but I added them and I get :

```
Camera.onError(unknown/unknown): [unknown/unknown] Waiting for the barcode module to be downloaded. Please wait. [unknown/unknown]: [unknown/unknown] Waiting for the barcode module to be downloaded. Please wait.
```

Actually, I am not reporting a bug, but more a strategy to adopt in this case. 
Since the error looks "unknown", what can it be, the best strategy to wait and to retry?

```
// The full error object
{
arguments: [
Camera.onError(unknown/unknown): [unknown/unknown] Waiting for the barcode module to be downloaded. Please wait., 
{
_cause: {}, 
_code: unknown/unknown, 
_message: [unknown/unknown] Waiting for the barcode module to be downloaded. Please wait., 
message: [unknown/unknown] Waiting for the barcode module to be downloaded. Please wait., 
name: unknown/unknown, 
stack: 
[unknown/unknown]: [unknown/unknown] Waiting for the barcode module to be downloaded. Please wait.
    at construct (native)
    at Wrapper (address at index.android.bundle:1:197548)
    at construct (native)
    at _createSuperInternal (address at index.android.bundle:1:2492550)
    at call (native)
    at CameraError (address at index.android.bundle:1:2493014)
    at construct (native)
    at _createSuperInternal (address at index....
}
]
}
```

Thank you!

### VisionCamera Version

3.6.13

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

erwanlpfr, 2023-12-13:
Tried and it works!
If I have time, I will suggest a PR to add it on expo plugin

kyjus25, 2024-01-04:
> Try adding this:
> 
> ```
> dependencies {
>   // ...
>   // Use this dependency to bundle the model with your app
>   implementation 'com.google.mlkit:barcode-scanning:17.2.0'
> }
> ```
> 
> to your app/build.gradle file.

@erwanlpfr is there a method that works for Expo? As far as I know I can't run a deployment because it requires a manual change to the android gradle file. Any luck on that PR?

erwanlpfr, 2024-01-04:
Hello !
I tried this : https://github.com/mrousavy/react-native-vision-camera/pull/2348
Let me know if it works for you!

kyjus25, 2024-01-04:
@erwanlpfr Fantastic! Thank you so much for the PR and quick reply!

closed 2024-01-08 by mrousavy