margelo/react-native-vision-camera #2257: 🐛 Camera has been disconnected

gsagergithub, 2023-12-06:
### What's happening?

Hi everyone. I find myself surrounded by the darkest darkness trying to use the rear camera of my Motorola Moto g6 plus, since I'm getting multiple errors. When I navigate to the screen containing the Camera component this one shows up:
`Camera.onError(session/camera-has-been-disconnected): [session/camera-has-been-disconnected] The given Camera device (id: 0) has been disconnected! Error: UNKNOWN_CAMERA_DEVICE_ERROR [session/camera-has-been-disconnected: [session/camera-has-been-disconnected] The given Camera device (id: 0) has been disconnected! Error: UNKNOWN_CAMERA_DEVICE_ERROR]`
And then, navigating back and forth again I'm getting these other two
`Camera.onError(unknown/unknown): [unknown/unknown] CameraDevice was already closed [unknown/unknown: [unknown/unknown] CameraDevice was already closed]`

`Camera.onError(session/camera-not-ready): [session/camera-not-ready] The Camera is not ready yet! Wait for the onInitialized() callback! [session/camera-not-ready: [session/camera-not-ready] The Camera is not ready yet! Wait for the onInitialized() callback!]`

I don't want to leave without telling you that I've tested the native camera application of the phone  and it's working properly,  and that I have at least three other devices where the application that integrates react native vision camera, runs well. One of them is a Motorola Moto e6 Plus. 

I would appreciate any kind of help/guidance to solve this issue. Thank you!

### Reproduceable Code

```tsx
const device = useCameraDevice('back');

<View>
  {
    isActive &&
    <Camera
      style={styles.container}
      device={device}
      isActive={isActive}
      codeScanner={codeScanner}
      enableZoomGesture={true}
    />
  }
</View>
```

### Relevant log output

```shell
2023-12-06 15:16:08.545 10591-10591 CameraView              pid-10591                            I  Updating CameraSession...
2023-12-06 15:16:08.547 10591-11047 CameraSession           pid-10591                            I  Updating CameraSession Configuration...
2023-12-06 15:16:08.548 10591-11047 CameraSession           pid-10591                            I  Successfully updated CameraSession Configuration! isActive: false
2023-12-06 15:16:08.559 10591-10591 CameraSession           pid-10591                            I  PreviewView Surface destroyed! Surface(name=null)/@0xff010d1
2023-12-06 15:16:08.559 10591-10591 CameraSession           pid-10591                            I  Destroying Preview Output...
2023-12-06 15:16:08.561 10591-10591 CameraSession           pid-10591                            I  Updating CameraSession Configuration...
2023-12-06 15:16:08.562 10591-10591 CameraSession           pid-10591                            I  Configuring Session for Camera #0...
2023-12-06 15:16:08.567 10591-10591 CameraSession           pid-10591                            I  Adding 1280 x 720 CodeScanner Output in Format #35...
2023-12-06 15:16:08.569   543-11051 ULog                    camerahalserver                      I  R AppRequest:2929 M[CameraDevice:187001] -  :mtkcam-AppStreamMgr #23595
2023-12-06 15:16:08.572   526-672   Camera3-OutputStream    cameraserver                         E  returnBufferCheckedLocked: Stream 2: Error queueing buffer to native window: No such device (-19)
2023-12-06 15:16:08.572   526-672   Camera3-Device          cameraserver                         E  Can't return buffer to its stream: No such device (-19)
2023-12-06 15:16:08.575 10591-10591 CreateCaptureSession    pid-10591                            I  Camera 0: Creating Capture Session #1002... Hardware Level: 1} | Outputs: [android.hardware.camera2.params.OutputConfiguration@28061169]
2023-12-06 15:16:08.637   543-11051 ULog                    camerahalserver                      I  R AppRequest:2930 M[CameraDevice:187001] -  :mtkcam-AppStreamMgr #23596
2023-12-06 15:16:08.639   526-672   Camera3-OutputStream    cameraserver                         E  returnBufferCheckedLocked: Stream 2: Error queueing buffer to native window: No such device (-19)
2023-12-06 15:16:08.639   526-672   Camera3-Device          cameraserver                         E  Can't return buffer to its stream: No such device (-19)
2023-12-06 15:16:08.721   543-11051 ULog                    camerahalserver                      I  R AppRequest:2931 M[CameraDevice:187001] -  :mtkcam-AppStreamMgr #23597
2023-12-06 15:16:08.724   526-672   Camera3-OutputStream    cameraserver                         E  returnBufferCheckedLocked: Stream 2: Error queueing buffer to native window: No such device (-19)
2023-12-06 15:16:08.724   526-672   Camera3-Device          cameraserver                         E  Can't return buffer to its stream: No such device (-19)
2023-12-06 15:16:08.794   543-11051 ULog                    camerahalserver                      I  R AppRequest:2932 M[CameraDevice:187001] -  :mtkcam-AppStreamMgr #23598
2023-12-06 15:16:08.797   526-672   Camera3-OutputStream    cameraserver                         E  returnBufferCheckedLocked: Stream 2: Error queueing buffer to native window: No such device (-19)
2023-12-06 15:16:08.797   526-672   Camera3-Device          cameraserver                         E  Can't return buffer to its stream: No such device (-19)
2023-12-06 15:16:08.862   543-11051 ULog                    camerahalserver                      I  R AppRequest:2933 M[CameraDevice:187001] -  :mtkcam-AppStreamMgr #23599
2023-12-06 15:16:08.865   526-672   Camera3-OutputStream    cameraserver                         E  returnBufferCheckedLocked: Stream 2: Error queueing buffer to native window: No such device (-19)
2023-12-06 15:16:08.865   526-672   Camera3-Device          cameraserver                         E  Can't return buffer to its stream: No such device (-19)
2023-12-06 15:16:08.865 10591-10591 Binder                  pid-10591                            D  <<<<< Spending more than 290 ms on Binder:android.hardware.camera2.ICameraDeviceUser#13
2023-12-06 15:16:08.867   526-999   CameraLatencyHistogram  cameraserver                         I  Stream 1 dequeueBuffer latency histogram (2934) samples:
2023-12-06 15:16:08.867   526-999   CameraLatencyHistogram  cameraserver                         I          5     10     15     20     25     30     35     40     45    inf (max ms)
2023-12-06 15:16:08.867   526-999   CameraLatencyHistogram  cameraserver                         I       100.00   0.00   0.00   0.00   0.00   0.00   0.00   0.00   0.00   0.00 (%)
2023-12-06 15:16:08.869   526-999   CameraLatencyHistogram  cameraserver                         I  Stream 2 dequeueBuffer latency histogram (2934) samples:
2023-12-06 15:16:08.869   526-999   CameraLatencyHistogram  cameraserver                         I          5     10     15     20     25     30     35     40     45    inf (max ms)
2023-12-06 15:16:08.869   526-999   CameraLatencyHistogram  cameraserver                         I       99.97   0.03   0.00   0.00   0.00   0.00   0.00   0.00   0.00   0.00 (%)
2023-12-06 15:16:08.887   543-543   mtkcam-AppStreamMgr     camerahalserver                      W  [0-ConfigHandler::checkStream] framework stream dataspace:0x08c20000(V0_JFIF|STANDARD_BT601_625|TRANSFER_SMPTE_170M|RANGE_FULL) {.v3_2 = {.id = 3, .streamType = OUTPUT, .width = 1280, .height = 720, .format = YCBCR_420_888, .usage = CPU_READ_NEVER | CPU_READ_RARELY | CPU_READ_OFTEN | CPU_WRITE_NEVER (0x3), .dataSpace = UNKNOWN | STANDARD_UNSPECIFIED | STANDARD_BT601_625 | TRANSFER_UNSPECIFIED | TRANSFER_LINEAR | TRANSFER_SRGB | TRANSFER_SMPTE_170M | RANGE_UNSPECIFIED | RANGE_FULL | V0_JFIF (0x8c20000), .rotation = ROTATION_0}, .physicalCameraId = "", .bufferSize = 0}
2023-12-06 15:16:08.889   543-543   BWC                     camerahalserver                      I  Profile_Change:[BWCPT_CAMERA_CAPTURE]:OFF,current concurrency is 0x0,DDR=,EMI scenario=BWL_SCN_ICFP OFF
2023-12-06 15:16:08.919   543-543   BWC                     camerahalserver                      I  Profile_Change:[BWCPT_CAMERA_CAPTURE]:ON,current concurrency is 0x10,DDR=,EMI scenario=BWL_SCN_ICFP ON
2023-12-06 15:16:08.922   543-543   MtkCam/ppl_context      camerahalserver                      I  [dump]     [APP     ] 03 1280x720  OUT t:0/r:0 maxBuffers:11 d/s:0x08c20000(V0_JFIF|STANDARD_BT601_625|TRANSFER_SMPTE_170M|RANGE_FULL) Hal-Client-usage:0x3(0|SW_READ_OFTEN) Real:0x32315659(YV12) Request:0x23(YCbCr_420_888) Override:0x23(YCbCr_420_888) Hal-usage:0x20033(0|SW_READ_OFTEN|SW_W [... cut ...]
### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

SayeedAfridi, 2023-12-08:
+1
Getting the same error.

raerta, 2023-12-10:
I got this error too. If you're getting the error but can use the camera, you can use onError to prevent the error from appearing in the app.

 onError={handleCameraError}
 
 `<Camera
        lowLightBoost
        enableZoomGesture
        zoom={2}
        style={StyleSheet.absoluteFill}
        device={device}
        isActive={isActive}
        codeScanner={codeScanner}
        focusable
        torch={torchIsOpen ? 'on' : 'off'}
        onError={handleCameraError}
      />`

`const handleCameraError = (e: any) => {
    console.log('error', e);
  };`

gsagergithub, 2023-12-11:
> I got this error too. If you're getting the error but can use the camera, you can use onError to prevent the error from appearing in the app.
>
> onError={handleCameraError}
>
> `<Camera lowLightBoost [...]

I can't use the camera at all. It shows as a black screen.

raerta, 2023-12-16:
Change your conditional rendering logic like this:

 `{
    device &&
    <Camera
      style={styles.container}
      device={device}
      isActive={isActive}
      codeScanner={codeScanner}
      enableZoomGesture={true}
    />
  }`

MicaelaWaigel, 2023-12-18:
> Change your conditional rendering logic like this:
> 
> `{ device && <Camera style={styles.container} device={device} isActive={isActive} codeScanner={codeScanner} enableZoomGesture={true} /> }`

It doesn't work anyway..

DevVibhor, 2023-12-18:
Facing the same issue.

haouarihk, 2023-12-18:
For me its super unstable and causes crashes everytime i either switch the camera
Or navigate to and from another page using the navigation library.
And sometimes when i show a overlay on top of the view.

It caused me a day of debugging, and it's still happening.
I'm on min sdk 26.
I use react native without expo bloat

Btw changing
 from `if(device == null) return ;`
 to `if(!device) return;` 
helped alot but still not enough 

Apparently device can be undefined(i think). Which is contradictory to the docs.

Thanks
IHateRNToDontWorry :)

paulovictor237, 2023-12-18:
I'm going through the same problem.
Android 12 and below: presents the same error
Android 13 and above: works normally

However, 25% of my app's user base are on Android version 12, making the use of the library unfeasible.

engrwaqasali, 2023-12-20:
+1
getting the same error
on first render the camera shows up correctly
but when navigate back and forth its black screen

engrwaqasali, 2023-12-20:
> +1 getting the same error code and video is attached on first render the camera shows up correctly but when navigate back and forth its black screen
> 
> i noticed one thing when i change the navigator from bottom tab, the camera then works just fine
> 
> ```
> import React, { ReactNode, useEffect } from 'react';
> import { Image, Linking, StyleSheet, View } from 'react-native';
> import {
>   Camera,
>   useCameraDevice,
>   useCameraPermission,
>   useCodeScanner,
> } from 'react-native-vision-camera';
> import { Colors } from '../theme';
> import { AppText, Button } from './theme';
> type Props = {
>   isActive: boolean;
>   onScan: (codes: string) => void;
>   children?: ReactNode | JSX.Element;
> };
> 
> const QRCodeScanner = ({ isActive, onScan, children }: Props) => {
>   const { hasPermission, requestPermission } = useCameraPermission();
>   const device = useCameraDevice('back');
> 
>   const codeScanner = useCodeScanner({
>     codeTypes: ['qr', 'ean-13'],
>     onCodeScanned: (codes) => {
>       console.log(`Scanned ${codes.length} codes!`);
>       if (codes?.length > 0) {
>         if (codes[0].value) {
>           console.log('Scanned code', codes[0].value);
>           onScan(codes[0].value);
>         }
>       }
>     },
>   });
> 
>   const getPermission = async () => {
>     if (!hasPermission) {
>       const res = await requestPermission();
>       if (!res) {
>         Linking.openSettings();
>       }
>     }
>   };
> 
>   useEffect(() => {
>     getPermission();
>   }, [hasPermission]);
> 
>   if (!hasPermission || !device)
>     return (
>       <View
>         style={{
>           flex: 1,
>           alignItems: 'center',
>           justifyContent: 'center',
>           backgroundColor: Colors.background,
>           margin: 32,
>           gap: 24,
>         }}
>       >
>         <AppText>Shido needs access to Camera</AppText>
>         <Button
>           title={'Open Settings'}
>           onPress={() => Linking.openSettings()}
>         />
>       </View>
>     );
>   return (
>     <View style={styles.container}>
>       <Camera
>         key={device.id}
>         isActive={isActive}
>         device={device}
>         codeScanner={codeScanner}
>         style={StyleSheet.absoluteFill}
>       />
>       <Image
>         style={styles.image}
>         source={require('../assets/images/Frame.png')}
>       />
>       {children}
>     </View>
>   );
> };
> 
> export default QRCodeScanner;
> 
> const styles = StyleSheet.create({
>   container: {
>     flex: 1,
>     padding: 24,
>     alignItems: 'center',
>     justifyContent: 'center',
>   },
>   closeButton: {
>     position: 'absolute',
>     top: 24,
>     left: 24,
>   },
>   image: {
>     width: 240,
>     height: 240,
>     alignSelf: 'center',
>     alignItems: 'center',
>     justifyContent: 'center',
>   },
> });
> ```
> 
>  WhatsApp.Video.2023-12-20.at.2.10.52.PM.mp4

https://github.com/mrousavy/react-native-vision-camera/issues/2291#issuecomment-1862274019 this solution worked for me for now

paulovictor237, 2024-01-13:
I managed to solve this bug by activating the `video` option. It's not a bug-fix, but it worked for me. Try doing the same as me and see if it works for you too.

```javascript
function App() {
  const device = useCameraDevice('back')
  if (!device) return null
  return (
    <Camera
      style={StyleSheet.absoluteFill}
      device={device}
      isActive={true}
      photo={true}
      video={true}
    />
  )
}
```

mrousavy, 2024-01-15:
I think this has been fixed in the latest version (here on main)

Please try the example app and let me know if the problem is gone!

closed 2024-01-15 by mrousavy

bglgwyng, 2024-01-17:
I'm still facing the same issue. I opened the example app and repeated flipping the direction.
My device is Galaxy S10, Android 12.

https://gist.github.com/bglgwyng/cfe64520b4fbd56a5c74fb153e892f4d

cauediademi, 2024-01-30:
> video

This works for me. I am having the same issue with Motorola Moto G5 in v3.8.2

fikretburhan, 2024-02-09:
Is there any progress on this matter, the problem still continues.

mrousavy, 2024-02-13:
Is this fixed in 3.9.x @bglgwyng ?

abakers, 2024-02-22:
> Is this fixed in 3.9.x @bglgwyng ?

Nope, we're still experiencing this issue. Version 3.9.0

Henriquehlr, 2024-02-26:
I'm facing the same issue on Android 12 when navigating between screens

version: 3.9.0

MarySnopok, 2024-03-01:
> I think this has been fixed in the latest version (here on main)
> 
> Please try the example app and let me know if the problem is gone!

+1

It was not fixed , we experience this issue for Android version >= 11 . Please reopen the issue

perseon, 2024-03-22:
Samsung S20FE / Android 13. Same error running the example app.

juicydevka, 2024-04-15:
The same 3.9.2

matheuscaet, 2024-04-23:
I'm getting this issue at 3.7.0 version, It's closed but I didn't get the resolution.
Should I upgrade to which version?

ADaRShsiNgh456, 2024-07-17:
Okk