margelo/react-native-vision-camera #2259: (ANDROID) Camera black screen when using "codeScanner" on second time screen gets focus 🐛

abdalamichel, 2023-12-07:
### What's happening?

ONLY IN ANDROID:

Using codeScanner, the camera breaks and remains black when moving from screens with react navigation (tabs)
The first time the screen gets mounted it works perfect.

The same code without codeScanner works perfect.

I tried 3 scenarios to fix it, but all have errors:
1) Setting the camera always active (isActive = true) and conditional rendering the while camera with useIsFocused();
     Result: black screen when screen gets focused on second time
2) Assign the camera property "isActive" to the result of useIsFocused()
     Result: app breaks when screen gets focused on second time
3) Setting the camera always active (isActive = true) 
     Result: app breaks when screen gets focused on second time

Version: "react-native-vision-camera": "^3.6.13",

### Reproduceable Code

```tsx
SCENARIO (1)

import { StyleSheet, Text } from "react-native";
import {
  Camera,
  useCameraDevices,
  useCodeScanner,
} from "react-native-vision-camera";

import type { NativeStackScreenProps } from "@react-navigation/native-stack";
import { useEffect, useState } from "react";
import { NavigationLinks } from "../../Navigations/NavigationLinks";
import { RoutesType } from "../../Navigations/RoutesType";
import { decodeAfipQR } from "../../Utils/afipHelper";
import { useIsFocused } from "@react-navigation/native";
import Loader from "../../components/Loader";

// @ts-ignore

type PropsType = NativeStackScreenProps<RoutesType, NavigationLinks.QR_SCANNER>;

export const QRScanner = ({ navigation }: PropsType) => {
  const [permissionProcessed, setPermissionProcessed] = useState(false);
  const [hasPermission, setHasPermission] = useState(false);
  const devices = useCameraDevices();
  const device = devices[0];

  const [barCodes, setBarCodes] = useState<string[]>([]);
  const focused = useIsFocused();
  const [isActiveCamera, setIsActiveCamera] = useState(true);

  const codeScanner = useCodeScanner({
    codeTypes: ["qr", "ean-13"],
    onCodeScanned: (codes) => {
      //       console.log(`Scanned ${codes.length} codes!`);

      try {
        setBarCodes(
          codes
            .filter((code) => {
              return code.value;
            })
            .map((code) => {
              if (!code.value) return "";

              return code.value;
            })
        );
      } catch (e) {
        alert("Error al procesar QR");
      }
    },
  });

  useEffect(() => {
    const doAsync = async () => {
      const status = await Camera.requestCameraPermission();

      console.log("PERMISSION: ", status);

      setPermissionProcessed(true);

      if (status === "granted") {
        setHasPermission(true);
      } else {
        setHasPermission(false);
      }
    };

    doAsync();
  }, []);

  useEffect(() => {
    setIsActiveCamera(focused);
    alert("Is active 2!!!: " + focused);
  }, [focused]);

  if (permissionProcessed && !hasPermission)
    return <Text>Camera permission is requiered</Text>;

  if (!device || !hasPermission) return <Text>Missing permission</Text>;

  if (!isActiveCamera) return <Loader />;

  return (
    <>
      <Camera
        style={StyleSheet.absoluteFill}
        device={device}
        isActive={true}
        codeScanner={codeScanner}
      />
    </>
  );
};

export default QRScanner;

SCENARIO (2)

import { StyleSheet, Text } from "react-native";
import {
  Camera,
  useCameraDevices,
  useCodeScanner,
} from "react-native-vision-camera";

import type { NativeStackScreenProps } from "@react-navigation/native-stack";
import { useEffect, useState } from "react";
import { NavigationLinks } from "../../Navigations/NavigationLinks";
import { RoutesType } from "../../Navigations/RoutesType";
import { decodeAfipQR } from "../../Utils/afipHelper";
import { useIsFocused } from "@react-navigation/native";
import Loader from "../../components/Loader";

// @ts-ignore

type PropsType = NativeStackScreenProps<RoutesType, NavigationLinks.QR_SCANNER>;

export const QRScanner = ({ navigation }: PropsType) => {
  const [permissionProcessed, setPermissionProcessed] = useState(false);
  const [hasPermission, setHasPermission] = useState(false);
  const devices = useCameraDevices();
  const device = devices[0];

  const [barCodes, setBarCodes] = useState<string[]>([]);
  const focused = useIsFocused();
  const [isActiveCamera, setIsActiveCamera] = useState(true);

  const codeScanner = useCodeScanner({
    codeTypes: ["qr", "ean-13"],
    onCodeScanned: (codes) => {
      //       console.log(`Scanned ${codes.length} codes!`);

      try {
        setBarCodes(
          codes
            .filter((code) => {
              return code.value;
            })
            .map((code) => {
              if (!code.value) return "";

              return code.value;
            })
        );
      } catch (e) {
        alert("Error al procesar QR");
      }
    },
  });

  useEffect(() => {
    const doAsync = async () => {
      const status = await Camera.requestCameraPermission();

      console.log("PERMISSION: ", status);

      setPermissionProcessed(true);

      if (status === "granted") {
        setHasPermission(true);
      } else {
        setHasPermission(false);
      }
    };

    doAsync();
  }, []);

  useEffect(() => {
    setIsActiveCamera(focused);
    alert("Is active 2!!!: " + focused);
  }, [focused]);

  if (permissionProcessed && !hasPermission)
    return <Text>Camera permission is requiered</Text>;

  if (!device || !hasPermission) return <Text>Missing permission</Text>;

  return (
    <>
      <Camera
        style={StyleSheet.absoluteFill}
        device={device}
        isActive={isActiveCamera}
        codeScanner={codeScanner}
      />
    </>
  );
};

export default QRScanner;

SCENARIO (3)

import { StyleSheet, Text } from "react-native";
import {
  Camera,
  useCameraDevices,
  useCodeScanner,
} from "react-native-vision-camera";

import type { NativeStackScreenProps } from "@react-navigation/native-stack";
import { useEffect, useState } from "react";
import { NavigationLinks } from "../../Navigations/NavigationLinks";
import { RoutesType } from "../../Navigations/RoutesType";
import { decodeAfipQR } from "../../Utils/afipHelper";
import { useIsFocused } from "@react-navigation/native";
import Loader from "../../components/Loader";

// @ts-ignore

type PropsType = NativeStackScreenProps<RoutesType, NavigationLinks.QR_SCANNER>;

export const QRScanner = ({ navigation }: PropsType) => {
  const [permissionProcessed, setPermissionProcessed] = useState(false);
  const [hasPermission, setHasPermission] = useState(false);
  const devices = useCameraDevices();
  const device = devices[0];

  const [barCodes, setBarCodes] = useState<string[]>([]);

  const codeScanner = useCodeScanner({
    codeTypes: ["qr", "ean-13"],
    onCodeScanned: (codes) => {
      //       console.log(`Scanned ${codes.length} codes!`);

      try {
        setBarCodes(
          codes
            .filter((code) => {
              return code.value;
            })
            .map((code) => {
              if (!code.value) return "";

              return code.value;
            })
        );
      } catch (e) {
        alert("Error al procesar QR");
      }
    },
  });

  useEffect(() => {
    const doAsync = async () => {
      const status = await Camera.requestCameraPermission();

      console.log("PERMISSION: ", status);

      setPermissionProcessed(true);

      if (status === "granted") {
        setHasPermission(true);
      } else {
        setHasPermission(false);
      }
    };

    doAsync();
  }, []);

  if (permissionProcessed && !hasPermission)
    return <Text>Camera permission is requiered</Text>;

  if (!device || !hasPermission) return <Text>Missing permission</Text>;

  return (
    <>
      <Camera
        style={StyleSheet.absoluteFill}
        device={device}
        isActive={true}
        codeScanner={codeScanner}
      />
    </>
  );
};

export default QRScanner;
```

### Relevant log output

```shell
NA
```

### Camera Device

```json
NA
```

### Device

Any Android

### VisionCamera Version

3.6.13

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

flaviodamaiajr, 2023-12-07:
I have the same issue! Any idea?

richterCLX, 2023-12-11:
Hey, this is the solution that seems to be working for me right now.

based on focus i replace the onCodeScanned function.

```js
 const qrcode = useMemo(() => {
    return {
      codeTypes: ['qr'],
      onCodeScanned: !focus
        ? () => null
        : qr => {
            // your qr code
          },
    };
  }, [focus]);
```

for the useCodeScanner hook, on refocus i get these errors on refocus, i am not sure if they are releated

```
ERROR  camera error [session/camera-has-been-disconnected: [session/camera-has-been-disconnected]
  The given Camera device (id: 0) has been disconnected! Error: CAMERA_IS_DISABLED_BY_ANDROID]
ERROR  camera error [unknown/unknown: [unknown/unknown] CameraDevice was already closed]
ERROR  camera error [unknown/unknown: [unknown/unknown] CameraDevice was already closed]
ERROR  camera error [session/camera-not-ready: [session/camera-not-ready]
  The Camera is not ready yet! Wait for the onInitialized() callback!]
```

FazilMuhammed, 2023-12-15:
I have same issue anyone worked

richterCLX, 2023-12-15:
It did not work as i hoped it would.

Currently i use some a react-navigation screen option unmountOnBlur to rerender the camera screen, whenever the screen is navigated to, then on error of the camera i navigate to the my homescreen.

For Replication of the error: Open your Camera Screen, bring it to background, use the camera in another app.

narayan-singh-solanki, 2023-12-19:
I'm also facing the same issue. Additionally, the second time when the camera is focused, it doesn't read the barcode, and I receive a warning message like 'The camera is not ready yet! Wait for the onInitialized() callback!'

Nadimkhan120, 2024-01-02:
i am also having the same issue, @mrousavy can you please check.

mrousavy, 2024-01-05:
Hey! Thanks for reporting this issue.

I've been working the past days on making sure the Camera lifecycle is safely handled and released when needed to make sure this crash doesn't occur anymore.

I just created a PR for this - can you please this to see if that fixes the issue for you? https://github.com/mrousavy/react-native-vision-camera/pull/2339

If this fixes your issue, please consider [💖 **sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy/) to support me / thank me for building VisionCamera and continuously improving it. 

If this does not fix your issue, please clone the repo, check out the branch `fix/blackscreen` (the PR above), run the Example app, reproduce your issue there and share the `adb logcat` logs with me (upload to pastebin or gist) so I can investigate this further.

Thank you! 🙏

closed 2024-01-08 by mrousavy