margelo/react-native-vision-camera #2261: Dual Capture not Possible🐛

willburstein, 2023-12-07:
### What's happening?

I want to be able to simultaneously capture too both the front and rear-facing cameras

### Reproduceable Code

```tsx
const devices = useCameraDevices();
const [front,back] = devices;

return (<>
<Camera ref={camera} photo={true} device={front} style={{ height: "100%", width: "100%" }}
              isActive={isActive} />
<Camera ref={camera} photo={true} device={back} style={{ height: "100%", width: "100%" }}
              isActive={isActive} />

</>)
```

### Relevant log output

```shell
N/A, only initializes one camera
```

### Camera Device

```json
N/A, only initializes one camera
```

### Device

iPhone XR

### VisionCamera Version

3.1.6

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

james-cohen, 2023-12-08:
Is it because you're using the same ref for both cameras? I use dual camera in my app without issue, the only caveat is you have to initialise one before the other

mrousavy, 2024-01-15:
This is not yet supported. There is a similar issue where I explained why, but I can't find it righgt now

closed 2024-01-15 by mrousavy