margelo/react-native-vision-camera #2265: Xcode Archive Error: Undefined symbols for WorkletRuntime::runGuarded 🔧

jenskuhrjorgensen, 2023-12-08:
### How were you trying to build the app?

Hi

I've run into an Xcode archive issue when using `react-native-vision-camera` v2.16.5 together with the newest version of `react-native-reanimated` (3.6.1). The issue arises because `react-native-reanimated` introduced some branching behaviour in v3.6.0 which relies on the `PRODUCTION` flag used when installing pods. This means that you have to run `PRODUCTION=1 pod install` in order to be able to create iOS release builds. In v3.5.4 of `react-native-reanimated` it was enough to just run `pod install` before creating iOS release builds. This is an issue for us because we have pods committed to the repo so we cannot rely on two different sets on pods based on whether you need to create debug or release builds.

I've filed an issue with `react-native-reanimated` (https://github.com/software-mansion/react-native-reanimated/issues/5465) but thought this might be relevant for users of `react-native-vision-camera` since that combination was what broke it for me. And also I hope that someone in here can come up with a solution to make it work. I don't know if the issue exists with `react-native-vision-camera` v3, but we cannot use that yet because it doesn't work with JSC (and we cannot yet update to Hermes due to its excluded JS features). 

The issue can be reproduced with the `react-native-vision-camera` example app when installing `react-native-reanimated` v3.6.1 and then making a call to `WorkletRuntime::runGuarded` e.g. like this:

```
#import "AppDelegate.h"

#import <React/RCTBundleURLProvider.h>
#import <RNReanimated/WorkletRuntime.h>

@implementation AppDelegate

std::shared_ptr<reanimated::WorkletRuntime> workletRuntime;

- (BOOL)application:(UIApplication *)application didFinishLaunchingWithOptions:(NSDictionary *)launchOptions
{
  self.moduleName = @"VisionCameraExample";
  // You can add your custom initial props in the dictionary below.
  // They will be passed down to the ViewController used by React Native.
  self.initialProps = @{};

  jsi::Runtime& runtime = *(jsi::Runtime*)"@123";
  jsi::Value value = jsi::Value(42);
  auto rt = reanimated::extractWorkletRuntime(runtime, value);
  jsi::Runtime &rtt = rt->getJSIRuntime();
  auto worklet = reanimated::extractShareableOrThrow<reanimated::ShareableWorklet>(rtt, value);
  workletRuntime = reanimated::extractWorkletRuntime(runtime, value.asObject(runtime));
  std::weak_ptr<reanimated::WorkletRuntime> weakWorkletRuntime = workletRuntime;
  auto workletRuntime = weakWorkletRuntime.lock();
  workletRuntime->runGuarded(worklet);

  return [super application:application didFinishLaunchingWithOptions:launchOptions];
}

- (NSURL *)sourceURLForBridge:(RCTBridge *)bridge
{
#if DEBUG
  return [[RCTBundleURLProvider sharedSettings] jsBundleURLForBundleRoot:@"index"];
#else
  return [[NSBundle mainBundle] URLForResource:@"main" withExtension:@"jsbundle"];
#endif
}

/// This method controls whether the `concurrentRoot`feature of React18 is turned on or off.
///
/// @see: https://reactjs.org/blog/2022/03/29/react-v18.html
/// @note: This requires to be rendering on Fabric (i.e. on the New Architecture).
/// @return: `true` if the `concurrentRoot` feature is enabled. Otherwise, it returns `false`.
- (BOOL)concurrentRootEnabled
{
  return true;
}

@end

### Full build logs

```tsx
ld: Undefined symbols:
  reanimated::getCallGuard(facebook::jsi::Runtime&), referenced from:
      void reanimated::runOnRuntimeGuarded<>(facebook::jsi::Runtime&, facebook::jsi::Value const&) in AppDelegate.o
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

### Project dependencies

```json
"dependencies": {
  "react-native": "0.72.6",
  "react-native-vision-camera": "2.16.5",
  "react-native-reanimated": "3.6.1",
  ...
}
```

### VisionCamera Version

2.16.5

### Target platforms

iOS

### Operating system

MacOS

### Can you build the VisionCamera Example app?

Yes, I can successfully build the Example app here

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

tomekzaw, 2023-12-08:
@jenskuhrjorgensen Can you check if #2270 fixes the problem?

jenskuhrjorgensen, 2023-12-08:
Thanks a lot @tomekzaw! I can confirm that #2270 fixes the issue 🥳

mrousavy, 2024-01-15:
Awesome work tomasz!

closed 2024-01-15 by mrousavy