margelo/react-native-vision-camera #2268: Pause device audio when recording starts❓

james-cohen, 2023-12-08:
### Question

If I have music playing on my device (iOS and Android), when I record using the native camera app the audio will pause until the recording has finished. However, when using this package the audio continues in the background when recording. Are there any methods or variables available for setting the audio mixing properties?

In [https://github.com/mrousavy/react-native-vision-camera/issues/448#issuecomment-925735438](this closed issue) it shows that .playAndRecord is already implemented. Is this a bug and should external audio be paused during recording?

### What I tried

Recording video on iOS and Android with music app playing in the background. Audio continues. Then try the same using native camera app, audio pauses and recommences when recording complete

### VisionCamera Version

3.6.13

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2024-01-02:
This is per design! Audio is automatically muxed to provide a seamless user experience - just like in Snapchat or Instagram.

mrousavy, 2024-01-02:
There is currently no way to configure that behaviour, so please create a feature request or contact me through my agency so we can develop that feature for you :)

closed 2024-01-02 by mrousavy