margelo/react-native-vision-camera #2274: What is the recommended way to draw landmarks over the video❓

adrian-bobev, 2023-12-10:
### Question

What is the recommended way to draw position landmarks of body parts over the live video?

### What I tried

I tried to create a simple MLKit Pose detection FrameProcessor. At the moment I am successfully logged the detected pose landmarks in the console. However, I am wondering what is the recommended way to draw those landmarks.

I saw some mentions of Skia, but not sure what is the current state and how I should use it.

### VisionCamera Version

3.6.13

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

rodgomesc, 2023-12-11:
i’m using react-native-skia with stable 30fps

adrian-bobev, 2023-12-12:
Thabjs for the answer. Just for future reference what I have done is the following:

```jsx
export default function App() {
  const [position, setPosition] = useState();
  const device = useCameraDevice('front');

  const onObjecteDetected = Worklets.createRunInJsFn(pos => {
    setPosition(pos);
  });

  const frameProcessor = useFrameProcessor(frame => {
    'worklet';

    const obj = detectObject(frame);
    
    onObjecteDetected({
      x: obj?.x,
      y: obj?.y
    });
  }, []);

  if (!device) {
    return null;
  }
  return (
    <>
      <Camera
        frameProcessor={frameProcessor}
        style={StyleSheet.absoluteFill}
        video={true}
        device={device}
        isActive={true}
        videoHdr={false}
      />
      <Canvas style={StyleSheet.absoluteFill}>
        <Circle cx={position.x} cy={position.y} r={10} color="red" />
      </Canvas>
    </>
  );
}
```

I am just using createRunInJsFn in order to update the state of the app component and to rerender it on every frame and use the position properties to draw.

rodgomesc, 2023-12-12:
try to stay far from the javascript thread, use `useValue` from skia instead of `useState`

adrian-bobev, 2023-12-13:
I manage to make it work. However, the useValue works only if the the value stored in it is number, I tried to use object  {
x: 10,
y: 20
} 

for example and it did not work.

PS: However, I just realized that this is not exactly I wont, because the skia overlay is not recorded in the video since it is not part of the frame. I guess in order to do it, I need to do the landmarks drawing in the native frame processor, am I right?

mrousavy, 2023-12-13:
Yea, recording that will be quite tricky.

The native Skia implementation would've handled all of that, but as @rodgomesc said, I removed it from the repo because it got way too complex for all the people that were not using Skia.

adrian-bobev, 2023-12-13:
@mrousavy, was there an issue with the implementation with native skia implemnetation or it was just because of the complexity that this change will bring to the library?

In other words, is it safe to find the change and use it or there was unresolved issue which was the cause of revert?

mrousavy, 2023-12-13:
The issue was just the complexity of the codebase because I had two codepaths, one Skia and one native.

It's somewhere in the V3 PR, you'll find it :)

adrian-bobev, 2023-12-14:
Thanks for the answers. WIth that I think we can close the issue since.

closed 2023-12-14 by adrian-bobev

mrousavy, 2024-04-12:
See https://github.com/mrousavy/react-native-vision-camera/pull/2727 👀

adrian-bobev, 2024-04-20:
Sounds promising!