margelo/react-native-vision-camera #2276: 🐛Android: Only the original thread that created a view hierarchy can touch its views

Robogram, 2023-12-11:
### What's happening?

I loaded the app but camera shows black screen with error message,

Camera.onError(unknown/unknown): [unknown/unknown] Only the original thread that created a view hierarchy can touch its views. unknown/unknown: [unknown/uknown] Only the original thread that created a view hierarchy can touch its views

### Reproduceable Code

```tsx
const frontDevice = useCameraDevice('front')
  const backDevice = useCameraDevice('back')

frontDevice != null && backDevice != null ? 
            <Camera
              style={full ? { height: '100%', width: '100%' } : { height: screenWidth, width: screenWidth }}
              ref={camera}
              device={cameraType == 'front' ? frontDevice : backDevice}
              isActive photo video audio
            />
            :
            <NoCameraDeviceError/>
```

### Relevant log output

```shell
I am using "expo run:android --device" not Android Studio
```

### Camera Device

```json
{
  "sensorOrientation": "landscape-left",
  "hardwareLevel": "limited",
  "maxZoom": 8,
  "minZoom": 1,
  "maxExposure": 200,
  "supportsLowLightBoost": true,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": false,
  "isMultiCam": false,
  "minExposure": -200,
  "name": "FRONT (1)",
  "hasFlash": false,
  "hasTorch": false,
  "position": "front",
  "id": "1"
}
```

### Device

Samsung Galaxy S20+

### VisionCamera Version

3.6.14

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [X] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

francesco-clementi-92, 2023-12-11:
Duplicated #2184

mrousavy, 2023-12-12:
Hey! I just created this PR to fix this issue: https://github.com/mrousavy/react-native-vision-camera/pull/2278 :)

closed 2023-12-12 by mrousavy

Robogram, 2023-12-12:
> Hey! I just created this PR to fix this issue: #2278 :)

Ok thank you very much. I'll check it out:)