margelo/react-native-vision-camera #2280: 🐛 [iOS] enableZoomGesture is broken after 3.6.13

adamandreasson, 2023-12-12:
### What's happening?

When using the native pinch to zoom with `enableZoomGesture={true}`, the pinching gesture results in zooming to something like 180x (max zoom) when moving just millimeters. It's virtually impossible to use the function.

I encountered this when updating to the latest version and looked through the recent minor version releases and [found this change in 3.6.13](https://github.com/mrousavy/react-native-vision-camera/commit/a5e10a8606d47694a2c36daf4b27af1bb809acad#diff-be8dd21530846a096dbc25e12308e33953ac39c593ae8687c6a0df6358b22935) that I suspect is causing the problem through some kind of feedback loop between React and Native

### Reproduceable Code

```tsx
function App() {
  const device = useCameraDevice('back')

  if (device == null) return <View />
  return (
    <Camera
      style={StyleSheet.absoluteFill}
      device={device}
      isActive={true}
      enableZoomGesture={true}
    />
  )
}
```

### Relevant log output

```shell
NA
```

### Camera Device

```json
{"formats":[{"minFps":60,"videoWidth":192,"minISO":33,"photoHeight":3024,"maxFps":60,"fieldOfView":65.35665130615234,"videoHeight":144,"autoFocusSystem":"phase-detection","supportsVideoHdr":false,"supportsPhotoHdr":false,"photoWidth":4032,"supportsDepthCapture":false,"maxISO":3168,"pixelFormats":["yuv","yuv","rgb","unknown","unknown","unknown"],"maxZoom":189,"videoStabilizationModes":["auto","off"]},{"minFps":60,"videoWidth":192,"minISO":33,"photoHeight":3024,"maxFps":60,"fieldOfView":65.35665130615234,"videoHeight":144,"autoFocusSystem":"phase-detection","supportsVideoHdr":false,"supportsPhotoHdr":false,"photoWidth":4032,"supportsDepthCapture":false,"maxISO":3168,"pixelFormats":["yuv","yuv","rgb","unknown","unknown","unknown"],"maxZoom":189,"videoStabilizationModes":["auto","off"]},{"minFps":60,"videoWidth":352,"minISO":33,"photoHeight":3024,"maxFps":60,"fieldOfView":60.91105270385742,"videoHeight":288,"autoFocusSystem":"phase-detection","supportsVideoHdr":false,"supportsPhotoHdr":false,"photoWidth":3696,"supportsDepthCapture":false,"maxISO":3168,"pixelFormats":["yuv","yuv","rgb","unknown","unknown","unknown"],"maxZoom":189,"videoStabilizationModes":["auto","off"]},{"minFps":60,"videoWidth":352,"minISO":33,"photoHeight":3024,"maxFps":60,"fieldOfView":60.91105270385742,"videoHeight":288,"autoFocusSystem":"phase-detection","supportsVideoHdr":false,"supportsPhotoHdr":false,"photoWidth":3696,"supportsDepthCapture":false,"maxISO":3168,"pixelFormats":["yuv","yuv","rgb","unknown","unknown","unknown"],"maxZoom":189,"videoStabilizationModes":["auto","off"]},{"minFps":60,"videoWidth":480,"minISO":33,"photoHeight":3024,"maxFps":60,"fieldOfView":65.35665130615234,"videoHeight":360,"autoFocusSystem":"phase-detection","supportsVideoHdr":false,"supportsPhotoHdr":false,"photoWidth":4032,"supportsDepthCapture":false,"maxISO":3168,"pixelFormats":["yuv","yuv","rgb","unknown","unknown","unknown"],"maxZoom":189,"videoStabilizationModes":["auto","off"]},{"minFps":60,"videoWidth":480,"minISO":33,"photoHeight":3024,"maxFps":60,"fieldOfView":65.35665130615234,"videoHeight":360,"autoFocusSystem":"phase-detection","supportsVideoHdr":false,"supportsPhotoHdr":false,"photoWidth":4032,"supportsDepthCapture":false,"maxISO":3168,"pixelFormats":["yuv","yuv","rgb","unknown","unknown","unknown"],"maxZoom":189,"videoStabilizationModes":["auto","off"]},{"minFps":60,"videoWidth":640,"minISO":33,"photoHeight":3024,"maxFps":60,"fieldOfView":65.35665130615234,"videoHeight":480,"autoFocusSystem":"phase-detection","supportsVideoHdr":false,"supportsPhotoHdr":false,"photoWidth":4032,"supportsDepthCapture":false,"maxISO":3168,"pixelFormats":["yuv","yuv","rgb","unknown","unknown","unknown"],"maxZoom":189,"videoStabilizationModes":["auto","off"]},{"minFps":60,"videoWidth":640,"minISO":33,"photoHeight":3024,"maxFps":60,"fieldOfView":65.35665130615234,"videoHeight":480,"autoFocusSystem":"phase-detection","supportsVideoHdr":false,"supportsPhotoHdr":false,"photoWidth":4032,"supportsDepthCapture":false,"maxISO":3168,"pixelFormats":["yuv","yuv","rgb","unknown","unknown","unknown"],"maxZoom":189,"videoStabilizationModes":["auto","off"]},{"minFps":60,"videoWidth":640,"minISO":33,"photoHeight":1512,"maxFps":60,"fieldOfView":65.35665130615234,"videoHeight":480,"autoFocusSystem":"phase-detection","supportsVideoHdr":false,"supportsPhotoHdr":false,"photoWidth":2016,"supportsDepthCapture":false,"maxISO":3168,"pixelFormats":["yuv","yuv","rgb","unknown","unknown","unknown"],"maxZoom":94.5,"videoStabilizationModes":["auto","off"]},{"minFps":60,"videoWidth":640,"minISO":33,"photoHeight":1512,"maxFps":60,"fieldOfView":65.35665130615234,"videoHeight":480,"autoFocusSystem":"phase-detection","supportsVideoHdr":false,"supportsPhotoHdr":false,"photoWidth":2016,"supportsDepthCapture":false,"maxISO":3168,"pixelFormats":["yuv","yuv","rgb","unknown","unknown","unknown"],"maxZoom":94.5,"videoStabilizationModes":["auto","off"]},{"minFps":60,"videoWidth":960,"minISO":33,"photoHeight":2384,"maxFps":60,"fieldOfView":70.29109191894531,"videoHeight":540,"autoFocusSystem":"phase-detection","supportsVideoHdr":false,"supportsPhotoHdr":false,"photoWidth":4224,"supportsDepthCapture":false,"maxISO":3168,"pixelFormats":["yuv","yuv","rgb","unknown","unknown","unknown"],"maxZoom":124.125,"videoStabilizationModes":["auto","cinematic","off","standard","cinematic-extended"]},{"minFps":60,"videoWidth":960,"minISO":33,"photoHeight":2384,"maxFps":60,"fieldOfView":70.29109191894531,"videoHeight":540,"autoFocusSystem":"phase-detection","supportsVideoHdr":false,"supportsPhotoHdr":false,"photoWidth":4224,"supportsDepthCapture":false,"maxISO":3168,"pixelFormats":["yuv","yuv","rgb","unknown","unknown","unknown"],"maxZoom":124.125,"videoStabilizationModes":["auto","cinematic","off","standard","cinematic-extended"]},{"minFps":60,"videoWidth":1024,"minISO":33,"photoHeight":3024,"maxFps":60,"fieldOfView":65.35665130615234,"videoHeight":768,"autoFocusSystem":"phase-detection","supportsVideoHdr":false,"supportsPhotoHdr":false,"photoWidth":4032,"supportsDepthCapture":false,"maxISO":3168,"pixelFormats":["yuv","yuv","rgb","unknown","unknown","unknown"],"maxZoom":171.875,"videoStabilizationModes":["auto","off","standard"]},{"minFps":60,"videoWidth":1024,"minISO":33,"photoHeight":3024,"maxFps":60,"fieldOfView":65.35665130615234,"videoHeight":768,"autoFocusSystem":"phase-detection","supportsVideoHdr":false,"supportsPhotoHdr":false,"photoWidth":4032,"supportsDepthCapture":false,"maxISO":3168,"pixelFormats":["yuv","yuv","rgb","unknown","unknown","unknown"],"maxZoom":171.875,"videoStabilizationModes":["auto","off","standard"]},{"minFps":30,"videoWidth":1280,"minISO":33,"photoHeight":2376,"maxFps":30,"fieldOfView":70.29109191894531,"videoHeight":720,"autoFocusSystem":"phase-detection","supportsVideoHdr":false,"supportsPhotoHdr":false,"photoWidth":4224,"supportsDepthCapture":false,"maxISO":3168,"pixelFormats":["yuv","yuv","rgb","unknown","unknown","unknown"],"maxZoom":123.75,"videoStabilizationModes":["auto","cinematic","off","standard","cinematic-extended"]},{"minFps":30,"videoWidth":1280,"minISO":33,"photoHeight":2376,"maxFps":30,"fieldOfView":70.29109191894531,"videoHeight":720,"autoFocusSystem":"phase-detection","supportsVideoHdr":false,"supportsPhotoHdr":false,"photoWidth":4224,"supportsDepthCapture":false,"maxISO":3168,"pixelFormats":["yuv","yuv","rgb","unknown","unknown","unknown"],"maxZoom":123.75,"videoStabilizationModes":["auto","cinematic","off","standard","cinematic-extended"]},{"minFps":30,"videoWidth":1280,"minISO":33,"photoHeight":2376,"maxFps":30,"fieldOfView":70.29109191894531,"videoHeight":720,"autoFocusSystem":"phase-detection","supportsVideoHdr":true,"supportsPhotoHdr":false,"photoWidth":4224,"supportsDepthCapture":false,"maxISO":3168,"pixelFormats":["yuv","yuv","rgb","unknown","unknown","unknown"],"maxZoom":123.75,"videoStabilizationModes":["auto","cinematic","off","standard","cinematic-extended"]},{"minFps":60,"videoWidth":1280,"minISO":33,"photoHeight":2384,"maxFps":60,"fieldOfView":70.29109191894531,"videoHeight":720,"autoFocusSystem":"phase-detection","supportsVideoHdr":false,"supportsPhotoHdr":false,"photoWidth":4224,"supportsDepthCapture":false,"maxISO":3168,"pixelFormats":["yuv","yuv","rgb","unknown","unknown","unknown"],"maxZoom":124.125,"videoStabilizationModes":["auto","cinematic","off","standard","cinematic-extended"]},{"minFps":60,"videoWidth":1280,"minISO":33,"photoHeight":2384,"maxFps":60,"fieldOfView":70.29109191894531,"videoHeight [... cut ...]
### Additional information

- [X] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

iliapnmrv, 2023-12-13:
I also can reproduce this issue. RNVC v3.6.15. IOS 17.1.2

jkadamczyk, 2023-12-29:
If you want a "quick fix" help yourselves with the library patch below
```
diff --git a/node_modules/react-native-vision-camera/ios/CameraView+Zoom.swift b/node_modules/react-native-vision-camera/ios/CameraView+Zoom.swift
index 74dc647..bc2c590 100644
--- a/node_modules/react-native-vision-camera/ios/CameraView+Zoom.swift
+++ b/node_modules/react-native-vision-camera/ios/CameraView+Zoom.swift
@@ -20,7 +20,7 @@ extension CameraView {
 
     // Update zoom React prop
     zoom = NSNumber(value: scale)
-    didSetProps(["zoom"])
+    didSetProps([])
   }
 
   func addPinchGestureRecognizer() {
```

mrousavy, 2024-01-02:
Thanks @jkadamczyk, fixed!

closed 2024-01-02 by mrousavy