margelo/react-native-vision-camera #2286: 🔧 Cannot build on Android with worklets-core

jslok, 2023-12-13:
### How were you trying to build the app?

yarn start --reset-cache

Getting an error during build when I have react-native-worklets-core installed for frame processors. If I delete and reinstall node_modules, I can build 0-2 times before the error comes back. Using an almost fresh copy of rn 0.73 with only other module being opencv for frame processor.
Tried earlier versions of worklets-core back to 0.2.0. Works fine without rnwc and/or with frame processors disabled.

Any help would be much appreciated.

The example app doesn't appear to be working for other reasons.

### Full build logs

```tsx
info Opening app on Android...
info A dev server is already running for this project on port 8081.
info Installing the app...
> Task :gradle-plugin:compileKotlin UP-TO-DATE
> Task :gradle-plugin:compileJava NO-SOURCE
> Task :gradle-plugin:pluginDescriptors UP-TO-DATE
> Task :gradle-plugin:processResources UP-TO-DATE
> Task :gradle-plugin:classes UP-TO-DATE
> Task :gradle-plugin:jar UP-TO-DATE
> Task :gradle-plugin:inspectClassesForKotlinIC UP-TO-DATE

> Configure project :openCVLib
OpenCV: 4.8.0 C:\Users\jslok\scan\android\openCVLib\build.gradle

> Configure project :react-native-vision-camera
[VisionCamera] react-native-worklets-core found, Frame Processors enabled!

> Task :app:generateCodegenSchemaFromJavaScript SKIPPED
> Task :app:generateCodegenArtifactsFromSchema SKIPPED
> Task :app:generatePackageList
> Task :app:preBuild
> Task :app:preDebugBuild
> Task :app:generateDebugBuildConfig UP-TO-DATE
> Task :openCVLib:preBuild UP-TO-DATE
> Task :openCVLib:preDebugBuild UP-TO-DATE
> Task :openCVLib:writeDebugAarMetadata UP-TO-DATE
> Task :react-native-vision-camera:prepareHeaders UP-TO-DATE
> Task :react-native-vision-camera:preBuild UP-TO-DATE
> Task :react-native-vision-camera:preDebugBuild UP-TO-DATE
> Task :react-native-vision-camera:writeDebugAarMetadata UP-TO-DATE
> Task :react-native-worklets-core:prepareHeaders UP-TO-DATE
> Task :react-native-worklets-core:preBuild UP-TO-DATE
> Task :react-native-worklets-core:preDebugBuild UP-TO-DATE
> Task :react-native-worklets-core:writeDebugAarMetadata UP-TO-DATE
> Task :app:checkDebugAarMetadata UP-TO-DATE
> Task :app:generateDebugResValues UP-TO-DATE
> Task :openCVLib:generateDebugResValues UP-TO-DATE
> Task :openCVLib:generateDebugResources UP-TO-DATE
> Task :openCVLib:packageDebugResources UP-TO-DATE
> Task :react-native-vision-camera:generateDebugResValues UP-TO-DATE
> Task :react-native-vision-camera:generateDebugResources UP-TO-DATE
> Task :react-native-vision-camera:packageDebugResources UP-TO-DATE
> Task :react-native-worklets-core:generateDebugResValues UP-TO-DATE
> Task :react-native-worklets-core:generateDebugResources UP-TO-DATE
> Task :react-native-worklets-core:packageDebugResources UP-TO-DATE
> Task :app:mapDebugSourceSetPaths UP-TO-DATE
> Task :app:generateDebugResources UP-TO-DATE
> Task :app:mergeDebugResources UP-TO-DATE
> Task :app:packageDebugResources UP-TO-DATE
> Task :app:parseDebugLocalResources UP-TO-DATE
> Task :app:createDebugCompatibleScreenManifests UP-TO-DATE
> Task :app:extractDeepLinksDebug UP-TO-DATE
> Task :openCVLib:extractDeepLinksDebug UP-TO-DATE
> Task :openCVLib:processDebugManifest UP-TO-DATE
> Task :react-native-vision-camera:extractDeepLinksDebug UP-TO-DATE
> Task :react-native-vision-camera:processDebugManifest UP-TO-DATE
[... 275 lines cut ...]
```

### VisionCamera Version

3.6.16

### Target platforms

Android

### Operating system

Windows

### Can you build the VisionCamera Example app?

No, I cannot build the Example app either

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

rodgomesc, 2023-12-13:
this also happens to me quite often, it builds after deleting `android/.gradle` folder

c-info, 2023-12-14:
having same issue when running the command cd android && ./gradlew clean && cd - && npx react-native run-android

gyanbhartip, 2024-01-03:
Has anyone been able to find a fix/reason/workaround for this error?

gyanbhartip, 2024-01-03:
deleting android/.gradle folder before building helped fix this for me.

gzqyl, 2024-01-04:
> deleting android/.gradle folder before building helped fix this for me.

this should be the correct answer, just test it, when build error occurs, clean project does not work, we need to delete the whole cache folder

dipudk29, 2024-01-04:
> deleting android/.gradle folder before building helped fix this for me.

Life savior.

divitrao, 2024-01-09:
getting this error here 

C/C++: CMake Generate step failed.  Build files cannot be regenerated correctly.
[CXX1429] error when building with cmake using C:\Users\divit\Nutrify\nutodaymobileapp\node_modules\react-native-vision-camera\android\CMakeLists.txt: -- Android: Targeting API '26' with architecture 'x86', ABI 'x86', and processor 'i686'
-- Android: Selected unified Clang toolchain
-- The C compiler identification is Clang 12.0.8
-- The CXX compiler identification is Clang 12.0.8
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Users/divit/AppData/Local/Android/Sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/windows-x86_64/bin/clang.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Users/divit/AppData/Local/Android/Sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/windows-x86_64/bin/clang++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done

C++ build system [configure] failed while executing:
    @echo off
    "C:\\Users\\divit\\AppData\\Local\\Android\\Sdk\\cmake\\3.22.1\\bin\\cmake.exe" ^
      "-HC:\\Users\\divit\\Nutrify\\nutodaymobileapp\\node_modules\\react-native-vision-camera\\android" ^
      "-DCMAKE_SYSTEM_NAME=Android" ^
      "-DCMAKE_EXPORT_COMPILE_COMMANDS=ON" ^
      "-DCMAKE_SYSTEM_VERSION=26" ^
      "-DANDROID_PLATFORM=android-26" ^
      "-DANDROID_ABI=x86" ^
      "-DCMAKE_ANDROID_ARCH_ABI=x86" ^
      "-DANDROID_NDK=C:\\Users\\divit\\AppData\\Local\\Android\\Sdk\\ndk\\23.1.7779620" ^
      "-DCMAKE_ANDROID_NDK=C:\\Users\\divit\\AppData\\Local\\Android\\Sdk\\ndk\\23.1.7779620" ^
      "-DCMAKE_TOOLCHAIN_FILE=C:\\Users\\divit\\AppData\\Local\\Android\\Sdk\\ndk\\23.1.7779620\\build\\cmake\\android.toolchain.cmake" ^
      "-DCMAKE_MAKE_PROGRAM=C:\\Users\\divit\\AppData\\Local\\Android\\Sdk\\cmake\\3.22.1\\bin\\ninja.exe" ^
      "-DCMAKE_CXX_FLAGS=-O2 -frtti -fexceptions -Wall -Wno-unused-variable -fstack-protector-all" ^
      "-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=C:\\Users\\divit\\Nutrify\\nutodaymobileapp\\node_modules\\react-native-vision-camera\\android\\build\\intermediates\\cxx\\Debug\\142u2c03\\obj\\x86" ^
      "-DCMAKE_RUNTIME_OUTPUT_DIRECTORY=C:\\Users\\divit\\Nutrify\\nutodaymobileapp\\node_modules\\react-native-vision-camera\\android\\build\\intermediates\\cxx\\Debug\\142u2c03\\obj\\x86" ^
      "-DCMAKE_BUILD_TYPE=Debug" ^
      "-DCMAKE_FIND_ROOT_PATH=C:\\Users\\divit\\Nutrify\\nutodaymobileapp\\node_modules\\react-native-vision-camera\\android\\.cxx\\Debug\\142u2c03\\prefab\\x86\\prefab" ^
      "-BC:\\Users\\divit\\Nutrify\\nutodaymobileapp\\node_modules\\react-native-vision-camera\\android\\.cxx\\Debug\\142u2c03\\x86" ^
      -GNinja ^
      "-DANDROID_STL=c++_shared" ^
      "-DNODE_MODULES_DIR=C:\\Users\\divit\\Nutrify\\nutodaymobileapp\\node_modules" ^
      "-DENABLE_FRAME_PROCESSORS=true"
  from C:\Users\divit\Nutrify\nutodaymobileapp\node_modules\react-native-vision-camera\android
VisionCamera: Frame Processors enabled!
CMake Error in CMakeLists.txt:
  Imported target "react-native-worklets-core::rnworklets" includes
  non-existent path

    "C:/Users/divit/Nutrify/nutodaymobileapp/node_modules/react-native-worklets-core/android/build/headers/rnworklets"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.

CMake Generate step failed.  Build files cannot be regenerated correctly.

i have "react-native-worklets-core": "^0.2.4", and  "react-native-vision-camera": "3.6.4",

closed 2024-01-15 by mrousavy

Heni-ghodbane, 2024-02-28:
We are still facing this error while building using Android Studio. Any suggestions? thanks 
These are the versions

"react-native-vision-camera": "^3.6.12",    
"react-native-worklets-core": "^0.3.0",

wuguishifu, 2024-04-03:
I'm still getting this issue after the following attempts to fix it:
1) deleting the android/.gradle folder
2) clearing all cache in android studio and restarting
3) working in a fresh isolated (linux) container
4) working on a fresh computer (mac os)

tomerh2001, 2024-05-27:
Also facing it now

```
    "react-native-vision-camera": "4.0.5",
    "react-native-worklets-core": "^1.3.3",
    "vision-camera-resize-plugin": "^3.1.0",
    "@ismaelmoreiraa/vision-camera-ocr": "^3.0.1",
```

This is the error:
```
> Task :prepareKotlinBuildScriptModel UP-TO-DATE
C/C++: VisionCamera: Frame Processors: ON!
C/C++: VisionCamera: Linking react-native-worklets...
C/C++: CMake Error in CMakeLists.txt:
C/C++:   Imported target "react-native-worklets-core::rnworklets" includes
C/C++:   non-existent path
C/C++:     "C:/projects/RC/x-workspace/projects/xmobile/node_modules/react-native-worklets-core/android/build/headers/rnworklets"
C/C++:   in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:
C/C++:   * The path was deleted, renamed, or moved to another location.
C/C++:   * An install or uninstall procedure did not complete successfully.
C/C++:   * The installation package was faulty and references files it does not
C/C++:   provide.
C/C++: CMake Generate step failed.  Build files cannot be regenerated correctly.
```

hungtrn75, 2024-07-02:
> Also facing it now
>
> ```
>     "react-native-vision-camera": "4.0.5",
>     "react-native-worklets-core": "^1.3.3",
>     "vision-camera-resize-plugin": "^3.1.0",
>     "@ismaelmoreiraa/vision-camera-ocr": [...]

Same issue

bonnmh, 2024-07-18:
> Also facing it now
>
> ```
>     "react-native-vision-camera": "4.0.5",
>     "react-native-worklets-core": "^1.3.3",
>     "vision-camera-resize-plugin": "^3.1.0",
>     "@ismaelmoreiraa/vision-camera-ocr": [...]

I encountered the same error, have you resolved the above error?

PrinceUBS, 2024-07-31:
I'm facing this issue! how you guys able to fix?

nppull, 2024-08-30:
> Android Studio Koala | 2024.1.1 Patch 2
> macOS 14.4
> react native 0.74.5
> react-native-vision-camera: 4.3.2
> react-native-worklets-core: 1.3.3

same error

thaindangelhack, 2024-08-30:
```
react-native: 0.74.5
react-native-reanimated: 3.10.1
react-native-vision-camera: 3.9.2
react-native-worklets-core: 1.3.3
```

I got the same error
I tried deleting the file many times like most suggested, but it didn't work for me
I also tried upgrading the libraries (react-native-reanimated, react-native-vision-camera, react-native-worklets-core) to different versions (including the latest) but still got the same error

I am waiting for a more accurate solution, can anyone help me?

nppull, 2024-11-01:
> ```
> react-native: 0.74.5
> react-native-reanimated: 3.10.1
> react-native-vision-camera: 3.9.2
> react-native-worklets-core: 1.3.3
> ```
>
> I got the same error I tried deleting the file many times like most [...]

react-native-vision-camera: 3.9.2
work with react-native-worklets-core: 0.3.0

thaindangelhack, 2024-11-01:
> react-native-vision-camera: 3.9.2 work with react-native-worklets-core: 0.3.0

thanks, maybe i will try later

i noticed that react-native-worklets-core was recently added (from react upgrade recommendation), the previous build didn't have it.

i'm not sure what the purpose of adding this lib is, but i removed it, everything still works

afiq234, 2024-11-04:
{
  "name": "TestDetection",
  "version": "0.0.1",
  "private": true,
  "scripts": {
    "android": "react-native run-android",
    "ios": "react-native run-ios",
    "lint": "eslint .",
    "start": "react-native start",
    "test": "jest"
  },
  "dependencies": {
    "react": "18.2.0",
    "react-native": "0.74.5",
    "react-native-reanimated": "^3.15.4",
    "react-native-vision-camera": "^4.5.2",
    "react-native-worklets-core": "^1.3.3"
  },
  "devDependencies": {
    "@babel/core": "^7.20.0",
    "@babel/preset-env": "^7.20.0",
    "@babel/runtime": "^7.20.0",
    "@react-native/babel-preset": "0.74.87",
    "@react-native/eslint-config": "0.74.87",
    "@react-native/metro-config": "0.74.87",
    "@react-native/typescript-config": "0.74.87",
    "@types/react": "^18.2.6",
    "@types/react-test-renderer": "^18.0.0",
    "babel-jest": "^29.6.3",
    "eslint": "^8.19.0",
    "jest": "^29.6.3",
    "prettier": "2.8.8",
    "react-test-renderer": "18.2.0",
    "typescript": "5.0.4"
  },
  "engines": {
    "node": ">=18"
  }
}

this one works for me

QSuraj, 2025-01-01:
Any update on this? Still facing the issue. 
```
CMake Error in CMakeLists.txt:
Imported target "react-native-worklets-core::rnworklets" includes
non-existent path
```

Packages -
```
"react-native-vision-camera": "4.6.3",
"react-native-worklets-core": "1.5.0",
"react-native-reanimated": "3.16.6",
```

luizricardo-co, 2025-01-21:
I'm facing same error

``
Imported target "react-native-worklets-core::rnworklets" includes

    "react-native": "0.76.6",
    "react-native-reanimated": "^3.16.7",
    "react-native-vision-camera": "^4.6.3",

    "react-native-worklets-core": "^1.5.0"

``

vorarohan, 2025-02-11:
> I'm facing this issue! how you guys able to fix?

I'm also facing same issue!

CMake Error in CMakeLists.txt:
Imported target "react-native-worklets-core::rnworklets" includes
non-existent path

alexgomes09, 2025-02-14:
Same error
```
CMake Error in CMakeLists.txt:
Imported target "react-native-worklets-core::rnworklets" includes
non-existent path
```
Tired of so many build errors. It's never-ending. It's been almost a week now I am trying to solve so many build errors :(

Josh2941, 2025-04-10:
Same issue, stuck for a long time. 

```
"react": "18.2.0",
"react-native": "0.73.6",
"react-native-vision-camera": "4.3.2",
"react-native-worklets-core": "1.3.3",
```

tgreco, 2025-04-14:
Same issue.

alameen-47, 2025-04-29:
SAME ISSUE ANY UPDATE HOW TO FIX IT 
A problem occurred configuring project ':react-native-worklets-core'.

  "react-native-vision-camera": "^4.6.4",
    "react-native-worklets-core": "^1.5.0",
    "react-native": "0.78.2",
    "react-native-reanimated": "^3.17.2",

dremex, 2025-04-29:
FWIW, I was bumping into all sorts of weirdness when trying to build Android, this solved it:

```
rm -rf ~/.gradle/caches
rm -rf ~/.gradle/daemon
```

alameen-47, 2025-04-29:
@ dremex hey could you help with passing data to an tf lite model from react native vision camera package 

I MESSED up with all issues and now struck

alameen-47, 2025-04-29:
`import {loadTensorflowModel} from 'react-native-fast-tflite';
import {Camera, useCameraDevices} from 'react-native-vision-camera';

`

 `  useEffect(() => {
    const loadModel = async () => {
      try {
        const loadedModel = await loadTensorflowModel(
          require('../../assets/models/movenet_singlePose.tflite'),
        );
        setModel(loadedModel);
        console.log('✅--MODEL LOADED--✅');
      } catch (error) {
        console.error('Error loading TensorFlow Lite model:', error);
      }
    };
    loadModel();
  }, []);

`
      <Camera
        className="flex-1 absolute inset-0"
        style={{
          flex: 1,
        }}
        ref={cameraRef}
        device={device}
        isActive={true}
        resizeMode="cover"
        photo={true}
        video={true}
        audio={false}
        enableZoomGesture={true}
      />

`

    "react-native-vision-camera": "^4.6.4",
    "react-native": "0.78.2",
    "react-native-fast-tflite": "^1.6.1",

`