margelo/react-native-vision-camera #2288: 🐛

deepakv-z14, 2023-12-14:
### What's happening?

application is crashing randomly in production with below error

```
Fatal Exception: java.lang.IllegalStateException: maxImages (3) has already been acquired, call #close before acquiring more.
       at android.media.ImageReader.acquireNextImage(ImageReader.java:527)
       at android.media.ImageReader.acquireLatestImage(ImageReader.java:411)
       at com.mrousavy.camera.core.CameraSession.configureOutputs$lambda$1(CameraSession.kt:299)
       at android.media.ImageReader$ListenerHandler.handleMessage(ImageReader.java:819)
       at android.os.Handler.dispatchMessage(Handler.java:106)
       at android.os.Looper.loop(Looper.java:264)
       at android.os.HandlerThread.run(HandlerThread.java:67)
```

### Reproduceable Code

```tsx
const device = useCameraDevice('back');

  const isFocused = useIsFocused();
 
<Camera
          style={StyleSheet.absoluteFill}
          device={device}
          isActive={isFocused}
          ref={scanRef}
          photo={true}
          orientation="portrait"
          onError={onCameraError}
        />
```

### Relevant log output

```shell
not available
```

### Camera Device

```json
not available
```

### Device

OPPO CPH2239

### VisionCamera Version

3.6.15

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-01-02:
No output logs available, no camera device available, and no reproduceable sample available - sorry but I cannot help you.

closed 2024-01-02 by mrousavy

deepakv-z14, 2024-01-02:
@mrousavy https://github.com/mrousavy/react-native-vision-camera/issues/1387 this seems to be same issue. And here we also have logcat output. Does this help?