margelo/react-native-vision-camera #2290: Cannot build on Android 🔧

c-info, 2023-12-14:
### How were you trying to build the app?

Cannot build on android getting error 
> Task :react-native-vision-camera:buildCMakeDebug[arm64-v8a] FAILED
C/C++: ld: error: undefined symbol: RNWorklet::JsiWorkletContext::getWorkletRuntime()

Already tried on given link 

https://github.com/mrousavy/react-native-vision-camera/issues/1708#issuecomment-1690448880

### Full build logs

```tsx
> Configure project :react-native-reanimated
Android gradle plugin: 8.1.1
Gradle: 8.3

> Configure project :react-native-vision-camera
[VisionCamera] react-native-worklets-core found, Frame Processors enabled!

> Task :react-native-reanimated:processDebugManifest
package="com.swmansion.reanimated" found in source AndroidManifest.xml: /Users/Samples/camera/node_modules/react-native-reanimated/android/src/main/AndroidManifest.xml.
Setting the namespace via the package attribute in the source AndroidManifest.xml is no longer supported, and the value is ignored.
Recommendation: remove package="com.swmansion.reanimated" from the source AndroidManifest.xml: /Users/Samples/camera/node_modules/react-native-reanimated/android/src/main/AndroidManifest.xml.

> Task :react-native-worklets-core:processDebugManifest
package="com.worklets" found in source AndroidManifest.xml: /Users/Samples/camera/node_modules/react-native-worklets-core/android/src/main/AndroidManifest.xml.
Setting the namespace via the package attribute in the source AndroidManifest.xml is no longer supported, and the value is ignored.
Recommendation: remove package="com.worklets" from the source AndroidManifest.xml: /Users/Samples/camera/node_modules/react-native-worklets-core/android/src/main/AndroidManifest.xml.

> Task :react-native-reanimated:compileDebugJavaWithJavac

> Task :react-native-worklets-core:compileDebugJavaWithJavac

> Task :react-native-vision-camera:compileDebugKotlin
w: file:///Users/Samples/camera/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraDevicesManager.kt:86:19 Parameter 'eventName' is never used
w: file:///Users/Samples/camera/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraDevicesManager.kt:89:23 Parameter 'count' is never used
w: file:///Users/Samples/camera/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt:7:44 'RCTEventEmitter' is deprecated. Deprecated in Java
w: file:///Users/Samples/camera/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt:19:28 'RCTEventEmitter' is deprecated. Deprecated in Java
w: file:///Users/Samples/camera/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt:19:57 'receiveEvent(Int, String!, WritableMap?): Unit' is deprecated. Deprecated in Java
w: file:///Users/Samples/camera/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt:26:28 'RCTEventEmitter' is deprecated. Deprecated in Java
w: file:///Users/Samples/camera/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt:26:57 'receiveEvent(Int, String!, WritableMap?): Unit' is deprecated. Deprecated in Java
w: file:///Users/Samples/camera/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt:33:28 'RCTEventEmitter' is deprecated. Deprecated in Java
w: file:///Users/Samples/camera/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt:33:57 'receiveEvent(Int, String!, WritableMap?): Unit' is deprecated. Deprecated in Java
w: file:///Users/Samples/camera/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt:51:28 'RCTEventEmitter' is deprecated. Deprecated in Java
w: file:///Users/Samples/camera/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt:51:57 'receiveEvent(Int, String!, WritableMap?): Unit' is deprecated. Deprecated in Java
w: file:///Users/Samples/camera/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt:57:28 'RCTEventEmitter' is deprecated. Deprecated in Java
w: file:///Users/Samples/camera/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt:57:57 'receiveEvent(Int, String!, WritableMap?): Unit' is deprecated. Deprecated in Java
w: file:///Users/Samples/camera/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt:97:28 'RCTEventEmitter' is deprecated. Deprecated in Java
w: file:///Users/Samples/camera/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt:97:57 'receiveEvent(Int, String!, WritableMap?): Unit' is deprecated. Deprecated in Java
w: file:///Users/Samples/camera/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraViewModule.kt:8:46 'ReactModule' is deprecated. Deprecated in Java
w: file:///Users/Samples/camera/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraViewModule.kt:23:2 'ReactModule' is deprecated. Deprecated in Java
w: file:///Users/Samples/camera/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/RecordingSession.kt:48:96 'constructor MediaRecorder()' is deprecated. Deprecated in Java
w: file:///Users/Samples/camera/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/extensions/CameraCharacteristics+getOutputSizes.kt:22:38 'get(Int, Int): CamcorderProfile!' is deprecated. Deprecated in Java
w: file:///Users/Samples/camera/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/extensions/CameraDevice+createCaptureSession.kt:60:12 'createCaptureSessionByOutputConfigurations((Mutable)List<OutputConfiguration!>!, CameraCaptureSession.StateCallback!, Handler?): Unit' is deprecated. Deprecated in Java
w: file:///Users/Samples/camera/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/extensions/RecordingSession+getRecommendedBitRate.kt:52:24 'get(Int, Int): CamcorderProfile!' is deprecated. Deprecated in Java
w: file:///Users/Samples/camera/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/extensions/RecordingSession+getRecommendedBitRate.kt:54:24 'get(Int): CamcorderProfile!' is deprecated. Deprecated in Java

> Task :react-native-vision-camera:configureCMakeDebug[arm64-v8a]
C/C++: VisionCamera: Frame Processors: ON!
C/C++: VisionCamera: Linking react-native-worklets...

> Task :react-native-worklets-core:buildCMakeDebug[armeabi-v7a]
C/C++: ninja: Entering directory `/Users/Samples/camera/node_modules/react-native-worklets-core/android/.cxx/Debug/5s1f185u/armeabi-v7a'
C/C++: In file included from /Users/Samples/camera/node_modules/react-native-worklets-core/cpp/WKTJsiWorkletContext.cpp:14:
C/C++: /Users/Samples/camera/node_modules/react-native-worklets-core/android/../cpp/decorators/WKTJsiSetImmediateDecorator.h:85:53: warning: format specifies type 'unsigned long' but the argument has type 'size_t' (aka 'unsigned int') [-Wformat]
C/C++:                   printf("ctx %lu: setImmediate\n", context->getContextId());
C/C++:                               ~~~                   ^~~~~~~~~~~~~~~~~~~~~~~
C/C++:                               %zu
C/C++: 1 warning generated.
[... 237 lines cut ...]
```

### VisionCamera Version

3.6.16

### Target platforms

Android

### Operating system

MacOS

### Can you build the VisionCamera Example app?

No, I cannot build the Example app either

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

hypesol, 2023-12-14:
I am also facing these crashes. even the example files are also not working

romancxx, 2023-12-14:
I had the same issue, and following the troubleshooting steps from the doc fixed the problem. Especially : 
```
./android/gradlew clean
rm -rf android/.gradle android/.idea android/app/build android/build
rm -rf package-lock.json yarn.lock node_modules
yarn   # or `npm i`
```

spsaucier, 2023-12-15:
That made it buildable but I still get the following when the JS is building:

```
Error: While trying to resolve module `@mgcrea/vision-camera-code-scanner` from file `/Users/stephensaucier/other/vision-camera-barcode-scanner/example/src/App.tsx`, the package `/Users/stephensaucier/other/vision-camera-barcode-scanner/example/node_modules/@mgcrea/vision-camera-code-scanner/package.json` was successfully found. However, this package itself specifies a `main` module field that could not be resolved (`/Users/stephensaucier/other/vision-camera-barcode-scanner/example/node_modules/@mgcrea/vision-camera-code-scanner/dist/index.js`.
```

alexdonh, 2023-12-27:
Having the same issue. clearing build folders, node_modules and cache did not do the trick for me. Any other workaround?

mrousavy, 2024-01-19:
Hey - yep this is a caching issue.

closed 2024-01-19 by mrousavy

Ahebiemarc, 2024-03-22:
This warning message is generated by the C/C++ compiler when compiling the code in the WKTJsiSetImmediateDecorator.h file of the react-native-worklets-core module.

The warning indicates that the format specified in the printf function does not match the type of the argument provided. Specifically, the %lu format is used to print a value of type unsigned long, but the argument provided is of type size_t, which is usually defined as an alias for unsigned int.

To correct this problem, you can simply change the format in the printf function to match the type of the argument. In this case you can use %zu for size_t:

node_modules/react-native-worklets-core/cpp/decorators/WKTJsiSetImmediateDecorator.h

printf("ctx %zu: setImmediate\n", context->getContextId());

mrousavy, 2024-03-22:
Thanks ChatGPT 👍

mlhtnc, 2024-04-17:
> I had the same issue, and following the troubleshooting steps from the doc fixed the problem. Especially :
> 
> ```
> ./android/gradlew clean
> rm -rf android/.gradle android/.idea android/app/build android/build
> rm -rf package-lock.json yarn.lock node_modules
> yarn   # or `npm i`
> ```

I have faced the same issue when I try to clone project and deploy again. When pushing to remote some files of course ignored and when you clone back, it creates cache issues afaiu.

gradlew clean did not work for me, it kept saying clean task not found, I don't know why. I skipped that.

I just did:

> rm -rf android (if you have changes, you will lose them!)
> rm -rf node_modules package-lock.json
> npm i