margelo/react-native-vision-camera #2291: ❓ Getting a Black screen when useing camera

FazilMuhammed, 2023-12-15:
### Question

  {isFocused && (
                      <Camera
                        style={StyleSheet.absoluteFill}
                        // style={{width: 500, height: 500}}
                        device={device}
                        codeScanner={codeScanner}
                        isActive={isActive}
                        lowLightBoost={true}
                        key={`${resetKey}`}
                        onInitialized={onInitialized}
                        onError={error => {
                          if (
                            error.message.includes(
                              'CameraDevice was already closed',
                            )
                          ) {
                            setResetKey(prev => prev + 1);
                          }
                        }}
                      />
                    )}

### What I tried

I need scan QR code I done these steps 

./android/gradlew clean
rm -rf android/.gradle android/.idea android/app/build android/build
rm -rf package-lock.json yarn.lock node_modules
yarn   # or `npm I`

still remains black screen

### VisionCamera Version

"react-native-vision-camera": "^3.6.17",

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

jslok, 2023-12-15:
3.6.17 gives me a black screen.
3.6.16 works.

Gruak, 2023-12-15:
When I wrap the Camera component with SafeAreaView then it works (no black screen). Weird.

AlexherDev69, 2023-12-15:
same here, but then it works when I flip the camera

> When I wrap the Camera component with SafeAreaView then it works (no black screen). Weird.

That worked for me too, thanks

M4Tdev, 2023-12-15:
I have the same issue, and I found out that when I turn on the flash it somehow refreshes/activates the camera itself and the image from it shows up.

android-imdad, 2023-12-15:
@mrousavy I did some digging, it seems the even though we set the Camera as "isActive" the camera module is showing isActive=false in the logs

> Successfully updated CameraSession Configuration! isActive: false

Turning on flash actually refreshes the camera view and removes the black screen to show the preview.

feardarkness, 2023-12-15:
@M4Tdev solution works for now. Just use flash with the same flag as isActive.

```
<Camera
   ...
   isActive={isCameraActive}
   flash={isCameraActive}
   />
```

jaimeneto85, 2023-12-15:
> When I wrap the Camera component with SafeAreaView then it works (no black screen). Weird.

I had the same problema on versions 3.6.17 and 3.6.16 as well.

but when I tried. to wrap with SafeAreaView:
```Typescript
<SafeAreaView style={StyleSheet.absoluteFill}>
          <Camera
            ref={ref}
            video
            format={format}
            device={devices}
            isActive={true}
            style={StyleSheet.absoluteFill}
          />
        </SafeAreaView>
        
```
OBS: without absolutFill the screen turn on White. Weird!

basitmir, 2023-12-16:
Same issue on 3.6.16
Sometimes it works with no black screen but sometimes it shows black screen
@mrousavy

FazilMuhammed, 2023-12-16:
any update from this @mrousavy

android-imdad, 2023-12-16:
> > When I wrap the Camera component with SafeAreaView then it works (no black screen). Weird.
> 
> I had the same problema on versions 3.6.17 and 3.6.16 as well.
> 
> but when I tried. to wrap with SafeAreaView:
> 
> ```ts
> <SafeAreaView style={StyleSheet.absoluteFill}>
>           <Camera
>             ref={ref}
>             video
>             format={format}
>             device={devices}
>             isActive={true}
>             style={StyleSheet.absoluteFill}
>           />
>         </SafeAreaView>
>         
> ```
> 
> OBS: without absolutFill the screen turn on White. Weird!

I am on version 3.5.1 because of the crash issue and this is still a persistent issue.

android-imdad, 2023-12-17:
> I had the same problema on versions 3.6.17 and 3.6.16 as well.
>
> but when I tried. to wrap with SafeAreaView:
>
> ```ts
> <SafeAreaView style={StyleSheet.absoluteFill}>
>           <Camera [...]

It seems this also does not work as expected, sometimes it loads the camera and other times its just a black screen

android-imdad, 2023-12-17:
Found a temporary fix for this issue, You need to update line number 42 in CameraConfiguration.Kt to 
`  var isActive: Boolean = true`
and line number 208 in CameraView.kt to
`config.isActive = true;`

This seems to fix the issue for now, not sure if there will be any unintended consequences but seems to be working fine.

FazilMuhammed, 2023-12-17:
@android-imdad where I find CameraConfiguration.Kt can share path please ?

jaimeneto85, 2023-12-17:
> Found a temporary fix for this issue, You need to update line number 42 in CameraConfiguration.Kt to ` var isActive: Boolean = true` and line number 208 in CameraView.kt to `config.isActive = [...]

the `config.isActive = true;`  for me make no sense, bacause config.isActive check `isActive && isAttachedToWindow`. Changing `config.isActive =  isActive` it's enough. The problem is `isAttachedToWindow` is changing the state. start with true (correctly) but change for false (this is the reason that works wrappign the code witn "SafeAreaView". I did a PR (#2299)  right now to change this line that I said.

android-imdad, 2023-12-18:
> the `config.isActive = true;` for me make no sense, bacause config.isActive check `isActive && isAttachedToWindow`. Changing `config.isActive = isActive` it's enough. The problem is [...]

Even though this fixes the camera preview, barcode or QR code scanning does not work as code scanner is being turned into null for some reason. I added a temporary fix for it by adding the scanning parameters manually in line number 348 in CameraSession.kt

`//    val codeScanner = configuration.codeScanner as? CameraConfiguration.Output.Enabled<CameraConfiguration.CodeScanner>

    var scannerOptions : List<CodeType> = listOf(CodeType.CODE_39,CodeType.CODE_128,CodeType.QR,CodeType.EAN_13);
    // CodeScanner Output
    val codeScanner = CameraConfiguration.Output.Enabled.create(
      CameraConfiguration.CodeScanner(scannerOptions)
    )`

FazilMuhammed, 2023-12-18:
@android-imdad when I replace from

android-imdad, 2023-12-18:
> @android-imdad when I replace from.
> 
> // val codeScanner = configuration.codeScanner as? CameraConfiguration.Output.Enabled<CameraConfiguration.CodeScanner>
> 
> to
> 
> var scannerOptions : List = listOf(CodeType.CODE_39,CodeType.CODE_128,CodeType.QR,CodeType.EAN_13); // CodeScanner Output val codeScanner = CameraConfiguration.Output.Enabled.create( CameraConfiguration.CodeScanner(scannerOptions) )
> 
> getting error
> 
> node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt: (361, 97): Unresolved reference: CodeType

use the following import

import com.mrousavy.camera.types.CodeType

Even after all this the scanner is still crashing, so I switched to react-native-camera-kit (only downside is, I can't get the type of code scanned but works perfectly for my use case

https://github.com/teslamotors/react-native-camera-kit

feardarkness, 2023-12-18:
@M4Tdev solution works (using flash as a new isActive flag on Android) for both camera and barcode scanner. I've noticed that I had to set an small delay when setting isActive to true in some android devices.
```
<Camera
  // all other properties
   isActive={isCameraActive}
   flash={isCameraActive}
   />
```

Ivanindiani, 2023-12-18:
I have the flash in a state, when changing the state the camera activates correctly.

fontesrp, 2023-12-18:
Initializing `isActive` as `false` and switching it to `true` a second after the Camera mounts work on my test devices as well:

```ts
  const [isActive, setIsActive] = useState(Platform.OS === 'ios')

  useEffect(() => {
    if (Platform.OS === 'ios') {
      return
    }

    if (permissionsGranted && device) {
      const timeout = setTimeout(() => setIsActive(true), 1e3)
      return () => clearTimeout(timeout)
    }

    setIsActive(false)
  }, [device, permissionsGranted])

  return appState === 'active' && isActive
```

FazilMuhammed, 2023-12-19:
@fontesrp can you share full screen code here. it will help to understand all

iqorlobanov, 2023-12-19:
> @fontesrp can you share full screen code here. it will help to understand all

Something like that. I actually had another issue with the camera on Android, but it was also fixed by setting isActive to true after a short delay.

```typescript
const [isCameraInitialized, setIsCameraInitialized] = useState(false);
const [isActive, setIsActive] = useState(Platform.OS === 'ios');

const isFocused = useIsFocused();
const isForeground = useIsForeground();

useEffect(() => {
      if (Platform.OS === 'ios') {
          return () => {};
      }

      let timeout: NodeJS.Timeout;

      if (isCameraInitialized) {
          timeout = setTimeout(() => setIsActive(true), 150);
      }

      setIsActive(false);
      return () => {
          clearTimeout(timeout);
      };
  }, [isCameraInitialized]);

const onInitialized = useCallback(() => {
    setIsCameraInitialized(true);
}, []);

<Camera
    ...
    isActive={isActive && isFocused && isForeground && isCameraInitialized}
    onInitialized={onInitialized}
/>
```

cu4nt0m, 2023-12-19:
3.5.1 Currently works for me, this is really weird to deal with.

paulovictor237, 2023-12-19:
> @fontesrp can you share full screen code here. it will help to understand all

it's worked

<details>
<summary>full code</summary>

```tsx
import React, { useEffect, useRef, useState } from 'react';
import { Button, StyleSheet, Text, View } from 'react-native';
import {
  Camera,
  useCameraDevice,
  CameraPosition,
  useCameraPermission,
  useMicrophonePermission,
} from 'react-native-vision-camera';

export default function CameraPeve() {
  const [active, setActive] = useState(false);
  const [hasPermission, setHasPermission] = useState(false);
  const [cameraType, setCameraType] = useState<CameraPosition>('back');
  const device = useCameraDevice(cameraType);
  const camera = useRef<Camera>(null);

  const {
    hasPermission: hasCameraPermission,
    requestPermission: requestCameraPermission,
  } = useCameraPermission();

  const {
    hasPermission: hasMicrophonePermission,
    requestPermission: requestMicrophonePermission,
  } = useMicrophonePermission();

  useEffect(() => {
    if (hasCameraPermission && hasMicrophonePermission) setHasPermission(true);
  }, [hasCameraPermission, hasMicrophonePermission]);

  useEffect(() => {
    if (!hasPermission || !device) return;
    const timeout = setTimeout(() => setActive(true), 1e3);
    return () => clearTimeout(timeout);
  }, [device, hasPermission]);

  const toggleCamera = () =>
    setCameraType((type) => (type === 'front' ? 'back' : 'front'));

  const takePhoto = async () => {
    try {
      const photo = await camera.current?.takePhoto();
      console.log('🐞 ~ photo:', photo);
    } catch (error) {
      console.log('🐞 ~ error:', error);
    }
  };

  if (!device || !hasPermission) {
    return (
      <View style={styles.container}>
        <Button
          onPress={requestCameraPermission}
          title="Camera permission"
          disabled={hasCameraPermission}
        />

        <Button
          onPress={requestMicrophonePermission}
          title="Microphone permission"
          disabled={hasMicrophonePermission}
        />
      </View>
    );
  }

  return (
    <View style={styles.container}>
      <Button
        onPress={() => setActive((p) => !p)}
        title={active ? 'Desativar Camera' : 'Active Camera'}
      />
      <Button onPress={toggleCamera} title={'Toggle Camera'} />
      <Button onPress={takePhoto} title={'Take Photo'} />
      {!active && <Text style={styles.warning}>Camera is not Active</Text>}
      <Camera
        ref={camera}
        isActive={active}
        device={device}
        photo={true}
        style={{ flex: 1 }}
      />
    </View>
  );
}

const styles = StyleSheet.create({
  warning: {
    fontSize: 30,
    fontWeight: 'bold',
    textAlign: 'center',
    color: 'red',
    backgroundColor: 'yellow',
  },
  container: {
    flex: 1,
    backgroundColor: 'white',
  },
  bold: {
    fontWeight: 'bold',
  },
});

```
</details>

SungKyuMM, 2023-12-20:
`setTimeout` solved this issue.
However, since I need to use the camera frequently, 1 second is too bothersome...

iqorlobanov, 2023-12-20:
>  `setTimeout` solved this issue.
> 
> However, since I need to use the camera frequently, 1 second is too annoying...

You can reduce the timeout to 150ms. This is enough to avoid the problem

SungKyuMM, 2023-12-20:
When I tested with 500ms, it didn't work, so I restored it to 1 second.

SungKyuMM, 2023-12-20:
When I tested with 500ms on S23, it didn't work, so I restored it to 1 second.

iqorlobanov, 2023-12-20:
> When I tested with 500ms in S23, it didn't work, so I restored it to 1 second.

Try setting a timeout after the camera is initialized (onInitialized() method)

SungKyuMM, 2023-12-20:
> Try setting a timeout after the camera is initialized (onInitialized() method)
> 
> [#2291 (comment)](https://github.com/mrousavy/react-native-vision-camera/issues/2291#issuecomment-1862274019)

When I used `onInitialized`, solved it. thank you!  :).

erwanlpfr, 2023-12-20:
I tried almost every workaround here and this always appeared on random devices.
I downgraded to the previous version : `"react-native-vision-camera": "^3.6.16"`.

andreasaker, 2023-12-22:
Encountered the same issue where "Camera" doesn't appear to register "isActive," resulting in a black screen.
After almost ripping my only hair strand i saw this thread and downgraded to "3.6.16". Now it works flawlessly again, thank you all. 👍
Downgraded from "3.6.17" to "3.6.16".

amilaupendra, 2023-12-22:
Downgraded from "3.6.17" to "3.6.16". It works.

BLOCKMATERIAL, 2023-12-27:
Downgraded from "3.6.17" to "3.6.16" also works.

ping, need hot-fix

phone: Google Pixel 4

mrousavy, 2024-01-05:
Hey! Thanks for reporting this issue.

I've been working the past days on making sure the Camera lifecycle is safely handled and released when needed to make sure this crash doesn't occur anymore.

I just created a PR for this - can you please this to see if that fixes the issue for you? https://github.com/mrousavy/react-native-vision-camera/pull/2339

If this fixes your issue, please consider [💖 **sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy/) to support me / thank me for building VisionCamera and continuously improving it. 

If this does not fix your issue, please clone the repo, check out the branch `fix/blackscreen` (the PR above), run the Example app, reproduce your issue there and share the `adb logcat` logs with me (upload to pastebin or gist) so I can investigate this further.

Thank you! 🙏

closed 2024-01-08 by mrousavy

aberic1992, 2024-01-25:
I just wanna add, that this is not fixed in the latest 2.x.x version. 
I solved this buy rerendering camera component by changing key prop on every focus.

mrousavy, 2024-01-25:
> I just wanna add, this is not fixed in the latest 2.x.x version.

Yes, because `main` branch is V3 since almost half a year

aberic1992, 2024-01-25:
@mrousavy I understand, but we just couldn't easily update to the latest version, and we needed solution asap.

tomerh2001, 2024-02-05:
On `3.8.2` this is still an issue

genaroNF, 2024-04-07:
Hello, not sure if what I am experiencing is realted to this error but I created an app using the example in the readme and got this problem where everything is black except for sources of light like a lightbulb I am adding a screenshot down below. Any idea what could be happening? [Here ](https://github.com/genaroNF/oso-hormiguero-mobile) is the code I am using

[image]

genaroNF, 2024-04-07:
I want to clarify that my camera is working just fine

Yugank-mialo, 2024-08-12:
<Camera
            style={StyleSheet.absoluteFill}
           ...
            isActive={true}
          />
          just add isActive={true} its worked for me

mduvarci12, 2024-11-13:
> Hello, not sure if what I am experiencing is realted to this error but I created an app using the example in the readme and got this problem where everything is black except for sources of light like a lightbulb I am adding a screenshot down below. Any idea what could be happening? [Here ](https://github.com/genaroNF/oso-hormiguero-mobile) is the code I am using
> 
> ![Screenshot_2024-04-07-17-12-06-032_com osohormigueroapp](https://private-user-images.githubusercontent.com/40221978/320311551-30ddce9d-bcf9-478f-9c11-b3ee6a4ff2cf.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzE1MTQ2NDAsIm5iZiI6MTczMTUxNDM0MCwicGF0aCI6Ii80MDIyMTk3OC8zMjAzMTE1NTEtMzBkZGNlOWQtYmNmOS00NzhmLTljMTEtYjNlZTZhNGZmMmNmLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTEzVDE2MTIyMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTViMTdiODRjYWIxMzI2NGIzNDZjMjNkNmE1MjNjYjkzZDc4M2ZiNmE0MjQ0MjIwZmRmZDQxOGRjZjRkMzk3NDkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.xHxITHyAV0XvBTwNn4eEJoSQ8adMdF2rOcMhbZNnuAA)

I have the same issue, no issues on some devices but it is very dark on some did you find a solution?

junpeace, 2024-12-09:
> Hello, not sure if what I am experiencing is realted to this error but I created an app using the example in the readme and got this problem where everything is black except for sources of light [...]

same issue... after i set exposure to 0 is working fine now

babldev, 2025-01-24:
I was also seeing `camera-is-restricted` errors if you leave the `<Camera isActiv={true}>` while the app backgrounds on Android.

I resolved this with:

```tsx
export default function App(): React.JSX.Element {
  const [appStateVisible, setAppStateVisible] = useState<AppStateStatus>();

  // Disable the camera when the app is not forgrounded.
  // Additionally, re-enable the camera when the app re-enters the foreground.
  // Without this, the OS may stop the camera session and <Camera> will not restart.
  useEffect(() => {
    const subscription = AppState.addEventListener('change', setAppStateVisible);
    return () => {
      subscription.remove();
    };
  }, []);

  return <Camera isActive={appStateVisible == "active"} />
}
```

SaikatGlitchCodes, 2025-02-26:
Drowning myself into space of solutions which ultimately raise more errors, Finally thing which worked for me was patching the package.

Version down your package to 3.1.x and see if it works if it does, compare code with latest version 4.2.1 and start understanding  difference in the code mostly check for camera1 file.

For me changing/reverting to previous version worked perfectly.
[image]
[image]
[image]

Finally you would want to patch it, follow : https://anu-thomas.medium.com/patch-package-in-react-native-c7786a15e279

Well that should work!, if not comment with screenshot what error are you getting