margelo/react-native-vision-camera #2293: 🐛 Camera freezes on first frame when running frameProcessors on Galaxy S8, Android 9

joncys, 2023-12-15:
### What's happening?

So I am testing frame processors on SAMSUNG Galaxy S8, running Android 9.0. The previous version works fine (v2), but on v3 the Camera freezes on the first frame, when (and only when) running frame processors. If I comment out the `frameProcessor` prop, it works fine. So in theory I am able to have the worklet initialized, it's only when I actually try to use it and bind it to the camera, it freezes and stops the camera from working. Also the console.logs from the frame processor are not being logged, so the worklet does not seem to be running as well (or binded).

I tried running the example app and the same thing happens over there as well. Will be pasting both the example app and my app logs. Both the logs are just launching the app and it arriving on the camera preview screen. I have dealt with permissions beforehand. I open the apps up and then the camera freezes, so I wait a bit, pressing on the screen so that it doesn't lock the phone, and then I open up the app list to close the app and end the session.

I'm super not familiar with native development, so I might be using the wrong lingo or something and this is quite as much context as I can possibly provide.

### Reproduceable Code

```tsx
// this is for my app, but nothing really special here, so I would
// assume that the problem is somewhere deep down

const frameProcessor = useFrameProcessor((frame) => {
    'worklet'
    console.log('hello')
    return null
  }, [])

  const device = useCameraDevice('back')

  if (device == null) return <Text>No camera</Text>
  return <Camera
    style={{
      flex: 1,
      width: '100%',
    }}
    device={device}
    isActive={true}
    frameProcessor={frameProcessor}
  />
```

### Relevant log output

```shell
It says, that my logs are too long, I've created a pastebin
https://pastebin.com/aEezEEZi

Nothing interesting on the React Native logs
```

### Camera Device

```json
{
  "sensorOrientation": "landscape-right",
  "hardwareLevel": "full",
  "maxZoom": 8,
  "minZoom": 1,
  "maxExposure": 8,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": true,
  "isMultiCam": false,
  "minExposure": -8,
  "name": "BACK (0)",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

SAMSUNG Galaxy S8

### VisionCamera Version

3.6.16

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [X] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Wondae-code, 2024-01-08:
Is there any progress??

I'm using similar device (Samsung Galaxy Note 8) and same OS. and 
I'm experiencing the same problem.

If you solved this problem, please share any Information.

stewartiee, 2024-01-18:
Experiencing the same on a new project, on a Pixel 3. First frame shows for a second, and then the app crashes.

Removing the `frameProcessor` on the Camera component, and it then shows fine.

yuraxdrumz, 2024-01-18:
experiencing the same issue on a one plus nord with android 11

gdaitis, 2024-01-19:
same exact problem on the same exact device. Has anyone taken a look at this?

fazzysyed, 2024-01-22:
```
  const format = React.useMemo(() => {
    const desiredWidth = 1920;
    const desiredHeight = 1080;
    if (device) {
      for (let index = 0; index < device.formats.length; index++) {
        const format = device.formats[index];
        if (format) {
          if (
            format.videoWidth == desiredWidth &&
            format.videoHeight == desiredHeight
          ) {
            return format;
          }
        }
      }
    }
    return undefined;
  }, [device?.formats]);
```

I had the same problem, but when I assigned the correct format, the camera worked fine

masanchez-hyve, 2024-01-25:
I'm having the same issue with POCO F3 Pro

mrousavy, 2024-01-30:
Thanks for sharing the logs @joncys - unfortunately I cannot access pastebin rn so I'll take another looks later, as an alternative you could use gist.github.com for hosting the logs :)

[image]

touchmarine, 2024-02-01:
I had the same issue, but it works with `pixelFormat="yuv"`. (Galaxy S8+, Android 9)

This is already the default pixel format on `main` (https://github.com/mrousavy/react-native-vision-camera/commit/11eb1f68e4912309589855ee81edea12f5261df4, https://github.com/mrousavy/react-native-vision-camera/commit/d64fc08e167a6f2787f6deccfb3ba0caf16557b0) and used by the codeScanner.

gdaitis, 2024-02-01:
@mrousavy here is the content from @joncys pastebin: https://gist.github.com/gdaitis/47f62b5b994b28f5edc55e7481d1f71e

mrousavy, 2024-02-01:
Hmm, there's nothing that's particularily interesting in those logs.... 🤔

joncys, 2024-02-01:
Tried running it with 3.9.0 beta and `pixelFormat={'yuv'}` set.
Now it does not freeze, but the whole app crashes, maybe that is something that you can work with.

Sharing logs for the crash here:
https://gist.github.com/joncys/303fcbadf05675fbf70ac001a7f29183

mrousavy, 2024-02-01:
```
2024-02-01 17:29:19.942 10876-10974 HermesVM                lt.joncys.example                    I  Compiling JS failed: 1:1:invalid empty parentheses '( )',  Buffer size 3 starts with: 280a29 and has protection mode(s): rw-p
2024-02-01 17:29:19.981 10876-10974 VideoPipeline           lt.joncys.example                    E  FrameProcessor/ImageReader pipeline threw an error!
                                                                                                    com.facebook.jni.CppException: Compiling JS failed: 1:1:invalid empty parentheses '( )' Buffer size 3 starts with: 280a29 and has protection mode(s): rw-p
```

@joncys This is a different issue and has been reported before a lot - it is basically some compatibility issue in your Babel plugins (Worklets vs Reanimated). Make sure to clean cache, update your versions, and try again.

Once that is gone let me know if 3.9 works :)

marcshilling, 2024-02-01:
All my users with POCO F2/F3 devices reporting this freeze. I'm already using `pixelFormat="yuv"` so can't be that. Unfortunately I don't have this device to test with/reproduce.

mrousavy, 2024-02-02:
@marcshilling are you on 3.9-beta.0?

marcshilling, 2024-02-02:
@mrousavy nah 3.8.2

brunoobatista, 2024-02-07:
I had this issue with the frameProcessor on a Samsung A8 with Android 9, but when I tested the RNVC example application, where there is the implementation of frameProcessors, it works fine. I tried applying what's there in an Expo project as well, but the same problem of freezing the camera upon opening occurs. When I created the project without Expo for testing, it worked normally.

rnvc: 3.8.2
expo: 49-50
rn: 0.72-0.73

krzychuuu132, 2024-02-10:
3.9.0-beta.2 the same issue ://

rampantslug, 2024-02-20:
I have a Poco F3 and using a frameprocessor that freezes on first frame. I am using scanOCR from vision-camera-ocr.
I'll do some more tests but if I just log inside the frameprocessor it is fine.
Cant be sure if it is the scanOCR call or the attempt to run in js using - Worklets.createRunInJsFn()
I'm on 3.8.2
I'll test using a worklet without processing OCR and report back

rampantslug, 2024-02-20:
Yep. Any attempt (just attempting to console.log frame details) to use a frameprocessor with a worklet causes it to freeze after first frame.

Thormeard, 2024-03-05:
Same issue. Empty frameProcessor (with just console.log), freezing. If I comment out frameProcessor prop, it works fine.

Built-in QR scanner is working (if this matter)

OnePlus Nord AC2003, Android 12, kernel 4.19.157-perf+

vision camera 3.9.1
RN 73.4
react-native-worklets-core 0.3.0

DanInDev, 2024-03-06:
Just had a very similiar issue. I fixed it by setting:

```bash
photo={true}
```

Also saw in a thread similiar that:

```bash
video={true}
```
Also worked for some.

Kavosh-m, 2024-03-17:
Same issue with some SAMSUNG devices.

`const format = useCameraFormat(device, [{ videoResolution: { width: 1920, height: 1080 } }, { fps: 30 }])`

```
<Camera
                        key={'camera1'}
                        style={StyleSheet.absoluteFill}
                        isActive={isActive}
                        device={device}
                        format={format}
                        frameProcessor={frameProcessor}
                        pixelFormat="yuv"
                        onStarted={() => {
                            console.log('camera is ready');
                            updateCropRegion();
                        }}
                        onError={error => {
                            logToSentry(error, true, 'Camera runtime error happend');
                        }}
                    />
```

mrousavy, 2024-03-19:
Hey! I just released a new V4 beta ([v4.0.0-beta.7](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v4.0.0-beta.7)) where I fixed a bunch of issues! 💪🚀 Can you test that and see if that fixes the issue for you? 😅 

```
yarn add react-native-vision-camera@beta
```

You might need to increase the `compileSdk` to the newest Android SDK (34) if you are behind.

mrousavy, 2024-03-25:
If that bug got fixed in latest V4 beta, please consider [💖 sponsoring me on GitHub 💖](http://github.com/sponsors/mrousavy/) so I can keep maintaining this library, fixing bugs and building new features! :)

closed 2024-03-25 by mrousavy