margelo/react-native-vision-camera #2294: 🐛 Android: Intermittent crash going from codeScanner to photo Camera

spsaucier, 2023-12-15:
### What's happening?

`navigation.navigate(...)` directly from Camera Scan screen to Camera Photo screen seems to cause a race which sometimes works properly but sometimes causes either an ANR or a crash.

ANR after dismissing this:
[image]

I tried building the Example app and was unable to get the JS build working:
```
Error: While trying to resolve module `@mgcrea/vision-camera-code-scanner` from file `vision-camera-barcode-scanner/example/src/App.tsx`, the package `vision-camera-barcode-scanner/example/node_modules/@mgcrea/vision-camera-code-scanner/package.json` was successfully found. However, this package itself specifies a `main` module field that could not be resolved (`vision-camera-barcode-scanner/example/node_modules/@mgcrea/vision-camera-code-scanner/dist/index.js`.
```

### Reproduceable Code

```tsx
// navigation.navigate(...) from Scan to Photo camera screens

const device = useCameraDevice('back', {
  physicalDevices: [
    'ultra-wide-angle-camera',
    'wide-angle-camera',
    'telephoto-camera',
  ],
});

// Scanner
{device != null && hasPermission && (
  <Camera
    key={`${resetKey}`}
    onError={error => {
      if (error.message.includes('CameraDevice was already closed')) {
        setResetKey(prev => prev + 1);
      }
    }}
    style={styles.preview}
    device={device}
    isActive={isFocused}
    codeScanner={onCodeScanned}
    orientation="portrait"
    format={format}
    resizeMode="contain"
    enableZoomGesture
    enableHighQualityPhotos={false}
    torch={useTorch ? 'on' : 'off'}
  ></Camera>
)}

// Photo
{preview && device != null && hasPermission ? (
  <Camera
    key={`${resetKey}`}
    onError={error => {
      if (
        error.message.includes('CameraDevice was already closed')
      ) {
        setResetKey(prev => prev + 1);
      }
    }}
    // @ts-ignore - ref appears to be forwarded properly to Camera component
    ref={cameraRef}
    device={device}
    isActive={isFocused}
    style={styles.preview}
    photo
    orientation="portrait"
    torch={useTorch ? 'on' : 'off'}
    onInitialized={() => setIsInitialized(true)}
    enableZoomGesture
    enableShutterSound={false}
  />
[... 237 lines cut ...]
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Moto G Stylus (2022), Google Pixel 6

### VisionCamera Version

3.6.12

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-01-05:
Hey! Thanks for reporting this issue.

I've been working the past days on making sure the Camera lifecycle is safely handled and released when needed, so this crash doesn't occur anymore.

I just created a PR for this - can you please this to see if that fixes the issue for you? https://github.com/mrousavy/react-native-vision-camera/pull/2339

If this fixes your issue, please consider [💖 **sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy/) to support me / thank me for building VisionCamera and continuously improving it. 

If this does not fix your issue, please clone the repo, check out the branch `fix/blackscreen` (the PR above), run the Example app, reproduce your issue there and share the `adb logcat` logs with me (upload to pastebin or gist) so I can investigate this further.

Thank you! 🙏

closed 2024-01-08 by mrousavy

mrousavy, 2024-02-06:
Hey, did you try the latest version?