margelo/react-native-vision-camera #2295: 🐛 [Android] android.os.ServiceSpecificException: supportsCameraApi:2983: Unknown camera ID 1

DzianisAleshkou, 2023-12-15:
### What's happening?

Black screen and unable to take a picture

### Reproduceable Code

```tsx
const backCamera = useCameraDevice("back", {
    physicalDevices: ["wide-angle-camera"]
  });
  const frontCamera = useCameraDevice("front");
  const [cameraType, setCameraType] = useState(CAMERA_TYPE.BACK);

  const device = cameraType === CAMERA_TYPE.BACK ? backCamera : frontCamera;

  const cameraFormat = useCameraFormat(device, [
    { videoResolution: { width: 1280, height: 720 } },
    { fps: 60 }
  ]);

 {!!device && !!hasPermission && (
        <>
          <Camera
            style={absoluteFill}
            ref={camera}
            isActive={isActive}
            device={device}
            photo={true}
            format={Platform.OS === "android" ? cameraFormat : undefined}
            frameProcessor={frameProcessor}
            pixelFormat="yuv"
            torch={torch}
          /> )}
```

### Relevant log output

```shell
android.os.Parcel.createExceptionOrNull Parcel.java:2400
android.os.Parcel.createException Parcel.java:2370
android.os.Parcel.readException Parcel.java:2353
android.os.Parcel.readException Parcel.java:2295
android.hardware.ICameraService$Stub$Proxy.supportsCameraApi ICameraService.java:1019
android.hardware.camera2.CameraManager.supportsCameraApiLocked CameraManager.java:1385
android.hardware.camera2.CameraManager.supportsCamera2ApiLocked CameraManager.java:1361
android.hardware.camera2.CameraManager.getCameraCharacteristics CameraManager.java:592
com.mrousavy.camera.core.PreviewView.resizeToInputCamera PreviewView.kt:50
com.mrousavy.camera.CameraView.setCameraId CameraView.kt:54
com.mrousavy.camera.CameraViewManager.setCameraId CameraViewManager.kt:41
java.lang.reflect.Method.invoke Method.java
com.facebook.react.uimanager.ViewManagersPropertyCache$PropSetter.updateViewProp ViewManagersPropertyCache.java:93
com.facebook.react.uimanager.ViewManagerPropertyUpdater$FallbackViewManagerSetter.setProperty ViewManagerPropertyUpdater.java:136
com.facebook.react.uimanager.ViewManagerPropertyUpdater.updateProps ViewManagerPropertyUpdater.java:56
com.facebook.react.uimanager.ViewManager.updateProperties ViewManager.java:86
com.facebook.react.uimanager.NativeViewHierarchyManager.updateProperties NativeViewHierarchyManager.java:142
com.facebook.react.uimanager.UIViewOperationQueue$UpdatePropertiesOperation.execute UIViewOperationQueue.java:93
com.facebook.react.uimanager.UIViewOperationQueue$1.run UIViewOperationQueue.java:915
com.facebook.react.uimanager.UIViewOperationQueue.flushPendingBatches UIViewOperationQueue.java:1026
com.facebook.react.uimanager.UIViewOperationQueue.access$2600 UIViewOperationQueue.java:47
com.facebook.react.uimanager.UIViewOperationQueue$DispatchUIFrameCallback.doFrameGuarded UIViewOperationQueue.java:1086
com.facebook.react.uimanager.GuardedFrameCallback.doFrame GuardedFrameCallback.java:29
com.facebook.react.modules.core.ReactChoreographer$ReactChoreographerDispatcher.doFrame ReactChoreographer.java:175
com.facebook.react.modules.core.ChoreographerCompat$FrameCallback$1.doFrame ChoreographerCompat.java:85
android.view.Choreographer$CallbackRecord.run Choreographer.java:1008
android.view.Choreographer.doCallbacks Choreographer.java:809
android.view.Choreographer.doFrame Choreographer.java:740
android.view.Choreographer$FrameDisplayEventReceiver.run Choreographer.java:995
android.os.Handler.handleCallback Handler.java:938
android.os.Handler.dispatchMessage Handler.java:99
android.os.Looper.loop Looper.java:246
android.app.ActivityThread.main ActivityThread.java:8615
java.lang.reflect.Method.invoke Method.java
com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run RuntimeInit.java:602
com.android.internal.os.ZygoteInit.main ZygoteInit.java:1130
```

### Camera Device

```json
I can't check it on that device because issue from crash reports
```

### Device

Galaxy Tab A (8.4", 2020)

### VisionCamera Version

3.6.11

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-01-30:
Hey - I think the issue has been fixed in the latest version of VisionCamera! Please try it :)

If not, just re-open the issue

closed 2024-01-30 by mrousavy

leileiChenGentleman, 2024-11-05:
supportsCameraApi:1765: Unknown camera ID 1

androidx.camera.core.CameraX.void lambda$initAndRetryRecursively$2(android.content.Context,java.util.concurrent.Executor,androidx.concurrent.futures.Callba
androidx.camera.core.InitializationException:java.lang.IllegalArgumentException: supportsCameraApi:1765: Unknown camera ID 1
2
androidx.camera.core.CameraX.void lambda$initAndRetryRecursively$2(android.content.Context,java.util.concurrent.Executor,androidx.concurrent.futures.CallbackToFutureAdapter$Completer,long)(CameraX.java:369)
3
......
4
Caused by:
5
android.os.ServiceSpecificException:supportsCameraApi:1765: Unknown camera ID 1
6
android.os.Parcel.readException(Parcel.java:2027)
7
android.os.Parcel.readException(Parcel.java:1959)
8
android.hardware.ICameraService$Stub$Proxy.supportsCameraApi(ICameraService.java:531)
9
android.hardware.camera2.CameraManager.supportsCameraApiLocked(CameraManager.java:714)
10
android.hardware.camera2.CameraManager.supportsCamera2ApiLocked(CameraManager.java:690)
11
android.hardware.camera2.CameraManager.getCameraCharacteristics(CameraManager.java:236)
12
androidx.camera.camera2.internal.compat.CameraManagerCompatBaseImpl.android.hardware.camera2.CameraCharacteristics getCameraCharacteristics(java.lang.String)(CameraManagerCompatBaseImpl.java:117)
13
androidx.camera.camera2.internal.compat.CameraManagerCompat.androidx.camera.camera2.internal.compat.CameraCharacteristicsCompat getCameraCharacteristicsCompat(java.lang.String)(CameraManagerCompat.java:189)
14
androidx.camera.camera2.internal.SupportedSurfaceCombination.void <init>(android.content.Context,java.lang.String,androidx.camera.camera2.internal.compat.CameraManagerCompat,androidx.camera.camera2.internal.CamcorderProfileHelper)(SupportedSurfaceCombination.java:144)
15
androidx.camera.camera2.internal.Camera2DeviceSurfaceManager.void init(android.content.Context,androidx.camera.camera2.internal.compat.CameraManagerCompat,java.util.Set)(Camera2DeviceSurfaceManager.java:109)
16
androidx.camera.camera2.internal.Camera2DeviceSurfaceManager.void <init>(android.content.Context,androidx.camera.camera2.internal.CamcorderProfileHelper,java.lang.Object,java.util.Set)(Camera2DeviceSurfaceManager.java:97)
17
androidx.camera.camera2.internal.Camera2DeviceSurfaceManager.void <init>(android.content.Context,java.lang.Object,java.util.Set)(Camera2DeviceSurfaceManager.java:69)
18
androidx.camera.camera2.internal.Camera2DeviceSurfaceManager.void <init>(android.content.Context,androidx.camera.camera2.internal.CamcorderProfileHelper,java.lang.Object,java.util.Set)(Camera2DeviceSurfaceManager.java:69)
19
androidx.camera.camera2.Camera2Config.androidx.camera.core.impl.CameraDeviceSurfaceManager lambda$defaultConfig$0(android.content.Context,java.lang.Object,java.util.Set)(Camera2Config.java:54)
20
androidx.camera.camera2.Camera2Config.androidx.camera.core.impl.CameraDeviceSurfaceManager $r8$lambda$kPfcZYYQs2fXKvw1cjC5GhRy6hM(android.content.Context,java.lang.Object,java.util.Set)(Camera2Config.java:0)
21
androidx.camera.camera2.Camera2Config$$InternalSyntheticLambda$1$b737bf8082c5f7ea267385a154eefca2640a55b974e0d2599d3c3cd43ba9dca3$1.androidx.camera.core.impl.CameraDeviceSurfaceManager newInstance(android.content.Context,java.lang.Object,java.util.Set)(Camera2Config.java:0)
22
androidx.camera.core.CameraX.void lambda$initAndRetryRecursively$2(android.content.Context,java.util.concurrent.Executor,androidx.concurrent.futures.CallbackToFutureAdapter$Completer,long)(CameraX.java:317)
23
androidx.camera.core.CameraX.void $r8$lambda$dnuYhxGsACX_lH-9SVlUVTdtTJg(androidx.camera.core.CameraX,android.content.Context,java.util.concurrent.Executor,androidx.concurrent.futures.CallbackToFutureAdapter$Completer,long)(CameraX.java:0)
24
androidx.camera.core.CameraX$$InternalSyntheticLambda$4$845f10987ef598ac292769067d92d63fd4aa661f29de3bb3ed47d03f5eaa76ed$0.void run()(CameraX.java:0)
25
java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1162)
26
java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:636)
27
java.lang.Thread.run(Thread.java:764)

VisionCamera Version 4.0.0