margelo/react-native-vision-camera #2296: If I using USB type external camera that not showing in android and it only shows the back(0 index) and front(1 index) but not showing the external(2 index).🐛

Sathishkumargrc, 2023-12-16:
### What's happening?

If I using USB type external camera that not showing in android and it only shows the back(0 index) and front(1 index) but not showing the external(2 index). can you please look around it and fix this. it's not showing the external device in -> getAvailableDevice(). 🐛 

### Reproduceable Code

```tsx
import {Alert, StyleSheet, Text, View} from 'react-native';
import React, {useEffect} from 'react';
import {
  Camera,
  useCameraDevice,
  useCameraPermission,
  useMicrophonePermission,
} from 'react-native-vision-camera';
import {useIsFocused} from '@react-navigation/native';
import {useAppState} from '@react-native-community/hooks';

const ExternalCameraScreen = () => {
  const {hasCamPermission, requestCamPermission} = useCameraPermission();
  const {hasMicPermission, requestMicPermission} = useMicrophonePermission();

  const NoCameraDeviceError = () => (
    <View>
      <Text
        style={{
          color: 'black',
          fontSize: 16,
          alignItems: 'center',
          justifyContent: 'center',
          alignSelf: 'center',
          marginTop: 10,
        }}>
        No camera device found
      </Text>
    </View>
  );

  const devices = Camera.getAvailableCameraDevices();

  const device = useCameraDevice('external');
  const usbCamera = useCameraDevice('external');
  const external = usbCamera || device;
  console.log('usbCamera >>>',usbCamera)

  const isFocused = useIsFocused();
  const appState = useAppState();
  const isActive = isFocused && appState === 'active';

    devices.map(device => {
      console.log('device', device.name);
    });
  }, []);

  if ((external || device || usbCamera) == null) return <NoCameraDeviceError />;
  return <Camera style={{flex: 1}} device={external} isActive={isActive} />;
  if (usbCamera != null) {
      return (
          <Camera
              style={{ flex: 1 }}
              device={usbCamera}
              isActive={isActive}
          />
      )
  } else {
      return (
          <Camera
              style={{ flex: 1 }}
              device={device}
              isActive={isActive}
          />
      )
  }
};

export default ExternalCameraScreen;

const styles = StyleSheet.create({});
```

### Relevant log output

```shell
LOG: usbCamera >>> undefined
LOG: device back(0), front(1)
```

### Camera Device

```json
{
  "sensorOrientation": "landscape-right",
  "hardwareLevel": "full",
  "maxZoom": 8,
  "minZoom": 1,
  "maxExposure": 8,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": true,
  "isMultiCam": false,
  "minExposure": -8,
  "name": "BACK (0)",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
},
{
  "sensorOrientation": "landscape-right",
  "hardwareLevel": "full",
  "maxZoom": 8,
  "minZoom": 1,
  "maxExposure": 8,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "name": "FRONT (1)",
  "position": "back",
  "id": "0"
}
```

### Device

Vivo Y73

### VisionCamera Version

3.6.14

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-01-19:
> If I using USB type external camera that not showing in android and it only shows the back(0 index) and front(1 index) but not showing the external(2 index). can you please look around it and fix this. it's not showing the external device in -> getAvailableDevice(). 🐛

I'm sorry, I don't understand what you mean.

If the USB Camera cannot be detected, it is probably not conforming to the UVC (USB Video Class), nothing I can do - it cannot be used in a native camera app either, unless you fully write a custom USB driver specific for that camera device - obviously that is a huge amount of effort and only a specific solution for that specific Camera.

closed 2024-01-19 by mrousavy

h4nkyn, 2024-07-23:
Hello! I am using a USB camera which conforms to UVC protocol (I have confirmed it is UVC protocol with the OEM, used UVC camera-specific apps, and even swapped it out with a logitech c920) - however, I cannot see 'external' in the app either. Is this an OS-specific issue, or perhaps it is a device filter I need to add? I'm in the progress of adding a device filter to the app, but running into gradle issues I will hopefully resolve tonight. I was wondering if you yourself have any experience in getting this working on a certain android phone/OS version, and if so I would love to know which one. Thank you

mrousavy, 2024-07-23:
@h4nkyn on which OS? Android or iOS

naufanituharish, 2024-11-06:
> @h4nkyn on which OS? Android or iOS

I got same problem using Logitech HD Webcam C615 (V-U0027), it's listed on UVC class device. I'm using android device (samsung S10+ with Android 12 OS) with vision camera v4.5.1.

Is there any external work around to setup external USB camera?
I've been testing the usb camera using USB Camera App on play store and it working,.

Thanks.

onmetafox, 2024-11-22:
In my app, USB camera and it is not working correctly.

Hyeeeein, 2025-01-10:
I’ve noticed the same issue. While the external camera works fine with the default camera app on Android devices, in my project, using useCameraDevice('external') logs undefined instead.