margelo/react-native-vision-camera #2297: 🐛 Image captured from takeSnapshot is partially black

PrernaBudhraja, 2023-12-16:
### What's happening?

On One plus 10T device, image captured using takeSnapshot function is partially black.

### Reproduceable Code

```tsx
const photo = await camera?.current?.takeSnapshot({
        skipMetadata: true,
        qualityPrioritization: 'speed',
        outputPath: destinationFile,
        jpegCompression: '70',
      });

<Camera
            key={`${cameraKey}`}
            style={{
              aspectRatio: 3 / 4,
            }}
            device={device}
            isActive={isFocused}
            ref={camera}
            photo
            video={isVideo}
            audio
            torch={torchStatus(flash)}
            enableZoomGesture
            animatedProps={animatedProps}
            format={{
              ...device?.formats[0],
              photoWidth: 1920,
              photoHeight: 1440,
              videoWidth: 640,
              videoHeight: 480,
            }}
            onError={(error) => {
              console.log('RnReportLibrary -> Camera -> VisionCamera component Error:', error);
            }}
            {...(isVideo ? { orientation: 'landscapeRight' } : {})}
          />
```

### Relevant log output

```shell
-
```

### Camera Device

```json
I don't have the device so cannot specify the device format.
```

### Device

One plus 10T

### VisionCamera Version

2.15.4

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-01-30:
Hey - sorry I cannot help you without any device or logs.

closed 2024-01-30 by mrousavy

hoanglam10499, 2024-04-08:
@mrousavy 
same, Samsung S23 Ultra device when use **takeSnapshot** part of the top is black after taking the photo, problem does not occur when used **takePhoto**, but it is slower than **takeSnapshot** :(
[image]

VisionCamera Version: 2.16.8
RN Version: 0.68.7

sagitarius29, 2024-04-25:
We have same problem in our application:
[image]

Samsung A53 with Android 14