margelo/react-native-vision-camera #2300: 🐛 project run issue

vijaykhandal1290, 2023-12-18:
### What's happening?

Project not running after install react-native-worklets-core package

### Reproduceable Code

```tsx
npm:

     "react": "18.2.0",
    "react-native": "0.73.0",
    "react-native-dropdown-picker": "^5.4.6",
    "react-native-reanimated": "^3.6.1",
    "react-native-vision-camera": "^3.6.16",
    "react-native-worklets-core": "^0.2.4"

Code: const processFrame = async (frame) => {
      // Access the frame data and perform face detection and alignment
      const faces = await frame.detectFaces();

      // Process the detected faces (e.g., draw overlays, log information)
      console.log('Detected faces:', faces);
    };

    if (camera.current) {
      camera.current.setFrameProcessor(processFrame);
      camera.current.start();
    }
```

### Relevant log output

```shell
Failed to install the app. Command failed with exit code 1: gradlew.bat app:installDebug -PreactNativeDevServerPort=8081 FAILURE: Build failed with an exception. * What went wrong: Execution failed for task ':react-native-vision-camera:buildCMakeDebug[arm64-v8a]'. > com.android.ide.common.process.ProcessException: ninja: Entering directory `D:\Projects\Learning\VideoSelf\node_modules\react-native-vision-camera\android\.cxx\Debug\155s2813\arm64-v8a' [1/16] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/OpenGLRenderer.cpp.o [2/16] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/PassThroughShader.cpp.o [3/16] Building CXX 
object CMakeFiles/VisionCamera.dir/src/main/cpp/OpenGLContext.cpp.o [4/16] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrameProcessorPlugin.cpp.o [5/16] Building CXX object 
CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrame.cpp.o [6/16] Building CXX object 
CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/FrameProcessorPluginHostObject.cpp.o [7/16] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCamera.cpp.o [8/16] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/FrameHostObject.cpp.o [9/16] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/VideoPipeline.cpp.o [10/16] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/JSIJNIConversion.cpp.o [11/16] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JVisionCameraScheduler.cpp.o [12/16] Building CXX object CMakeFiles/VisionCamera.dir/D_/Projects/Learning/VideoSelf/node_modules/react-native-vision-camera/cpp/JSITypedArray.cpp.o [13/16] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/VisionCameraProxy.cpp.o [14/16] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp.o [15/16] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JVisionCameraProxy.cpp.o [16/16] Linking CXX shared library ..\..\..\..\build\intermediates\cxx\Debug\155s2813\obj\arm64-v8a\libVisionCamera.so FAILED: ../../../../build/intermediates/cxx/Debug/155s2813/obj/arm64-v8a/libVisionCamera.so 
cmd.exe /C "cd . && E:\Android\Sdk\ndk\25.1.8937393\toolchains\llvm\prebuilt\windows-x86_64\bin\clang++.exe --target=aarch64-none-linux-android21 --sysroot=E:/Android/Sdk/ndk/25.1.8937393/toolchains/llvm/prebuilt/windows-x86_64/sysroot -fPIC -g -DANDROID -fdata-sections -ffunction-sections -funwind-tables -fstack-protector-strong -no-canonical-prefixes -D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security -O2 -frtti -fexceptions -Wall -Wno-unused-variable -fstack-protector-all -fno-limit-debug-info -Wl,--build-id=sha1 -Wl,--no-rosegment -Wl,--fatal-warnings -Wl,--gc-sections -Wl,--no-undefined -Qunused-arguments -shared -Wl,-soname,libVisionCamera.so -o ..\..\..\..\build\intermediates\cxx\Debug\155s2813\obj\arm64-v8a\libVisionCamera.so CMakeFiles/VisionCamera.dir/D_/Projects/Learning/VideoSelf/node_modules/react-native-vision-camera/cpp/JSITypedArray.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCamera.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/VideoPipeline.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/PassThroughShader.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/OpenGLContext.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/OpenGLRenderer.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/FrameHostObject.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/FrameProcessorPluginHostObject.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/JSIJNIConversion.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/VisionCameraProxy.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrame.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrameProcessorPlugin.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JVisionCameraProxy.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JVisionCameraScheduler.cpp.o E:/Android/Sdk/ndk/25.1.8937393/toolchains/llvm/prebuilt/windows-x86_64/sysroot/usr/lib/aarch64-linux-android/21/liblog.so -landroid C:/Users/Vijay/.gradle/caches/transforms-3/0859cc2ebf96bc372b2c46ef5b293d59/transformed/jetified-react-android-0.73.0-debug/prefab/modules/jsi/libs/android.arm64-v8a/libjsi.so C:/Users/Vijay/.gradle/caches/transforms-3/0859cc2ebf96bc372b2c46ef5b293d59/transformed/jetified-react-android-0.73.0-debug/prefab/modules/reactnativejni/libs/android.arm64-v8a/libreactnativejni.so C:/Users/Vijay/.gradle/caches/transforms-3/057fda56c2cfc52d70a46042a30c6944/transformed/jetified-fbjni-0.5.1/prefab/modules/fbjni/libs/android.arm64-v8a/libfbjni.so -lGLESv2 -lEGL -latomic -lm && cd ." ld: error: undefined symbol: RNWorklet::JsiWorkletContext::getWorkletRuntime() >>> referenced by JFrameProcessor.cpp:37 (D:/Projects/Learning/VideoSelf/node_modules/react-native-vision-camera/android/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp:37) >>> CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp.o:(vision::JFrameProcessor::callWithFrameHostObject(std::__ndk1::shared_ptr<vision::FrameHostObject> const&) const) >>> referenced by JVisionCameraProxy.cpp:57 (D:/Projects/Learning/VideoSelf/node_modules/react-native-vision-camera/android/src/main/cpp/frameprocessor/java-bindings/JVisionCameraProxy.cpp:57) >>> CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JVisionCameraProxy.cpp.o:(vision::JVisionCameraProxy::~JVisionCameraProxy()) ld: error: undefined symbol: RNWorklet::JsiWorkletContext::invokeOnJsThread(std::__ndk1::function<void (facebook::jsi::Runtime&)>&&) >>> referenced by JFrameProcessor.cpp:50 (D:/Projects/Learning/VideoSelf/node_modules/react-native-vision-camera/android/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp:50) >>> CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp.o:(vision::JFrameProcessor::callWithFrameHostObject(std::__ndk1::shared_ptr<vision::FrameHostObject> const&) const) >>> referenced by WKTJsiWorklet.h:369 (D:/Projects/Learning/VideoSelf/node_modules/react-native-worklets-core/android/build/headers/rnworklets/react-native-worklets-core/WKTJsiWorklet.h:369) >>> CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp.o:(RNWorklet::WorkletInvoker::~WorkletInvoker()) ld: error: undefined symbol: RNWorklet::JsiWorkletContext::runtimeMappings >>> referenced by __tree:0 (E:/Android/Sdk/ndk/25.1.8937393/toolchains/llvm/prebuilt/windows-x86_64/sysroot/usr/include/c++/v1/__tree:0) >>> CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp.o:(RNWorklet::WorkletInvoker::call(facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned long)) >>> referenced by __tree:0 (E:/Android/Sdk/ndk/25.1.8937393/toolchains/llvm/prebuilt/windows-x86_64/sysroot/usr/include/c++/v1/__tree:0) >>> CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp.o:(RNWorklet::WorkletInvoker::call(facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned long)) ld: error: undefined symbol: RNWorklet::JsiWorkletContext::invokeOnWorkletThread(std::__ndk1::function<void 
[... cut ...]
### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

closed 2023-12-18 by mrousavy

samsonroyquickplay, 2023-12-20:
@mrousavy why is this closed ? Any fix for this ? 

react-native: 0.73.1
react-native-vision-camera: 3.6.17

vijaykhandal1290, 2023-12-20:
yes , I am also not understood why is this closed ?

mrousavy, 2023-12-20:
because it is the wrong issue template. Plesae use the build issue template.

vijaykhandal1290, 2023-12-20:
can you provide me proper code link because your sample code in not run in latest android studio and react native version. Please provide me write code link .

Thanks

samsonroyquickplay, 2023-12-21:
https://github.com/mrousavy/react-native-vision-camera/issues/2309 @mrousavy