margelo/react-native-vision-camera #2301: 🐛 Black Screen in Samsung S22 S23 ultra and some other devices when frame processor is on.

satyajitdas95, 2023-12-18:
### What's happening?

Whenever I am using frameprocessor the black screen is coming . Few devices i am listing below
Samsung s22 ultra
Samsung s23 Ultra
Oneplus 10R
Motorola Edge

With-out `frameprocessor` everything is fine. Probably resolution of processing frame and capturing in device causing the issue. as i debugged frameprocessor is working fine. Its able to detect faces wheather 1 or 2. That means only the preivew have the problem.

### Reproduceable Code

```tsx
const format = getCameraFormat(device, [
    {videoResolution: {width: 480, height: 720}},
    {fps: 30},
  ]);

const frameProcessor = useFrameProcessor(frame => {
    'worklet';

    try {
      runAtTargetFps(DETECTION_FPS, () => {
        // console.log(`Detected objects.`, data.number_of_faces);
        // setNumberOfFaces(data.number_of_faces);

        const data = detectFaces(frame);
        onFaceDetected(data.number_of_faces);

        // console.log('face number----', data.number_of_faces);

        // console.log(
        //   'face tilt data----',
        //   data.face_tilted_left,
        //   data.face_tilted_right,
        // );

        onFaceTiltDetected(data.face_tilted_left || data.face_tilted_right);
        // sv.value = data.number_of_faces;
        // console.log(`Detected objects.`, data.number_of_faces);
      });
    } catch (exception) {}
  }, []);

return (
    <SafeAreaView>
      <Camera
        ref={camera}
        style={StyleSheet.absoluteFill}
        device={device}
        // isActive={isFocussed}
        onInitialized={onInitialized}
        onError={onError}
        enableZoomGesture={true}
        zoom={0.6}
        photo={true}
        video={true}
        audio={false}
        format={format}
        pixelFormat="yuv"
        orientation="portrait"
        enableHighQualityPhotos={true}
        frameProcessor={frameProcessor}
      />

      {/* <CustomText
        style={{
          fontSize: 50,
          fontWeight: '800',
          color: 'red',
[... 220 lines cut ...]
  "hasTorch": false,
  "position": "front",
  "id": "1"
}
```

### Device

Samsung s22 ultra Android 13 and Android 14,Samsung s23 Ultra Android 13 and Android 14, Oneplus 10R Android 13,Motorola Edge Android 13

### VisionCamera Version

3.6.17

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

jaimeneto85, 2023-12-18:
looks like duplicate. It's same issue: #2291

thiagosimon, 2023-12-18:
Hey, guys!

I encountered an issue in my app on Android 11 physical devices. The successful solution for reading QR codes involved the following settings:

```
"react-native-vision-camera": "3.6.12",
"react-native-worklets-core": "^0.2.2",
"react-native": "0.72.7"
```

If you're using CodeScanner, you can achieve success with just the following statement:

```javascript
const onCodeScanned = useMemo(() => {
    return {
        codeTypes: ['qr'],
        onCodeScanned: !isActive
            ? () => null
            : (codes: Code[]) => {
                // Reading performed
            },
    };
}, []);

I had to lock the library version at 3.6.12, and at least for me, this solution worked. 
I hope it helps you too!

sudeepgupta11, 2023-12-19:
I am also facing the same issue with Samsung A Series devices.

CodeReset, 2023-12-19:
I sometimes encounter a black screen or black stripes during `takeSnapshot` on version 2.15.2 of react-native-vision-camera. It may not be related to react-native-vision-camera but rather to Android 14, Samsung 22, or Samsung 23. I haven't been able to reproduce the issue on other devices.

Imamyable, 2023-12-20:
Maybe something that might narrow down the possible causes of the error. I have tested it on two different samsung phones (s20 and s22 and for me the problem seems to appear when 
* frameprocessors are enabled
* pixelFormat is set to yuv
* android version is 11

When I'm using the s22 phone with android 13, frameproccessors on and pixelformat yuv everything seems to work well.  I'm thinking there might be an issue with the available pixelformats on different android versions?

ambrishshukla001, 2023-12-20:
same issue always open black screen camera is not opening

Keyur-Rao, 2023-12-28:
import {Camera, useCameraDevice} from 'react-native-vision-camera';

const camera = useRef(null);
const device = useCameraDevice('front');
const [showCamera, setShowCamera] = useState(false);

<SafeAreaView style={StyleSheet.absoluteFill}>
            <Camera
              ref={camera}
              device={device}
              isActive={showCamera}
              photo={true}
              style={StyleSheet.absoluteFill}
            />
 </SafeAreaView>

mrousavy, 2024-01-05:
Hey! Thanks for reporting this issue.

I've been working the past days on making sure the Camera lifecycle is safely handled and released when needed, so this crash doesn't occur anymore.

I just created a PR for this - can you please this to see if that fixes the issue for you? https://github.com/mrousavy/react-native-vision-camera/pull/2339

If this fixes your issue, please consider [💖 **sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy/) to support me / thank me for building VisionCamera and continuously improving it. 

If this does not fix your issue, please clone the repo, check out the branch `fix/blackscreen` (the PR above), run the Example app, reproduce your issue there and share the `adb logcat` logs with me (upload to pastebin or gist) so I can investigate this further.

Thank you! 🙏

closed 2024-01-08 by mrousavy