margelo/react-native-vision-camera #2308: 🐛 Still face issue andoid black screen when navigate to camera page

majidhidile, 2023-12-20:
### What's happening?

 const device = useCameraDevice(isFront ? 'front' : 'back');
  const format = useCameraFormat(device, [
    { videoResolution: 'max' },
    { photoResolution: 'max' },
    { videoResolution: { width: 1280, height: 720 } },
    { photoResolution: { width: 1280, height: 720 } },
  ]);
 <Camera
            orientation="portrait"
            format={format}
            device={device}
            isActive={true}
            ref={cameraRef}
            onError={(e) => {
              console.log('eRROR-->', e);
            }}
            style={[StyleSheet.absoluteFill]}
            photo={true}
            video={true}
            audio={true}
            enableZoomGesture={true}
          ></Camera>

this is my code on the camera page. when navigating this page I get a black screen.
but I temporarily solved  

  const handleCameraActive = () => {
    setisActive(false);
    setTimeout(() => {
      setisActive(true);
    }, 1500);
  };
  useEffect(() => {
    handleCameraActive();
  }, [isFront]);

also above code works very well on iOS device.

### Reproduceable Code

```tsx
<Camera
orientation="portrait"
format={format}
device={device}
isActive={true}
ref={cameraRef}
onError={(e) => {
console.log('eRROR-->', e);
}}
style={[StyleSheet.absoluteFill]}
photo={true}
video={true}
audio={true}
enableZoomGesture={true}
>
```

### Relevant log output

```shell
getting black screen when navigating to camera page.
```

### Camera Device

```json
{
  "sensorOrientation": "landscape-right",
  "hardwareLevel": "full",
  "maxZoom": 7,
  "minZoom": 0.6150000095367432,
  "maxExposure": 144,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera",
    "ultra-wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": true,
  "isMultiCam": true,
  "minExposure": -144,
  "name": "BACK (0)",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

android

### VisionCamera Version

3.6.17

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [X] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

ivosabev, 2023-12-20:
I have the same issues. I used the setTimout hack, but it doesn't work on all devices. I am also using `react-navigation`.

majidhidile, 2023-12-20:
> I have the same issues. I used the setTimout hack, but it doesn't work on all devices. I am also using `react-navigation`.

could you give example?

vinhtruong93, 2023-12-25:
I also have this issue, and i found out a way to resolve it.
Add method `onInitialized` check camera initialize, then set flag `isActive = true`. This means value `isActive=true` must after process initialize of camera is completed.
Hope this useful to you!

trilam1409, 2023-12-28:
> I also have this issue, and i found out a way to resolve it. Add method `onInitialized` check camera initialize, then set flag `isActive = true`. This means value `isActive=true` must after process initialize of camera is completed. Hope this useful to you!

Thank you. It worked for me. Could you find the root cause of this issue?

mohamed1796, 2023-12-31:
> I also have this issue, and i found out a way to resolve it. Add method `onInitialized` check camera initialize, then set flag `isActive = true`. This means value `isActive=true` must after process initialize of camera is completed. Hope this useful to you!

can you share your code

trilam1409, 2024-01-05:
You should downgrade the library to v3.6.16.

#2340

mrousavy, 2024-01-05:
Hey! Thanks for reporting this issue.

I've been working the past days on making sure the Camera lifecycle is safely handled and released when needed, so this crash doesn't occur anymore.

I just created a PR for this - can you please this to see if that fixes the issue for you? https://github.com/mrousavy/react-native-vision-camera/pull/2339

If this fixes your issue, please consider [💖 **sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy/) to support me / thank me for building VisionCamera and continuously improving it. 

If this does not fix your issue, please clone the repo, check out the branch `fix/blackscreen` (the PR above), run the Example app, reproduce your issue there and share the `adb logcat` logs with me (upload to pastebin or gist) so I can investigate this further.

Thank you! 🙏

closed 2024-01-08 by mrousavy

GolliAch, 2024-01-10:
@mrousavy Still having black screen issues with android on initial app launch using v3.7.0

mrousavy, 2024-01-10:
Please create a new issue with adb logcat logs and repro.