margelo/react-native-vision-camera #2309: 🔧

samsonroyquickplay, 2023-12-20:
### How were you trying to build the app?

I tried to run the app using yarn react-native run-android and the build failed 

### Full build logs

```tsx
Execution failed for task ':react-native-vision-camera:buildCMakeDebug[arm64-v8a]'.
> com.android.ide.common.process.ProcessException: ninja: Entering directory `/*/handset/node_modules/react-native-vision-camera/android/.cxx/Debug/326l4s4h/arm64-v8a'
  [1/16] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/OpenGLRenderer.cpp.o
  [2/16] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/PassThroughShader.cpp.o
  [3/16] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/OpenGLContext.cpp.o
  [4/16] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrameProcessorPlugin.cpp.o
  [5/16] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrame.cpp.o
  [6/16] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/FrameProcessorPluginHostObject.cpp.o
  [7/16] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCamera.cpp.o
  [8/16] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/FrameHostObject.cpp.o
  [9/16] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/VideoPipeline.cpp.o
  [10/16] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/JSIJNIConversion.cpp.o
  [11/16] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JVisionCameraScheduler.cpp.o
  [12/16] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/VisionCameraProxy.cpp.o
  [13/16] Building CXX object CMakeFiles/VisionCamera.dir/*/handset/node_modules/react-native-vision-camera/cpp/JSITypedArray.cpp.o
  [14/16] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp.o
  [15/16] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JVisionCameraProxy.cpp.o
  [16/16] Linking CXX shared library ../../../../build/intermediates/cxx/Debug/326l4s4h/obj/arm64-v8a/libVisionCamera.so
  FAILED: ../../../../build/intermediates/cxx/Debug/326l4s4h/obj/arm64-v8a/libVisionCamera.so 
  : && /Users/samsonroy/Library/Android/sdk/ndk/25.1.8937393/toolchains/llvm/prebuilt/darwin-x86_64/bin/clang++ --target=aarch64-none-linux-android21 --sysroot=/Users/samsonroy/Library/Android/sdk/ndk/25.1.8937393/toolchains/llvm/prebuilt/darwin-x86_64/sysroot -fPIC -g -DANDROID -fdata-sections -ffunction-sections -funwind-tables -fstack-protector-strong -no-canonical-prefixes -D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security  -O2 -frtti -fexceptions -Wall -Wno-unused-variable -fstack-protector-all -fno-limit-debug-info  -Wl,--build-id=sha1 -Wl,--no-rosegment -Wl,--fatal-warnings -Wl,--gc-sections -Wl,--no-undefined -Qunused-arguments -shared -Wl,-soname,libVisionCamera.so -o ../../../../build/intermediates/cxx/Debug/326l4s4h/obj/arm64-v8a/libVisionCamera.so CMakeFiles/VisionCamera.dir/*/handset/node_modules/react-native-vision-camera/cpp/JSITypedArray.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCamera.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/VideoPipeline.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/PassThroughShader.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/OpenGLContext.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/OpenGLRenderer.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/FrameHostObject.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/FrameProcessorPluginHostObject.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/JSIJNIConversion.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/VisionCameraProxy.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrame.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrameProcessorPlugin.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JVisionCameraProxy.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JVisionCameraScheduler.cpp.o  /Users/samsonroy/Library/Android/sdk/ndk/25.1.8937393/toolchains/llvm/prebuilt/darwin-x86_64/sysroot/usr/lib/aarch64-linux-android/21/liblog.so  -landroid  /Users/samsonroy/.gradle/caches/transforms-3/584646f62842007117c884443f5d4262/transformed/jetified-react-android-0.73.1-debug/prefab/modules/jsi/libs/android.arm64-v8a/libjsi.so  /Users/samsonroy/.gradle/caches/transforms-3/584646f62842007117c884443f5d4262/transformed/jetified-react-android-0.73.1-debug/prefab/modules/reactnativejni/libs/android.arm64-v8a/libreactnativejni.so  /Users/samsonroy/.gradle/caches/transforms-3/7c7ebfac43f8f21d37b50ac28c04d4da/transformed/jetified-fbjni-0.5.1/prefab/modules/fbjni/libs/android.arm64-v8a/libfbjni.so  -lGLESv2  -lEGL  -latomic -lm && :
  ld: error: undefined symbol: RNWorklet::JsiWorkletContext::getWorkletRuntime()
  >>> referenced by JFrameProcessor.cpp:37 (/*/handset/node_modules/react-native-vision-camera/android/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp:37)
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp.o:(vision::JFrameProcessor::callWithFrameHostObject(std::__ndk1::shared_ptr<vision::FrameHostObject> const&) const)
  >>> referenced by JVisionCameraProxy.cpp:57 (/*/handset/node_modules/react-native-vision-camera/android/src/main/cpp/frameprocessor/java-bindings/JVisionCameraProxy.cpp:57)
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JVisionCameraProxy.cpp.o:(vision::JVisionCameraProxy::~JVisionCameraProxy())
  
  ld: error: undefined symbol: RNWorklet::JsiWorkletContext::invokeOnJsThread(std::__ndk1::function<void (facebook::jsi::Runtime&)>&&)
  >>> referenced by JFrameProcessor.cpp:50 (/*/handset/node_modules/react-native-vision-camera/android/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp:50)
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp.o:(vision::JFrameProcessor::callWithFrameHostObject(std::__ndk1::shared_ptr<vision::FrameHostObject> const&) const)
  >>> referenced by WKTJsiWorklet.h:369 (/*/handset/node_modules/react-native-worklets-core/android/build/headers/rnworklets/react-native-worklets-core/WKTJsiWorklet.h:369)
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp.o:(RNWorklet::WorkletInvoker::~WorkletInvoker())
  
  ld: error: undefined symbol: RNWorklet::JsiWorkletContext::runtimeMappings
  >>> referenced by __tree:0 (/Users/samsonroy/Library/Android/sdk/ndk/25.1.8937393/toolchains/llvm/prebuilt/darwin-x86_64/sysroot/usr/include/c++/v1/__tree:0)
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp.o:(RNWorklet::WorkletInvoker::call(facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned long))
  >>> referenced by __tree:0 (/Users/samsonroy/Library/Android/sdk/ndk/25.1.8937393/toolchains/llvm/prebuilt/darwin-x86_64/sysroot/usr/include/c++/v1/__tree:0)
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp.o:(RNWorklet::WorkletInvoker::call(facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned long))
  
  ld: error: undefined symbol: RNWorklet::JsiWorkletContext::invokeOnWorkletThread(std::__ndk1::function<void (RNWorklet::JsiWorkletContext*, facebook::jsi::Runtime&)>&&)
  >>> referenced by WKTJsiWorklet.h:372 (/*/handset/node_modules/react-native-worklets-core/android/build/headers/rnworklets/react-native-worklets-core/WKTJsiWorklet.h:372)
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp.o:(RNWorklet::WorkletInvoker::~WorkletInvoker())
  
  ld: error: undefined symbol: RNWorklet::JsiWorkletContext::defaultInstance
  >>> referenced by memory:3887 (/Users/samsonroy/Library/Android/sdk/ndk/25.1.8937393/toolchains/llvm/prebuilt/darwin-x86_64/sysroot/usr/include/c++/v1/memory:3887)
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp.o:(RNWorklet::JsiWorkletContext::getDefaultInstance())
[... cut ...]
### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

samsonroyquickplay, 2023-12-20:
@mrousavy Can you take a look at this ?

KraljSamo, 2023-12-21:
Also experiencing this on with:

```
    "react": "18.2.0",
    "react-native": "0.72.5",
    "react-native-vision-camera": "3.6.17",
    "react-native-worklets-core": "0.2.4",
```

```
    ext {
        buildToolsVersion = "33.0.0"
        minSdkVersion = 26
        compileSdkVersion = 33
        targetSdkVersion = 33
        kotlinVersion = "1.8.20"
        
        // We use NDK 23 which has both M1 support and is the side-by-side NDK version from AGP.
        ndkVersion = "23.1.7779620"
    }
```

KraljSamo, 2023-12-21:
This fixed the problem:

```
./android/gradlew clean
rm -rf android/.gradle android/.idea android/app/build android/build
rm -rf package-lock.json yarn.lock node_modules
yarn   # or `npm i`
```

samsonroyquickplay, 2023-12-21:
"dependencies": {
    "react-native": "0.73.1",
    "react-native-reanimated": "3.6.1",
    "react-native-vision-camera": "3.6.16",
    "react-native-worklets-core": "^0.2.4",
}

Downgrading from 3.6.17 to 3.6.16 fixed the above issue

closed 2023-12-21 by samsonroyquickplay