margelo/react-native-vision-camera #2310: React-native Vision Camera unable to zoom in iOS

muralivegesna1381, 2023-12-20:
### What's happening?

I am unable to zoom in on ios devices. I am able to zoom in on android devices without any problem but unable to zoom on iPhones or iPads.

### Reproduceable Code

```tsx
<View style={styles.container}>
                <Camera
                    ref={camera}
                    style={StyleSheet.absoluteFill}
                    isActive={true}
                    photo={true}
                    orientation="portrait"
                    device={device}
                    torch={cameraFlash}
                    focusable={true}
                    enableZoomGesture={true}
                    hdr={cameraHDR}
                    lowLightBoost={cameraNightMode}
                />

                <View style={styles.captionFrame}>
                    <Image style={styles.imageStyleFrame} source={fullBodyFlip ? require('./../../../../assets/images/bfiGuide/png/dog_frame_fullbody2.png') : frameImgArray[route.params?.indexPos]} />
                </View>
            </View >
```

### Relevant log output

```shell
nothing happening when doing pinch to zoom
```

### Camera Device

```json
{
      "pixelFormat": "420v",
      "videoStabilizationModes": [
        "off",
        "cinematic"
      ],
      "maxZoom": 8,
      "minISO": 50,
      "maxISO": 3200,
      "autoFocusSystem": "none",
      "supportsPhotoHDR": true,
      "colorSpaces": [
        "yuv"
      ],
      "videoWidth": 1920,
      "videoHeight": 1080,
      "photoWidth": 1920,
      "frameRateRanges": [
        {
          "maxFrameRate": 60,
          "minFrameRate": 1
        },
        {
          "maxFrameRate": 15,
          "minFrameRate": 15
        },
        {
          "maxFrameRate": 20,
          "minFrameRate": 15
        },
        {
          "maxFrameRate": 20,
          "minFrameRate": 20
        },
        {
          "maxFrameRate": 24,
          "minFrameRate": 24
        },
        {
          "maxFrameRate": 30,
          "minFrameRate": 8
        },
        {
          "maxFrameRate": 30,
          "minFrameRate": 10
        },
        {
          "maxFrameRate": 30,
          "minFrameRate": 15
        },
        {
          "maxFrameRate": 30,
          "minFrameRate": 30
        }
      ],
      "photoHeight": 1080,
      "supportsVideoHDR": false,
      "isHighestPhotoQualitySupported": false,
      "fieldOfView": 66.3177012556713
    }
```

### Device

iphone

### VisionCamera Version

2.15.6

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

ferostabio, 2023-12-20:
Happens to me as well. All good on Android, but nothing happens on iOS -with similar code.

alexandreerick, 2024-01-15:
Same thing is happening here. Any update on this?

alexandreerick, 2024-01-15:
A little workaround is to use the `zoom` props to control the zoom with a button.
Create a state with:  `const [cameraZoom, setCameraZoom] = React.useState(device?.neutralZoom);` and then update state with something like this
```
const handleAddZoom = () => {
    setCameraZoom(oldValue => {
      if (oldValue) {
        const zoomToSet = oldValue + 1;
        if (zoomToSet > device?.maxZoom) return oldValue;

        return zoomToSet;
      }

      return 1.0;
    });
  };
```
You need to check if the zoom you are setting is bigger then `maxZoom`.
Create a function to remove zoom and then use the `cameraZoom` state on the `zoom` props.