margelo/react-native-vision-camera #2311: ❓ Preview stretched in my app but not in the example on same device

Flocurry, 2023-12-21:
### Question

I don't understand why my camera preview is stretched in my Android app but not with your example on my tablet device...

### What I tried

I use the same CameraPage.tsx file in my app like in your example.
The format below props given to the camera component is the same in my app than in your example

{
   "format":{
      "autoFocusSystem":"contrast-detection",
      "fieldOfView":85.2381275912114,
      "maxFps":30,
      "maxISO":1600,
      "maxZoom":4,
      "minFps":1,
      "minISO":100,
      "photoHeight":1600,
      "photoWidth":2560,
      "pixelFormats":[
         "yuv",
         "native"
      ],
      "supportsDepthCapture":false,
      "supportsPhotoHdr":false,
      "supportsVideoHdr":false,
      "videoHeight":960,
      "videoStabilizationModes":[
         "off",
         "off"
      ],
      "videoWidth":1440
   }
}

### VisionCamera Version

3.6.17

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

bglgwyng, 2023-12-21:
#2142 Maybe the same issue

Flocurry, 2023-12-21:
I found why the preview is stretched on my app.
This is because my app is locked in landscape
I have android:screenOrientation="landscape" in the AndroidManifest.xml file

So How can I solve my problem with the landscape orientation ?

closed 2024-01-02 by mrousavy

mrousavy, 2024-01-11:
Hey! Does this fix your issue maybe? https://github.com/mrousavy/react-native-vision-camera/pull/2377 (still testing)

mrousavy, 2024-01-11:
Hey!

After 8 hours of debugging, I finally found the culprit! I fixed the preview stretching issue in this PR: https://github.com/mrousavy/react-native-vision-camera/pull/2377

If you appreciate my dedication and free support, please consider [💖 **sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy/) so I can keep providing fixes and building out new features for my open-source libraries in my free time.

Flocurry, 2024-01-12:
Hello, I've tested the 3.7.1 and i've still the issue.

In the version 3.7.0, I've made a patch.

I've added the below line in PreviewView.kt file 

private val aspectRatio: Float get() = size.width.toFloat() / size.height.toFloat()

and replace the method onMeasure by below lines

@SuppressLint("DrawAllocation")
  override fun onMeasure(widthMeasureSpec: Int, heightMeasureSpec: Int) {
    super.onMeasure(widthMeasureSpec, heightMeasureSpec)
    val viewWidth = MeasureSpec.getSize(widthMeasureSpec)
    val viewHeight = MeasureSpec.getSize(heightMeasureSpec)

    Log.i(TAG, "PreviewView onMeasure($viewWidth, $viewHeight)")

    val newWidth: Int
    val newHeight: Int

    val actualRatio = if (viewWidth > viewHeight) aspectRatio else 1f / aspectRatio
    if (viewWidth < viewHeight * actualRatio) {
        newHeight = viewHeight
        newWidth = (viewHeight * actualRatio).roundToInt()
    } else {
        newWidth = viewWidth
        newHeight = (viewWidth / actualRatio).roundToInt()
    }

    Log.d(TAG, "Measured dimensions set: $newWidth x $newHeight")
    setMeasuredDimension(newWidth, newHeight)
  }

mrousavy, 2024-02-06:
Created a PR to fix this issue once and for all: https://github.com/mrousavy/react-native-vision-camera/pull/2519 💪🚀 
 **Thanks to everyone who sponsored me on GitHub to fix this bug!!** 💖

Try the PR and see if it works (by checking out that branch and running the example app)

Flocurry, 2024-02-07:
Hello @mrousavy, i've tested the 3.9.0-beta.2 and there's still de issue.

You can see below the preview scene and how the photo taken looks like

[image]
[image]

Flocurry, 2024-02-09:
Is there a release planned soon that fixes this issue ?

Flocurry, 2024-02-16:
Hello @mrousavy, i've tested the 3.9.0-beta.6 version, i have these log in my Logcat

[image]

I suppose this is why my imge is zoomed... I hope it can help you to fix the issue.

mrousavy, 2024-02-16:
@Flocurry are you in landscape mode or portrait

Flocurry, 2024-02-16:
I'm in landscape mode

mrousavy, 2024-02-16:
Yea okay, only portrait works atm. See https://github.com/mrousavy/react-native-vision-camera/issues/1891 for more details.

Flocurry, 2024-03-04:
Hello Mark.

Is it normal that the 4.0.0-beta.4 fix my issue ?

With this version, sometimes I have the below error when I open the camera

'Camera.onError(session/invalid-output-configuration): Failed to configure the Camera Session because the output/stream configurations are invalid!', { [session/invalid-output-configuration: Failed to configure the Camera Session because the output/stream configurations are invalid!]
                                                                                                      name: 'session/invalid-output-configuration',
                                                                                                      _code: 'session/invalid-output-configuration',
                                                                                                      _message: 'Failed to configure the Camera Session because the output/stream configurations are invalid!',
                                                                                                      _cause: 
                                                                                                       { stacktrace: 'java.util.concurrent.TimeoutException: Future[androidx.camera.core.impl.utils.futures.ListFuture@1f7eb6a] is not done within 5000 ms.\n\tat androidx.camera.core.impl.utils.futures.Futures.lambda$makeTimeoutFuture$1(Futures.java:439)\n\tat androidx.camera.core.impl.utils.futures.Futures$$ExternalSyntheticLambda0.call(Unknown Source:6)\n\tat androidx.camera.core.impl.utils.executor.HandlerScheduledExecutorService$HandlerScheduledFuture.run(HandlerScheduledExecutorService.java:245)\n\tat android.os.Handler.handleCallback(Handler.java:938)\n\tat android.os.Handler.dispatchMessage(Handler.java:99)\n\tat android.os.Looper.loopOnce(Looper.java:226)\n\tat android.os.Looper.loop(Looper.java:313)\n\tat android.os.HandlerThread.run(HandlerThread.java:67)\n',
                                                                                                         message: 'Future[androidx.camera.core.impl.utils.futures.ListFuture@1f7eb6a] is not done within 5000 ms.' } }
2024-03-04 16:52:20.201  1374-26404 Camera2-Fr...cessorBase cameraserver                         E  FrameProcessorBase: Error waiting for new frames: Connection timed out (-110)

and this error disappear if i go back and reopen the camera.

Flocurry, 2024-03-14:
Hello @mrousavy, the above issue persists with the 4.0.0-beta.6 version

mrousavy, 2024-03-14:
Which issue? Preview stretching or output configuration error?

I cannot reproduce the output configuration error anymore in the latest beta.

Flocurry, 2024-03-14:
output configuration error