margelo/react-native-vision-camera #2313: 🐛 `zoom` props doesn't work properly with camera formats of specified resolution

bglgwyng, 2023-12-21:
### What's happening?

When setting the zoom to a value other than 1 and combining it with specified camera formats and resolutions, as described in the reproducible code section, the preview results in a black screen.

### Reproduceable Code

```tsx
Modify the example app as

  const format = useCameraFormat(device, [
    { fps: targetFps },
    { videoAspectRatio: screenAspectRatio },
    { videoResolution: { width: 1920, height: 1080 } }, // it was 'max' before
    { photoAspectRatio: screenAspectRatio },
    { photoResolution: { width: 1920, height: 1080 } }, // it was 'max' before
  ])

Then set `zoom` value as any other value than `1`, for example
```tsx
<Camera zoom={1.5} {...otherProps} />
```

Then the preview
```

### Relevant log output

```shell
[unknown/unknown: Error Domain=AVFoundationErrorDomain Code=-11819 "Cannot Complete Action" UserInfo={NSLocalizedDescription=Cannot Complete Action, NSLocalizedRecoverySuggestion=Try again later.}]
```

### Camera Device

```json
{
  "supportsLowLightBoost": false,
  "isMultiCam": false,
  "hasTorch": true,
  "id": "com.apple.avfoundation.avcapturedevice.built-in_video:0",
  "sensorOrientation": "landscape-right",
  "hasFlash": true,
  "name": "Back Camera",
  "minZoom": 1,
  "supportsFocus": true,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsRawCapture": false,
  "neutralZoom": 1,
  "maxZoom": 121.875,
  "hardwareLevel": "full",
  "position": "back"
}
```

### Device

iPhone 13 (iOS 16.6.1)

### VisionCamera Version

3.6.8

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

bglgwyng, 2023-12-21:
#2310 #2280 These issues can have the same cause. Please check if you described the camera format with specified resolution vaules.

mrousavy, 2024-01-09:
Lol, this is a fun one. I'll dig around the Apple forums to see why this is happening after I fix the Preview stretching issue - thanks for the bug report @bglgwyng !

bglgwyng, 2024-01-20:
Yes, the problem still exists.

mrousavy, 2024-01-22:
From [a post in the apple developer forums](https://forums.developer.apple.com/forums/thread/679862):

[image]

mrousavy, 2024-01-22:
And [someone on Stackoverflow said](https://stackoverflow.com/a/29786924/5281431) this might be the `stopRunning` method timing out, but we are not using that in this scenario I think (that's why the logs would've been useful and why I always ask for logs btw 😉) but yea I think we're not using stopRunning here right?

bglgwyng, 2024-01-22:
@mrousavy Is this reported behavior reproduced on your device?
I just tested it and I found that my previous description was a bit misleading.
There's no crash and no black screen problem, but the frame rates get so low(about 1 frame in 3 seconds).

mrousavy, 2024-01-22:
No, I haven't seen that issue before - sorry :/ 

Let's keep it open tho, but honestly not sure if there's anything I can do.

frenbergFNX, 2024-02-08:
> Lol, this is a fun one. I'll dig around the Apple forums to see why this is happening after I fix the Preview stretching issue - thanks for the bug report @bglgwyng !

Hi @mrousavy and a big thanks for the amazing job on this lib

I see you mention a Preview stretching issue that I cant find any issues about but am experiencing too on 3.8.2. Do you have any progress on that?

mrousavy, 2024-02-08:
Yes, there are like 6 issues about that and in the latest 3.9. beta release I fixed it.

mrousavy, 2024-04-22:
Hey - I think this issue has been fixed in [VisionCamera 4.0.0](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v4.0.0). 🥳 

Please try V4 and let me know if you still experience this issue;
- if not, please consider [💖 **sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy/) to support the development of VisionCamera and thank me for my time spent on fixing bugs and building new features.
- if you still see this issue, please comment and we can re-open this. But please update your native logs with the native (Xcode/Android Studio) logs from running VisionCamera V4 so I can investigate this.

closed 2024-04-22 by mrousavy