margelo/react-native-vision-camera #2314: 🐛 App crashes when trying to open the camera multiple times

abdelrahman-muntaser, 2023-12-21:
### What's happening?

I am using the camera with a frame processor for scanning document, my camera code is inside a modal and on camera open i am setting the modal to visible and sending the visible to the camera, it works for the first time and second time then it crash the application 

### Reproduceable Code

```tsx
const device = useCameraDevice(cameraType);
  if (device == null) {
    return (
      <View style={styles.loadingView}>
        <Text style={styles.loadingText}>Loading</Text>
      </View>
    );
  }
  return (
    <Camera
      device={device}
      isActive={isVisible}
      zoom={zoom}
      style={EStyleSheet.absoluteFill}
      frameProcessor={frameProcessor}
      ref={ref}
      pixelFormat={'yuv'}
      foc
      torch={isVisible ? torch : 'off'}>
      {props.children}
    </Camera>
```

### Relevant log output

```shell
2023-12-21 14:20:26.313 32608-349   libc++abi               com.tahaluf.idify                    E  terminating with uncaught exception of type std::runtime_error: Unable to retrieve jni environment. Is the thread attached?
2023-12-21 14:20:26.313 32608-349   libc                    com.tahaluf.idify                    A  Fatal signal 6 (SIGABRT), code -1 (SI_QUEUE) in tid 349 (hades), pid 32608 (m.tahaluf.idify)
```

### Camera Device

```json
{
  "sensorOrientation": "landscape-left",
  "hardwareLevel": "full",
  "maxZoom": 8,
  "minZoom": 1,
  "maxExposure": 200,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": true,
  "isMultiCam": false,
  "minExposure": -200,
  "name": "FRONT (1)",
  "hasFlash": false,
  "hasTorch": false,
  "position": "front",
  "id": "1"
}
```

### Device

Galaxy S23

### VisionCamera Version

3.6.16

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-01-05:
Hey! Thanks for reporting this issue.

I've been working the past days on making sure the Camera lifecycle is safely handled and released when needed, so this crash doesn't occur anymore.

I just created a PR for this - can you please this to see if that fixes the issue for you? https://github.com/mrousavy/react-native-vision-camera/pull/2339

If this fixes your issue, please consider [💖 **sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy/) to support me / thank me for building VisionCamera and continuously improving it. 

If this does not fix your issue, please clone the repo, check out the branch `fix/blackscreen` (the PR above), run the Example app, reproduce your issue there and share the `adb logcat` logs with me (upload to pastebin or gist) so I can investigate this further.

Thank you! 🙏

tytom2003, 2024-01-08:
> Hey! Thanks for reporting this issue.
>
> I've been working the past days on making sure the Camera lifecycle is safely handled and released when needed, so this crash doesn't occur anymore.
>
> I just [...]

In expo, can solve the camera crash issue?

Ashutosh-Tiwari-Ukh, 2024-01-08:
@mrousavy I'm facing this app crash randomly even when the camera opens for the first time. But it's happening specifically in one tester's device and I can't somehow fetch the error logs out of it. I am hoping that the fix of this issue will solve my crash as well.

closed 2024-01-08 by mrousavy

Ashutosh-Tiwari-Ukh, 2024-01-08:
@mrousavy No I haven't but the fixes list does mention the issue I am facing. When can I expect a release? I am currently using latest version i.e. `3.1.17`.

mrousavy, 2024-01-08:
Please test the change. Download the repo and swap out your `android/` folders. I'm close to one, but I'm not an actual magician - I don't know if the issue is fixed on your end with my changes if I don't have any logs or repros.