margelo/react-native-vision-camera #2316: 🐛 (Android) NullPointerException -  com.mrousavy.camera.core.VideoPipeline in removeRecordingSessionOutputSurface

lucasqrib, 2023-12-21:
### What's happening?

Some users are facing a NullPointerExcpetion crash in our production app.
It's not constantly crashing, so I could not identify a pattern to reproduce the issue.

### Reproduceable Code

```tsx
// Not the whole component, but I took the relevant part out I believe
const Component = () => {
  const { hasPermission, requestPermission } = useCameraPermission();
  const device = useCameraDevice('back');
  const isFocused = useIsFocused();
  const appState = useAppState();
  const isCameraActive = isFocused && appState === 'active';

  const frameProcessor = useFrameProcessor(
    createOnFrameReceiveHandler(),
    [],
  );

  //... useEffect to request the permission

  // Use an empty component if the device is null or doesn't have permission
  if (!device || !hasPermission) {
    return <View style={tw('flex-1 bg-neutral-950')} />;
  }

  return (
    <View>
      {device && (
        <Camera
          style={tw('flex-1')}
          device={device}
          isActive={isCameraActive}
          frameProcessor={frameProcessor}
          pixelFormat="yuv"
          onError={onCameraError}
        />
      )}
    </View>
  );
};

const createOnFrameReceiveHandler = () => {
  const setScanResult = Worklets.createRunInJsFn(jsSetScanResult);
  return (frame: Frame) => {
    'worklet';
    const data = scanOCR(frame);
    const result = data?.result?.blocks?.map(block => block.text);

    setScanResult(result);
  };
};

const onCameraError = (error: any) => {
  console.error(error);
};
```

### Relevant log output

```shell
java.lang.NullPointerException: java.lang.NullPointerException
    at com.mrousavy.camera.core.VideoPipeline.removeRecordingSessionOutputSurface(VideoPipeline.kt)
    at com.mrousavy.camera.core.VideoPipeline.setRecordingSessionOutput(VideoPipeline.kt:60)
    at com.mrousavy.camera.core.CameraSession.updateVideoOutputs(CameraSession.kt:28)
    at com.mrousavy.camera.core.CameraSession.setFrameProcessor(CameraSession.kt:3)
    at com.mrousavy.camera.CameraView.setFrameProcessor$react_native_vision_camera_release(CameraView.kt:5)
    at com.mrousavy.camera.frameprocessor.VisionCameraProxy.setFrameProcessor$lambda$0(VisionCameraProxy.kt:15)
    at com.mrousavy.camera.frameprocessor.VisionCameraProxy.b(VisionCameraProxy.kt:1)
    at com.mrousavy.camera.frameprocessor.c.run(R8$$SyntheticClass:7)
    at android.os.Handler.handleCallback(Handler.java:942)
    at android.os.Handler.dispatchMessage(Handler.java:99)
    at android.os.Looper.loopOnce(Looper.java:226)
    at android.os.Looper.loop(Looper.java:313)
    at android.app.ActivityThread.main(ActivityThread.java:8762)
    at java.lang.reflect.Method.invoke(Method.java)
    at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:604)
    at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1067)
```

### Camera Device

```json
It only happened on the production app, so I could not get it :/
```

### Device

SM-G781V (Android 13), SM-S908U1 (Android 13), SM-G781V (Android 13), moto g(7) optimo maxx(XT1955DL)(Android 10)

### VisionCamera Version

3.6.17

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-01-29:
Hey - hmm not sure about this, I need the full logs to be able to help you.

Maybe worth a try, this could fix it: https://github.com/mrousavy/react-native-vision-camera/pull/2453

lucasqrib, 2024-01-29:
@mrousavy thanks for the reply! 
The full logs are a bit tricky since I could not find a way to reproduce it locally.
But your PR may fix it indeed because, for right now, I just patched the library adding a try-catch in this exact spot, and my crash rate dropped.

closed 2024-01-29 by mrousavy