margelo/react-native-vision-camera #2318: 🐛 App crashes on launch  after obfuscated by dexguard in Android

burakakyol, 2023-12-22:
### What's happening?

We're using the camera for scanning the barcode. We had no error while running on debug and release mode and the application worked as expected when the obfuscation was disabled. However, after obfuscating the application by using dexguard(enterprise proguard) , the application started to crash on splash screen. I couldn't find any keep rule in the documentation. 

### Reproduceable Code

```tsx
const device = useCameraDevice('back');

   device && (
            <View style={{ flex: 1 }}>
              <Camera
                testID="rncamera-view"
                fps={30}
                codeScanner={codeScanner}
                device={device}
                style={StyleSheet.absoluteFill}
                isActive={true}
              />
              <BarcodeMask
                outerMaskOpacity={0.5}
                lineAnimationDuration={1200}
                width={Dimensions.get('screen').width * 0.7}
                height={Dimensions.get('screen').height * 0.25}
              />
              {overlay && barCode === '' && (
                <CameraOverlay>
                  <Image
                    source={warningImage}
                    style={{ width: 40, height: 40, flexDirection: 'column', marginRight: 20 }}
                  />
                  <CameraOverlayText>{t('overlay')}</CameraOverlayText>
                </CameraOverlay>
              )}
            </View>
```

### Relevant log output

```shell
2023-12-22 09:29:53.334 28312-28312 SoLoader                com.reactnativeproject.test           V  libjscexecutor.so not found on /data/data/com.reactnativeproject.test/lib-main
2023-12-22 09:29:53.334 28312-28312 SoLoader                com.reactnativeproject.test           V  libjscexecutor.so not found on /data/app/~~YWbcVz0p-0omcS7eIIwWqg==/com.reactnativeproject.test-Ypvl8TOzMBYvTEKDBqMl-Q==/lib/arm64
2023-12-22 09:29:53.334 28312-28312 SoLoader                com.reactnativeproject.test           V  libjscexecutor.so not found on /system/lib64
2023-12-22 09:29:53.334 28312-28312 SoLoader                com.reactnativeproject.test           V  libjscexecutor.so not found on /vendor/lib64
2023-12-22 09:29:53.336 28312-28312 SoLoader                com.reactnativeproject.test           E  couldn't find DSO to load: libjscexecutor.so
                                                                                                    	SoSource 0: com.facebook.soloader.ApkSoSource[root = /data/data/com.reactnativeproject.test/lib-main flags = 1]
                                                                                                    	SoSource 1: com.facebook.soloader.DirectorySoSource[root = /data/app/~~YWbcVz0p-0omcS7eIIwWqg==/com.reactnativeproject.test-Ypvl8TOzMBYvTEKDBqMl-Q==/lib/arm64 flags = 0]
                                                                                                    	SoSource 2: com.facebook.soloader.DirectorySoSource[root = /system/lib64 flags = 2]
                                                                                                    	SoSource 3: com.facebook.soloader.DirectorySoSource[root = /vendor/lib64 flags = 2]
                                                                                                    	Native lib dir: /data/app/~~YWbcVz0p-0omcS7eIIwWqg==/com.reactnativeproject.test-Ypvl8TOzMBYvTEKDBqMl-Q==/lib/arm64
                                                                                                     result: 0
2023-12-22 09:29:53.338 28312-28312 SoLoader                com.reactnativeproject.test           V  libhermes.so not found on /data/data/com.reactnativeproject.test/lib-main
2023-12-22 09:29:53.338 28312-28312 SoLoader                com.reactnativeproject.test           D  libhermes.so found on /data/app/~~YWbcVz0p-0omcS7eIIwWqg==/com.reactnativeproject.test-Ypvl8TOzMBYvTEKDBqMl-Q==/lib/arm64
2023-12-22 09:29:53.338 28312-28312 SoLoader                com.reactnativeproject.test           D  Not resolving dependencies for libhermes.so
2023-12-22 09:29:53.387 28312-28312 SoLoader                com.reactnativeproject.test           V  libhermes_executor.so not found on /data/data/com.reactnativeproject.test/lib-main
2023-12-22 09:29:53.387 28312-28312 SoLoader                com.reactnativeproject.test           D  libhermes_executor.so found on /data/app/~~YWbcVz0p-0omcS7eIIwWqg==/com.reactnativeproject.test-Ypvl8TOzMBYvTEKDBqMl-Q==/lib/arm64
2023-12-22 09:29:53.387 28312-28312 SoLoader                com.reactnativeproject.test           D  Not resolving dependencies for libhermes_executor.so
2023-12-22 09:29:53.790 28312-30992 SoLoader                com.reactnativeproject.test           V  libfbjni.so not found on /data/data/com.reactnativeproject.test/lib-main
2023-12-22 09:29:53.790 28312-30992 SoLoader                com.reactnativeproject.test           V  libfbjni.so not found on /data/app/~~YWbcVz0p-0omcS7eIIwWqg==/com.reactnativeproject.test-Ypvl8TOzMBYvTEKDBqMl-Q==/lib/arm64
2023-12-22 09:29:53.791 28312-30992 SoLoader                com.reactnativeproject.test           V  libfbjni.so not found on /system/lib64
2023-12-22 09:29:53.791 28312-30992 SoLoader                com.reactnativeproject.test           V  libfbjni.so not found on /vendor/lib64
2023-12-22 09:29:53.793 28312-30992 SoLoader                com.reactnativeproject.test           E  couldn't find DSO to load: libfbjni.so
                                                                                                    	SoSource 0: com.facebook.soloader.ApkSoSource[root = /data/data/com.reactnativeproject.test/lib-main flags = 1]
                                                                                                    	SoSource 1: com.facebook.soloader.DirectorySoSource[root = /data/app/~~YWbcVz0p-0omcS7eIIwWqg==/com.reactnativeproject.test-Ypvl8TOzMBYvTEKDBqMl-Q==/lib/arm64 flags = 0]
                                                                                                    	SoSource 2: com.facebook.soloader.DirectorySoSource[root = /system/lib64 flags = 2]
                                                                                                    	SoSource 3: com.facebook.soloader.DirectorySoSource[root = /vendor/lib64 flags = 2]
                                                                                                    	Native lib dir: /data/app/~~YWbcVz0p-0omcS7eIIwWqg==/com.reactnativeproject.test-Ypvl8TOzMBYvTEKDBqMl-Q==/lib/arm64
                                                                                                     result: 0
2023-12-22 09:29:53.796 28312-30992 AndroidRuntime          com.reactnativeproject.test           E  FATAL EXCEPTION: create_react_context
                                                                                                    Process: com.reactnativeproject.test, PID: 28312
[... 43 lines cut ...]
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Redmi Note 11 ( Android 11), Galaxy A32 ( Android 11)

### VisionCamera Version

3.6.13

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-12-22:
Hey - sorry I don't really think this is related to VisionCamera.

Maybe try removing this line: https://github.com/mrousavy/react-native-vision-camera/blob/7905f2c057869698a4f5afa3182bec4cd2a5fe58/package/android/build.gradle#L145

mrousavy, 2024-01-08:
Hey - did you try to remove this line? If yes, did it work?

burakakyol, 2024-01-08:
Hi @mrousavy 
I'm sorry for not getting back to you sooner.

We removed the line you mentioned, but it didn't work. I'll get back to you tomorrow with the details.

Thank you.

mrousavy, 2024-01-09:
Okay thanks - let me know because this is weird haven't seen this before

burakakyol, 2024-01-10:
Hi @mrousavy

I removed the line you mentioned. However, I got a different crash from the application.

```
 couldn't find DSO to load: libjscexecutor.so
                                                                                                    	SoSource 0: com.facebook.soloader.ApkSoSource[root = /data/data/tr.sisal.streetvendor.test/lib-main flags = 1]
                                                                                                    	SoSource 1: com.facebook.soloader.DirectorySoSource[root = /data/app/~~jU4dZ_nfHW8CmZVGk6dvLA==/tr.sisal.streetvendor.test-OPAHM0LywtyiQJVkT_Q7UQ==/lib/arm64 flags = 0]
                                                                                                    	SoSource 2: com.facebook.soloader.DirectorySoSource[root = /system/lib64 flags = 2]
                                                                                                    	SoSource 3: com.facebook.soloader.DirectorySoSource[root = /vendor/lib64 flags = 2]
                                                                                                    	Native lib dir: /data/app/~~jU4dZ_nfHW8CmZVGk6dvLA==/tr.sisal.streetvendor.test-OPAHM0LywtyiQJVkT_Q7UQ==/lib/arm64
                                                                                                     result: 0

```

 libjscexecutor.so is also different .so file that is placed in the gradle file of the library. Should I remove it ?

OmmAshutosh, 2024-01-10:
1) first install this in your device
npm install @react-native-camera/core @react-native-camera/barcode 
2)make sure you have linked the packages correctly. If you are using React Native 0.60 or above, auto-linking should work.
3)try this code according to your preferences;

import React, { useEffect, useState } from 'react';
import { StyleSheet, View, Dimensions, Image } from 'react-native';
import Camera from '@react-native-camera/core';
import BarcodeMask from '@react-native-camera/barcode-mask';
import { useCameraDevice } from '@react-native-camera/core';

const CameraScreen = () => {
 const device = useCameraDevice('back');
 const [barCode, setBarCode] = useState('');

 const handleBarCodeRead = (e) => {
    setBarCode(e.data);
 };

 useEffect(() => {
    if (barCode !== '') {
      // Your code here, for example:
      // navigate to a new screen and pass the barCode value
    }
 }, [barCode]);

 return (
    <View style={styles.container}>
      {device && (
        <Camera
          style={StyleSheet.absoluteFill}
          device={device}
          onBarCodeRead={handleBarCodeRead}
        >
          <BarcodeMask />
        </Camera>
      )}
      {barCode && (
        <View style={styles.overlay}>
          <Image
            source={require('./path/to/warning/image.png')}
            style={styles.warningImage}
          />
          <Text style={styles.warningText}>Barcode Scanned: {barCode}</Text>
        </View>
      )}
    </View>
 );
};

const styles = StyleSheet.create({
 container: {
    flex: 1,
 },
 overlay: {
    position: 'absolute',
    top: 0,
    left: 0,
    right: 0,
    bottom: 0,
    backgroundColor: 'rgba(0, 0, 0, 0.5)',
    justifyContent: 'center',
    alignItems: 'center',
 },
 warningImage: {
    width: 40,
    height: 40,
 },
 warningText: {
    color: 'white',
    fontSize: 16,
    marginTop: 20,
 },
});

export default CameraScreen;

closed 2024-01-30 by mrousavy

burakakyol, 2024-02-28:
We solved the issue by applying the dexguard rules below:

`-keepresourcefiles lib/**/libfb*.so,lib/**/libreact*.so,lib/**/libyoga*.so,lib/**/libhermes.so,lib/**/libhermes_executor.so,lib/**/libjsi*.so,*/lib/**/libfb*.so,*/lib/**/libreact*.so,*/lib/**/libyoga*.so,*/lib/**/libhermes.so,*/lib/**/libhermes_executor.so,*/lib/**/libjsi*.so`