margelo/react-native-vision-camera #2319: [V2] Crash on iPhone 15 Pro + iOS 17 (AVCaptureDeviceFormat.toDictionary()) 🐛

sommcz, 2023-12-22:
### What's happening?

I have reported and simulated crashes on iOS 17, only on iPhone 15 Pro affected (i've tested iphone 15, 12 and it works ok).

It looks similiar like https://github.com/mrousavy/react-native-vision-camera/issues/1840, crash is on different place.

I'm using Xcode14 (i had reverted back when V2 build was not possible on xcode 15 due to some errors) and RNVC V2.

Sadly, i don't have device available for debugging, i have just production crash reports and crash on BrowserStack real device cloud.

### Reproduceable Code

```tsx
const devices = useCameraDevices();
	const device = devices.back;

// ....

	return (
		<>
			{device && hasPermission && (
				<View style={styles.container}>
					{isForeground && (
						<Camera
							style={StyleSheet.absoluteFill}
							device={device}
							isActive={!disabled}
							frameProcessor={boundFrameProcessor}
							frameProcessorFps={frameProcessorFps}
							preset="medium"
							onLayout={handleLayoutChange}
						/>
					)}

....
```

### Relevant log output

```shell
Incident Identifier: A67BE188-5BF8-4B0B-979F-3378000178E4
Hardware Model:      iPhone16,1
Process:             MyRNVCApp [82017]
AppVariant:          1:iPhone15,2:16
Code Type:           ARM-64 (Native)
Role:                Foreground
Parent Process:      launchd [1]
Coalition:           com.MyRNVCApp [3039]

OS Version:          iPhone OS 17.0.3 (21A360)
Baseband Version:    1.00.03
Report Version:      104

Exception Type:  EXC_BREAKPOINT (SIGTRAP)
Exception Codes: 0x0000000000000001, 0x000000019472f86c
Termination Reason: SIGNAL 5 Trace/BPT trap: 5
Terminating Process: exc handler [82017]

Triggered by Thread:  0

Thread 0 name:
Thread 0 Crashed:
0   libswiftCore.dylib            	0x000000019472f86c _assertionFailure(_:_:file:line:flags:) + 264 (AssertCommon.swift:144)
1   MyRNVCApp                   	0x0000000104aea884 AVCaptureDeviceFormat.toDictionary() + 3016 (AVCaptureDevice.Format+toDictionary.swift:0)
2   MyRNVCApp                   	0x0000000104aff2ec closure #1 in closure #2 in closure #1 in CameraViewManager.getAvailableCameraDevices(_:reject:) + 4 (CameraViewManager.swift:136)
3   MyRNVCApp                   	0x0000000104aff2ec specialized Collection.map<A>(_:) + 112 (<compiler-generated>:0)
4   MyRNVCApp                   	0x0000000104aff2ec closure #2 in closure #1 in CameraViewManager.getAvailableCameraDevices(_:reject:) + 1816 (CameraViewManager.swift:135)
5   MyRNVCApp                   	0x0000000104afea48 specialized Collection.map<A>(_:) + 128 (<compiler-generated>:0)
6   MyRNVCApp                   	0x0000000104afea48 closure #1 in CameraViewManager.getAvailableCameraDevices(_:reject:) + 968
7   MyRNVCApp                   	0x0000000104afea48 0x10470c000 + 4139592 (CameraViewManager.swift:118)
8   MyRNVCApp                   	0x0000000104b0483c specialized withPromise(_:_:) + 48 (Promise.swift:52)
9   MyRNVCApp                   	0x0000000104aff560 specialized withPromise(_:_:) + 12 (CameraViewManager.swift:68)
10  MyRNVCApp                   	0x0000000104aff560 CameraViewManager.getAvailableCameraDevices(_:reject:) + 12 (<compiler-generated>:0)
11  MyRNVCApp                   	0x0000000104aff560 @objc CameraViewManager.getAvailableCameraDevices(_:reject:) + 196
12  CoreFoundation                	0x000000019b2b7134 __invoking___ + 148 (:-1)
```

### Camera Device

```json
{}
```

### Device

iPhone 15 Pro

### VisionCamera Version

2.16.1

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

billoneillighttiger, 2023-12-23:
I have got the same issue. react-native-vision-camera version: 3.6.16

robertgotitapp, 2023-12-26:
Same issue logged in [1840](https://github.com/mrousavy/react-native-vision-camera/issues/1840)

djmaluy, 2023-12-27:
same problem

SHINGPH, 2024-01-18:
i also got an issue with this it's crashing on 15 pro/max version @mrousavy any solutions to this?

edritech93, 2024-01-19:
same issue for me, crash on iPhone 15 ProMax
using react-native-vision-camera: 2.16.6

Ardazafer, 2024-01-19:
same issue here, i'm using v2.15.2

reubber, 2024-01-22:
same issue ;/

unicornRainbows, 2024-01-24:
Same issue here, please help someone. :)

mrousavy, 2024-01-30:
> Same issue here, please help someone. :)

> same issue ;/

> same issue for me, crash on iPhone 15 ProMax

> i also got an issue with this it's crashing on 15 pro/max version @mrousavy any solutions to this?

> same

> I have got the same issue. react-native-vision-camera version: 3.6.16

please stop spamming - if an issue is that important for you to get it fixed, consider sponsoring me on GitHub to pay me for fixing an issue for you.

---

Thanks @Ardazafer for the sponsorship on GitHub!

I've gone ahead and ran the Example app on my iPhone 15 Pro on the current `v2` branch, and it starts perfectly fine. Are there any ways to reproduce this issue?

mrousavy, 2024-01-30:
Did this fix the issue for you guys? https://github.com/mrousavy/react-native-vision-camera/commit/503114b9d1c61d60fa3c69b3bd4cb084758fbf5b

mrousavy, 2024-01-30:
I'm pretty sure that fixed the issue in 2.16.7, if not we can still re-open the issue @Ardazafer.

closed 2024-01-30 by mrousavy

Ardazafer, 2024-02-24:
@mrousavy Yeap that fixed the issue, thanks for the good work man!

gaganpreet-scalex, 2024-03-18:
Any update on this issue?

vantran-agilityio, 2024-12-12:
Happing til now