margelo/react-native-vision-camera #2324: Getting camera details not working

jigarsolanki0212, 2023-12-28:
### Question

**When I try to get the camera details, it does not work on Android, but it works fine on iOS**

- I am using it to make the camera recording ultra-wide angle and normal,
- for that, I use min zoom as an ultra-wide angle and neutral zoom as a normal camera.
- it works fine for IOS devices but I try on Android if the device has multiple cameras then still it gives me only one camera device.

**please respond to me as soon as possible, as I have to submit this work soon**

### What I tried

 const devices = useCameraDevices();
    const deviceAndroid = switchCameraToFront ? devices?.front : devices.back;
    const isWideAngleCamera = deviceAndroid?.devices.includes('ultra-wide-angle-camera');
    
    

### VisionCamera Version

"react-native-vision-camera": "^2.16.2",

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

Jigar2311, 2024-01-01:
@mrousavy  ?

closed 2024-01-01 by mrousavy

Jigar2311, 2024-01-03:
@mrousavy i still can't find the solution to manage ultra wide angle camera for android i have done it for ios but for android i can't understand what to do can you please help.