margelo/react-native-vision-camera #2325: 🐛  Camera.onError(unknown/unknown):  outputs must be a non-null, non-empty list

tytom2003, 2023-12-29:
### What's happening?

when i run my program, i get " Camera.onError(unknown/unknown): [unknown/unknown] outputs must be a non-null, non-empty list [unknown/unknown: [unknown/unknown] outputs must be a non-null, non-empty list]". i use "Camera.requestCameraPermission();". i got "granted", but i use "const {hasPermission, requestPermission} = useCameraPermission();". hasPermission get false. My camera is get black camera. How to solve the issue? i tried use react-native-vision-camera 3.6.17 and 3.6.16. i still get black screen. 
i hope my program can open camera. so, i can add other code in my program.
Thank you very much.

### Reproduceable Code

```tsx
below is my program 
export default function Barcode() {
  const device = useCameraDevice('back');
  //const device = devices.back;
  const {hasPermission, requestPermission} = useCameraPermission();

  useEffect(() => {
    getpremission();
  }, []);

  const getpremission = async () => {
    const status = await Camera.requestCameraPermission();
    if (status === 'denied') {
      console.log('permission not granted');
    }

    if (device === null) {
      console.log('null');
      return <NoCameraDeviceError />;
    }

    console.log('54: ' + JSON.stringify(status));
    console.log('55: ' + JSON.stringify(hasPermission));
    if (hasPermission === false) {
      console.log('false');
      requestPermission();
    }
  };

  // eslint-disable-next-line react-hooks/rules-of-hooks
  const onError = useCallback((error: CameraRuntimeError) => {
    console.error(error);
  }, []);

  return (
    <>
      <Camera style={StyleSheet.absoluteFill} device={device} isActive={true} />
    </>
  );
}
```

### Relevant log output

```shell
ERROR  Camera.onError(unknown/unknown): [unknown/unknown] outputs must be a non-null, non-empty list [unknown/unknown: [unknown/unknown] outputs must be a non-null, non-empty list]
```

### Camera Device

```json
import {
  Camera,
  useCameraDevice,
  NoCameraDeviceError,
  useCodeScanner,
  useCameraPermission,
} from 'react-native-vision-camera';
const device = useCameraDevice('back'); 
<Camera style={StyleSheet.absoluteFill} device={device} isActive={true} />
```

### Device

Android

### VisionCamera Version

3.6.17 and 3.6.16

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

ikbalkazanc, 2024-01-03:
🙋🏻‍♂️

tytom2003, 2024-01-04:
> 🙋🏻‍♂️

I solved the black screen when i open apps at first time. 

reference: (discord - SteelMan)
Hi, I've not the code here, but my code was something like:

{device &&
  <Camera
    device={device}
    isActive={true}
  />
}

and I got black screen.

After I tried something like this:

const [isActive, setActive] = useState(false);
useEffect(() => {
  if (hasPermission) {
    setTimeout(() => setActive(true), 500); // I'm not sure "setTimeout" is needed
  }
}, [hasPermission]);

return device && (
  <Camera
    device={device}
    isActive={isActive}
  />
)

closed 2024-01-04 by tytom2003