margelo/react-native-vision-camera #2328: "code": "capture/video-not-enabled", "message": "[capture/video-not-enabled] Video capture is disabled! Pass `video={true}` to enable video recordings.

remorr25, 2023-12-29:
### What's happening?

I have follow the document and integrate this package for video recording app. I found that whenever I start recording it is giving me below error.

"code": "capture/video-not-enabled", "message": "[capture/video-not-enabled] Video capture is disabled! Pass `video={true}` to enable video recordings.

I have added video={true} in the Camera component then also I getting same error again and again.

My package json mentioned below

{
   "name": "NewsClips",
   "version": "0.0.1",
   "private": true,
   "scripts": {
      "android": "react-native run-android",
      "ios": "react-native run-ios",
      "lint": "eslint .",
      "start": "react-native start",
      "test": "jest",
      "release": "cd android && ./gradlew assembleRelease",
      "brelease": "cd android && ./gradlew bundleRelease"
   },
   "dependencies": {
      "react": "18.2.0",
      "react-native": "0.73.1",
      "react-native-vision-camera": "^3.6.17"
   },
   "devDependencies": {
      "@babel/core": "^7.20.0",
      "@babel/preset-env": "^7.20.0",
      "@babel/runtime": "^7.20.0",
      "@react-native/babel-preset": "^0.73.18",
      "@react-native/eslint-config": "^0.73.1",
      "@react-native/metro-config": "^0.73.2",
      "@react-native/typescript-config": "^0.73.1",
      "@types/react": "^18.2.6",
      "@types/react-test-renderer": "^18.0.0",
      "babel-jest": "^29.6.3",
      "eslint": "^8.19.0",
      "jest": "^29.6.3",
      "prettier": "2.8.8",
      "react-test-renderer": "18.2.0",
      "typescript": "5.0.4"
   },
   "engines": {
      "node": ">=18"
   }
}

### Reproduceable Code

```tsx
function App(): React.JSX.Element {
  const camera = useRef(null);
  const [cameraPermission, setCameraPermission] = useState('');
  const [isCamActive, setCamActive] = useState(false);
  const [videoPath, setVideoPath] = useState();
  const [photoPath, setPhotoPath] = useState();

  const device = useCameraDevice('back');
 
  useEffect(() => {
    (async () => {
      const cameraPermissionStatus = await Camera.requestCameraPermission();
      setCameraPermission(cameraPermissionStatus);
    })();
  }, []);

  const handleRecordVideo = async () => {
    try {
      camera.current?.startRecording({
        onRecordingFinished: video => setVideoPath(video.path),
        onRecordingError: error => console.error(error),
      });
    } catch (e) {
      console.log(e);
    }
  };

  const handleStopVideo = async () => {
    try {
      await camera?.current?.stopRecording();
    } catch (e) {
      console.log(e);
    }
  };

  const initialize = () => {
    console.log('camera initialize');
    setCamActive(true);
  };

  if (device == null) return <ActivityIndicator />;
  if (cameraPermission == 'authorized') {
    return <Text>{cameraPermission}</Text>;
  }
  return (
    <View style={styles.screen}>
      <SafeAreaView style={styles.saveArea}>
        <View style={styles.header}>
          <Text style={styles.headerText}>React Native Camera Libraries</Text>
        </View>
      </SafeAreaView>

      <View style={styles.caption}>
        <Text style={styles.captionText}>
          Welcome To React-Native-Vision-Camera Tutorial
        </Text>
      </View>
      <View>
        <Camera
          ref={camera}
          style={[styles.camera, styles.photoAndVideoCamera]}
          device={device}
          onInitialized={initialize}
          isActive={isCamActive}
          video={true}
        />
        <TouchableOpacity style={styles.btn} onPress={handleRecordVideo}>
          <Text style={styles.btnText}>Start recording</Text>
        </TouchableOpacity>
      </View>
    </View>
  );
}

const styles = StyleSheet.create({
  screen: {
    flex: 1,
    backgroundColor: '#EEF2E6',
  },
  saveArea: {
    backgroundColor: '#3D8361',
  },
  header: {
    height: 50,
    backgroundColor: '#3D8361',
    alignItems: 'center',
    justifyContent: 'center',
  },
  headerText: {
    color: '#ffffff',
    fontSize: 20,
  },
  caption: {
    height: 100,
    justifyContent: 'center',
    alignItems: 'center',
  },
  captionText: {
    color: '#100F0F',
    fontSize: 16,
    fontWeight: '600',
  },
  camera: {
    height: 460,
    width: '92%',
    alignSelf: 'center',
  },
  photoAndVideoCamera: {
    height: 360,
  },
  barcodeText: {
    paddingHorizontal: 16,
    paddingVertical: 20,
    textAlign: 'center',
    color: '#100F0F',
    fontSize: 24,
  },
  pickerSelect: {
    paddingVertical: 12,
  },
  image: {
    marginHorizontal: 16,
    paddingTop: 8,
    width: 80,
    height: 80,
  },
  dropdownPickerWrapper: {
    paddingHorizontal: 16,
    paddingBottom: 16,
    zIndex: 9,
  },
  btnGroup: {
    margin: 16,
    flexDirection: 'row',
  },
  btn: {
    backgroundColor: '#63995f',
    margin: 13,
    paddingHorizontal: 20,
    paddingVertical: 16,
    borderRadius: 8,
  },
  btnText: {
    color: '#ffffff',
    fontSize: 20,
    textAlign: 'center',
  },
  video: {
    marginHorizontal: 16,
    height: 100,
    width: 80,
    position: 'absolute',
    right: 0,
    bottom: -80,
  },
});
```

### Relevant log output

```shell
No log available
```

### Camera Device

```json
{
  "sensorOrientation": "landscape-right",
  "hardwareLevel": "full",
  "maxZoom": 10,
  "minZoom": 1,
  "maxExposure": 144,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": true,
  "isMultiCam": false,
  "minExposure": -144,
  "name": "BACK (0)",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Redmi note 10s

### VisionCamera Version

3.6.17

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-01-02:
Are you sure you have waited until the Camera is initialized? This error can never occur unless you really don't enable `video={true}`..

remorr25, 2024-01-02:
@mrousavy Yes I tested many times with multiple device while recording the video, I face same issue.
For capturing photo I didn't face this issue.

axeleschholz, 2024-01-03:
I am running into an identical issue, but with `photo={true}`. Pulling the props on the camera ref immediately before taking the picture confirms that the `photo` prop is set to true. But the promise doesn't resolve and throws the `capture/photo-not-enabled` error, advising passing `photo={true}` to the Camera element to enable capture.

This error is intermittent and seemingly random. However, when it will occur, the camera often displays as blurry.

mrousavy, 2024-01-05:
Hey! Thanks for reporting this issue.

I've been working the past days on making sure the Camera lifecycle is safely handled and released when needed, so this crash doesn't occur anymore.

I just created a PR for this - can you please this to see if that fixes the issue for you? https://github.com/mrousavy/react-native-vision-camera/pull/2339

If this fixes your issue, please consider [💖 **sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy/) to support me / thank me for building VisionCamera and continuously improving it. 

If this does not fix your issue, please clone the repo, check out the branch `fix/blackscreen` (the PR above), run the Example app, reproduce your issue there and share the `adb logcat` logs with me (upload to pastebin or gist) so I can investigate this further.

Thank you! 🙏

closed 2024-01-08 by mrousavy