margelo/react-native-vision-camera #2330: 🐛 (Android) Camera works on initial build and view blacks out on refresh

czack9131, 2023-12-31:
### What's happening?

This is day 2 of trying to make this library work with Expo. I couldn't figure out why, even with the example code provided in the tutorial/docs, my app would request permissions and the camera would display a preview, but when I refreshed my codebase the camera would cease to function. The preview would go black and all of the camera action hooks threw device not found errors. It took clearing the expo cache and rebuilding and reinstalling the app for the camera to function again. The cycle repeats...

I eventually created a dummy project with only the base Expo installs and version ^3.6.17 of RNVC. My App.tsx component held only the base elements/hooks and the problem persisted. No permutation of when/how I called the hooks influenced the camera behavior. No errors were reported by the camera component, just no output. I downgraded RNVC to 3.6.16 as suggested for a tangentially related issue and the same code worked perfectly with none of the aforementioned issues. 

My theory is that when the codebase was refreshed, the camera resource was never released (hence why the camera would work only the first time the app was opened on a fresh install). I also noticed that the little green led that appears on the top right corner of Galaxy phones when the camera resource is in use only appeared on that first run. 

Please let me know if I've missed something completely obvious.

Using: 
- Expo 49.0.21
- RNVV 3.6.17 (Issue)
- RNVC 3.6.16 (Working)

### Reproduceable Code

```tsx
export default function App() {
  const cameraRef = React.useRef<Camera>(null);
  const [cameraPermissionStatus, setCameraPermissionStatus] =
    React.useState<CameraPermissionStatus>("not-determined");
  const device = useCameraDevice("front");

  const requestCameraPermission = React.useCallback(async () => {
    console.log("Requesting camera permission...");
    const permission = await Camera.requestCameraPermission();
    console.log(`Camera permission status: ${permission}`);

    setCameraPermissionStatus(permission);
  }, []);

  React.useEffect(() => {
    if (cameraPermissionStatus !== "granted") {
      requestCameraPermission();
    }
  }, [cameraPermissionStatus]);

  return (
    <View style={{flex:1}}>
      <Camera
        ref={cameraRef}
        device={device!}
        isActive={true}
        style={StyleSheet.absoluteFill}
        onError={console.error}
      />
    </View>
  );
}
```

### Relevant log output

```shell
› Reloading apps
Android Bundling complete 45ms
 LOG  Requesting camera permission...
 LOG  Camera permission status: granted
```

### Camera Device

```json
{
  hardwareLevel: "full",
  hasFlash: false,
  hasTorch: false,
  id: "1",
  isMultiCam: false,
  maxExposure: 200,
  maxZoom: 8,
  minExposure: -200,
  minZoom: 1,
  name: "FRONT (1)",
  neutralZoom: 1,
  physicalDevices: ["wide-angle-camera"],
  position: "front",
  sensorOrientation: "landscape-left",
  supportsFocus: true,
  supportsLowLightBoost: false,
  supportsRawCapture: true
}
```

### Device

Samsung Galaxy S22+

### VisionCamera Version

3.6.17

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [X] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

remorr25, 2024-01-01:
@czack9131  please set `isActive={true}` by use some state and set the state on onInitialized prop method like below:-

export default function App() {
  const cameraRef = React.useRef<Camera>(null);
  const [isCamActive, setCamActive] = React.useState(false);
  const [cameraPermissionStatus, setCameraPermissionStatus] =
    React.useState<CameraPermissionStatus>("not-determined");
  const device = useCameraDevice("front");

  const requestCameraPermission = React.useCallback(async () => {
    console.log("Requesting camera permission...");
    const permission = await Camera.requestCameraPermission();
    console.log(`Camera permission status: ${permission}`);

    setCameraPermissionStatus(permission);
  }, []);

  React.useEffect(() => {
    if (cameraPermissionStatus !== "granted") {
      requestCameraPermission();
    }
  }, [cameraPermissionStatus]);
  
  const initialize = () => {
    console.log('camera initialize');
    setCamActive(true);
  };

  return (
    <View style={{flex:1}}>
      <Camera
        ref={cameraRef}
        device={device!}
        isActive={true}
        onInitialized={initialize}
          isActive={isCamActive}
        style={StyleSheet.absoluteFill}
        onError={console.error}
      />
    </View>
  );
}

mrousavy, 2024-01-05:
Hey! Thanks for reporting this issue.

I've been working the past days on making sure the Camera lifecycle is safely handled and released when needed, so this crash doesn't occur anymore.

I just created a PR for this - can you please this to see if that fixes the issue for you? https://github.com/mrousavy/react-native-vision-camera/pull/2339

If this fixes your issue, please consider [💖 **sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy/) to support me / thank me for building VisionCamera and continuously improving it. 

If this does not fix your issue, please clone the repo, check out the branch `fix/blackscreen` (the PR above), run the Example app, reproduce your issue there and share the `adb logcat` logs with me (upload to pastebin or gist) so I can investigate this further.

Thank you! 🙏

closed 2024-01-08 by mrousavy

faizan72560, 2024-05-06:
video not getting stopped when recording is started in version 3.6.16

faizan72560, 2024-05-06:
can you suggest me any version in which all the things is working fine