margelo/react-native-vision-camera #2333: ✨ is ther way to auto zoom to a specific QR, IOS native camera is able to do that

rajatgupta2123, 2024-01-01:
### What feature or enhancement are you suggesting?

is therw way to auto zoom to a specific QR, IOS native camera is able to do that 

### What Platforms whould this feature/enhancement affect?

iOS, Android

### Alternatives/Workarounds

-

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

jslok, 2024-01-02:
Use the qr/barcode scanner frameprocessor. It gives you the location and size of the qr code and you can use that to zoom.

mrousavy, 2024-01-02:
exactly what @jslok said. I don't think the Camera itself should do that, that should be your application's responsibility.

closed 2024-01-02 by mrousavy