margelo/react-native-vision-camera #2338: How to record video from a front camera without a mirror❓

HorbachAndrii, 2024-01-02:
### Question

I am analyzing recorded video and for me, it matters to record video without a mirror.

For IOS, I added some hack. I changed this `connection.isVideoMirrored = false` (from true to false) at `node_modules/react-native-vision-camera/ios/Extensions/AVCaptureOutput+mirror.swift`.

But for Android, I don't know how to record video from a front camera without a mirror.

Maybe somebody knows how can I record video from a front camera without a mirror?

### What I tried

I changing the Matrix for transformMatrix in VideoPipeline.kt and other stuff, but nothing helped

### VisionCamera Version

3.6.16

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

HorbachAndrii, 2024-01-04:
So I added also some hack that works for android.

VideoPipeline.kt `node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/VideoPipeline.kt` 

- update this line https://github.com/mrousavy/react-native-vision-camera/blob/ef4e9fabf9410dc5630dee8220f281089c941495/package/android/src/main/java/com/mrousavy/camera/core/VideoPipeline.kt#L165
`onFrame(transformMatrix)` -> `onFrame(transformMatrix, isMirrored)`

- update this line https://github.com/mrousavy/react-native-vision-camera/blob/ef4e9fabf9410dc5630dee8220f281089c941495/package/android/src/main/java/com/mrousavy/camera/core/VideoPipeline.kt#L212
`private external fun onFrame(transformMatrix: FloatArray)` 
->
 `private external fun onFrame(transformMatrix: FloatArray, isMirrored: Boolean)`

VideoPipeline.cpp `node_modules/react-native-vision-camera/android/src/main/cpp/VideoPipeline.cpp`

- update this line 
`void VideoPipeline::onFrame(jni::alias_ref<jni::JArrayFloat> transformMatrixParam) {`
->
`void VideoPipeline::onFrame(jni::alias_ref<jni::JArrayFloat> transformMatrixParam, bool isMirrored) {`

VideoPipeline.h `node_modules/react-native-vision-camera/android/src/main/cpp/VideoPipeline.h`
 `update this line void onFrame(jni::alias_ref<jni::JArrayFloat> transformMatrix);`
 ->
  `void onFrame(jni::alias_ref<jni::JArrayFloat> transformMatrix, bool isMirrored);`

closed 2024-01-04 by HorbachAndrii

CyxouD, 2024-04-30:
It doesn't work in my case on iOS with the same react-native-vision-camera. Could you say what iOS device you use? Also, could you send your Camera component JSX (if something differs here)?

HorbachAndrii, 2024-04-30:
> It doesn't work in my case on iOS with the same react-native-vision-camera. Could you say what iOS device you use? Also, could you send your Camera component JSX (if something differs here)?

What VisionCamera Version are you using? 

> Could you say what iOS device you use?
it works for all IOS devices

CyxouD, 2024-04-30:
@HorbachAndrii thank you for your quick response! 
> What VisionCamera Version are you using?
I tried with 3.6.16 (as you mentioned) and 3.9.2. I tested with iPhone 11.

HorbachAndrii, 2024-04-30:
here is my AVCaptureOutput+mirror.swift file

```
//
//  AVCaptureOutput+mirror.swift
//  mrousavy
//
//  Created by Marc Rousavy on 18.01.21.
//  Copyright © 2021 mrousavy. All rights reserved.
//

import AVFoundation

extension AVCaptureOutput {
  /**
   Mirrors the video output if possible.
   */
  func mirror() {
    connections.forEach { connection in
      if connection.isVideoMirroringSupported {
        connection.automaticallyAdjustsVideoMirroring = false
        connection.isVideoMirrored = false
      }
    }
  }

  /**
   Sets the target orientation of the video output.
   This does not always physically rotate image buffers.

   - For Preview, an orientation hint is used to rotate the layer/view itself.
   - For Photos, an EXIF tag is used.
   - For Videos, the buffers are physically rotated if available, since we use an AVCaptureVideoDataOutput instead of an AVCaptureMovieFileOutput.
   */
  func setOrientation(_ orientation: Orientation) {
    // Set orientation for each connection
    connections.forEach { connection in
      #if swift(>=5.9)
        if #available(iOS 17.0, *) {
          // Camera Sensors are always in landscape rotation (90deg).
          // We are setting the target rotation here, so we need to rotate by landscape once.
          let cameraOrientation = orientation.rotateBy(orientation: .landscapeLeft)
          let degrees = cameraOrientation.toDegrees()

          // TODO: Don't rotate the video output because it adds overhead. Instead just use EXIF flags for the .mp4 file if recording.
          //       Does that work when we flip the camera?
          if connection.isVideoRotationAngleSupported(degrees) {
            connection.videoRotationAngle = degrees
          }
        } else {
          if connection.isVideoOrientationSupported {
            connection.videoOrientation = orientation.toAVCaptureVideoOrientation()
          }
        }
      #else
        if connection.isVideoOrientationSupported {
          connection.videoOrientation = orientation.toAVCaptureVideoOrientation()
        }
      #endif
    }
  }
}

```

CyxouD, 2024-05-01:
> Did you 'Clean Build Folder' before building the app?
@HorbachAndrii Thank you! I tried it with 3.9.1 and 3.9.2, but unfortunately it didn't work. I didn't clean the build folder with 3.6.16 because it had another bugs, but I tested with logs that changes were reflected.
I will check if it works with other versions of iPhone (not 11)

mrousavy, 2024-07-02:
Hey all! I just spent some time to build a new feature: I created a PR to add a `isMirrored` prop to Camera: https://github.com/mrousavy/react-native-vision-camera/pull/3036

With this prop you can enable or disable output mirroring 🎉🥳

If you appreciate my time and dedication towards building new features and constantly improving this library, please [**consider sponsoring me on GitHub**](https://github.com/sponsors/mrousavy/)  :)

CyxouD, 2024-07-03:
@mrousavy thank you for introducing new features! However, in this issue, we needed Preview to be not mirrored in the front camera, but it isn't supported:

> To avoid dizzyness for users, the Preview will always be mirrored in the front Camera, but not in the back Camera. This is the same behaviour as in the native iOS Camera app.

mrousavy, 2024-07-03:
This is intentional - if the preview was mirrored the app would be pretty unusable because every movement goes in the wrong direction. All camera apps mirror the front preview.

If you really really need the preview not mirrored for whatever reason, you can just apply a `scaleX: -1` transform to the `<Camera>` view style.