margelo/react-native-vision-camera #2340: 🐛 Camera blacks out on Android on latest releases

mgcrea, 2024-01-02:
### What's happening?

I've started to encounter a black screen on Android while working in development mode, `<Camera />` starts black and stays black 100% of the time. I can fix it by adding a timeout to toggle the the isActive prop after one second.

It used to work flawlessly before (same device same OS version) but I'm not sure which version introduced the regression as I develop mainly on iOS. Could try to bisect the version if interested.

I've captured two adb traces, one when the bug occurs, one with a working start with the timeout.

There seems to be related issues but not exactly the same:
- https://github.com/mrousavy/react-native-vision-camera/issues/2330
- https://github.com/mrousavy/react-native-vision-camera/issues/2301

### Reproduceable Code

```tsx
const devices = useCameraDevices();
  const device = devices.find(({position}) => position === 'back');
  return <Camera
           style={StyleSheet.absoluteFill}
           device={device}
           resizeMode="contain"
         />
```

### Relevant log output

```shell
https://gist.github.com/mgcrea/8bce7771c8dbca5fdd1c0358c89fcd9e
```

### Camera Device

```json
{
  "sensorOrientation": "landscape-right",
  "hardwareLevel": "full",
  "maxZoom": 8,
  "minZoom": 1,
  "maxExposure": 72,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": true,
  "isMultiCam": false,
  "minExposure": -72,
  "name": "BACK (0)",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

OnePlus 6

### VisionCamera Version

3.6.17

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

piotrgomola, 2024-01-02:
I had the same issue with my S10, I followed the example in:
https://github.com/mrousavy/react-native-vision-camera/blob/main/package/example/src/CodeScannerPage.tsx

I noticed that going into background and back with the app resolved the issue. I modified a bit the code and it works for me now:

First I made `isActive` a state:
```
...
  const isFocused = useIsFocused()
  const isForeground = useIsForeground()
  const [isActive, setActive] = useState(false)
...
```

Then in the Camera I set this state in onInitialized callback:

```
<Camera
    style={StyleSheet.absoluteFill}
    device={device}
    isActive={isActive}
    onInitialized={() => {setActive(isFocused && isForeground)}}
    codeScanner={codeScanner}
    enableZoomGesture={true}
    />

dmackinn, 2024-01-02:
I have this same issue using 3.6.17 with RN 0.73.1. Switching to 3.6.16 gets rid of the issue.

juliacremus, 2024-01-03:
the error seems to be on the device definition, since I changed back to front and it opens the camera, but when I take the photo it gets black again. When I change again to back device it opens just for one photo and then blacks out.

I'm interested in this bug resolution too

tytom2003, 2024-01-04:
> ### What's happening?
> I've started to encounter a black screen on Android while working in development mode, `<Camera />` starts black and stays black 100% of the time. I can fix it by adding a timeout to toggle the the isActive prop after one second.
> 
> It used to work flawlessly before (same device same OS version) but I'm not sure which version introduced the regression as I develop mainly on iOS. Could try to bisect the version if interested.
> 
> I've captured two adb traces, one when the bug occurs, one with a working start with the timeout.
> 
> There seems to be related issues but not exactly the same:
> 
> * [🐛 (Android) Camera works on initial build and view blacks out on refresh #2330](https://github.com/mrousavy/react-native-vision-camera/issues/2330)
> * [🐛 Black Screen in Samsung S22 S23 ultra and some other devices when frame processor is on. #2301](https://github.com/mrousavy/react-native-vision-camera/issues/2301)
> 
> ### Reproduceable Code
> ```tsx
> const devices = useCameraDevices();
> const device = devices.find(({position}) => position === 'back');
> return <Camera
>          style={StyleSheet.absoluteFill}
>          device={device}
>          resizeMode="contain"
>        />
> ```
> 
> ### Relevant log output
> ```shell
> https://gist.github.com/mgcrea/8bce7771c8dbca5fdd1c0358c89fcd9e
> ```
> 
> ### Camera Device
> ```json
> {
>   "sensorOrientation": "landscape-right",
>   "hardwareLevel": "full",
>   "maxZoom": 8,
>   "minZoom": 1,
>   "maxExposure": 72,
>   "supportsLowLightBoost": false,
>   "neutralZoom": 1,
>   "physicalDevices": [
>     "wide-angle-camera"
>   ],
>   "supportsFocus": true,
>   "supportsRawCapture": true,
>   "isMultiCam": false,
>   "minExposure": -72,
>   "name": "BACK (0)",
>   "hasFlash": true,
>   "hasTorch": true,
>   "position": "back",
>   "id": "0"
> }
> ```
> 
> ### Device
> OnePlus 6
> 
> ### VisionCamera Version
> 3.6.17
> 
> ### Can you reproduce this issue in the VisionCamera Example app?
> Yes, I can reproduce the same issue in the Example app here
> 
> ### Additional information
> * [ ]  I am using Expo
> * [x]  I have enabled Frame Processors (react-native-worklets-core)
> * [x]  I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
> * [x]  I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
> * [x]  I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Hoping the way can help you to solve black screen issue.

reference: (discord - SteelMan)
Hi, I've not the code here, but my code was something like:

{device &&

}

and I got black screen.

After I tried something like this:

const [isActive, setActive] = useState(false);
useEffect(() => {
if (hasPermission) {
setTimeout(() => setActive(true), 500); // I'm not sure "setTimeout" is needed
}
}, [hasPermission]);

return device && (

)

tytom2003, 2024-01-04:
Hoping the way can help you to solve black screen problem at the first time open apps.

reference: (discord - SteelMan)
Hi, I've not the code here, but my code was something like:

{device &&

}

and I got black screen.

After I tried something like this:

const [isActive, setActive] = useState(false);
useEffect(() => {
if (hasPermission) {
setTimeout(() => setActive(true), 500); // I'm not sure "setTimeout" is needed
}
}, [hasPermission]);

return device && (
  <Camera
    device={device}
    isActive={isActive}
  />
)

iaheck, 2024-01-04:
Same issue with Expo 48.0.21 and RN 0.71.14. 3.6.16 works fine for me too as you suggested, thanks

mrousavy, 2024-01-05:
Hey! Thanks for reporting this issue.

I've been working the past days on making sure the Camera lifecycle is safely handled and released when needed, so this crash doesn't occur anymore.

I just created a PR for this - can you please this to see if that fixes the issue for you? https://github.com/mrousavy/react-native-vision-camera/pull/2339

If this fixes your issue, please consider [💖 **sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy/) to support me / thank me for building VisionCamera and continuously improving it. 

If this does not fix your issue, please clone the repo, check out the branch `fix/blackscreen` (the PR above), run the Example app, reproduce your issue there and share the `adb logcat` logs with me (upload to pastebin or gist) so I can investigate this further.

Thank you! 🙏

ashish293, 2024-01-05:
For me this worked-

`  const [isActive, setIsActive] = useState(false);
useEffect(() => {
    setTimeout(() => {
      setIsActive(true);
    }, 500);
  }, []);

<Camera
          style={StyleSheet.absoluteFill}
          ref={cameraRef}
          device={device}
          isActive={isActive}
          onError={handleCameraError}
        />
`

mrousavy, 2024-01-05:
That is not a fix, but rather a really dirty workaround.

ashish293, 2024-01-05:
> That is not a fix, but rather a really dirty workaround.

I tried for downgrading version from `3.6.17 to 3.6.16` but didn't worked.
Any suggestion as I need for my current project

mrousavy, 2024-01-05:
> Any suggestion as I need for my current project

Did you read what I wrote above? I built a fix for this in PR #2339....

daangelov, 2024-01-05:
With a fresh install on the example app on Android 12L (API 32) I can confirm that the black screen is not there but other bugs are still present like this one #2210. Here is a video, hope it helps:

[new.webm](https://github.com/mrousavy/react-native-vision-camera/assets/22197040/5b9838bc-f187-4e84-a271-1b57413cf862)

mrousavy, 2024-01-05:
Someone else said that the blackscreen is not fixed, can you confirm that it is fixed for you? https://github.com/mrousavy/react-native-vision-camera/issues/2343#issuecomment-1878978238

mgcrea, 2024-01-08:
Issue is indeed fixed, thank you!