margelo/react-native-vision-camera #2341: 🐛 App crash using QR scanner (useCodeScanner)

Alejandro095, 2024-01-02:
### What's happening?

The app crashes after scanning a QR code and navigate to another screen (using React Native Navigation) The QR code is successfully scanned but when I want to navigate to another page after (inside onCodeScanned callback) and the app sometimes crashes.

If I unmount the <Camera /> component before navigating to another page it also crash, example.

```js
  const [isActive, setIsActive] = React.useState(true);
  const [isMount, setIsMount] = React.useState(true);
  const device = useCameraDevice('back');

  const codeScanner = useCodeScanner({
    codeTypes: ['qr', 'ean-13'],
    onCodeScanned: codes => {
      setIsActive(false);
      setIsMount(false);  // <-- sometimes crash
      props.navigation.navigate('ANOTHER_SCREEN'); // React native navigation
    },
  });

  return (
      <Camera
        style={styles.camera}
        device={device}
        isActive={true}
        audio={false}
        codeScanner={codeScanner}
      />
    );
```

Is this a mistake or am I doing it wrong?

### Reproduceable Code

```tsx
import React, {useEffect} from 'react';
import {StyleSheet, Text} from 'react-native';

import {
  Camera,
  useCameraDevice,
  useCameraPermission,
  useCodeScanner,
} from 'react-native-vision-camera';

export default function VisionScreen(props: any) {
  const {hasPermission, requestPermission} = useCameraPermission();
  const device = useCameraDevice('back');

  useEffect(() => {
    if (!hasPermission) {
      console.log(requestPermission());
    }
  }, [hasPermission, requestPermission]);

  const codeScanner = useCodeScanner({
    codeTypes: ['qr', 'ean-13'],
    onCodeScanned: codes => {
      props.navigation.navigate('ANOTHER_SCREEN'); // <-- crash
    },
  });

  if (device == null || !hasPermission) {
    return <Text>HasPermission False</Text>;
  }

  return (
    <Camera
      style={styles.camera}
[... 208 lines cut ...]
  "hasTorch": false,
  "position": "back",
  "id": "0"
}
```

### Device

Samung Tab A7

### VisionCamera Version

3.6.16

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

klabusta, 2024-01-05:
Same here on Xiaomi Redmi Note 10 and weirdly only happening when navigating aways from the camera. 3.6.16

coderOnHere, 2024-01-08:
I have the same problem. Sometimes, especially when the qr is in front of the camera before opening, the application crashes. Does anyone know what might be the problem? It should read qr information and navigate to another screen.

Happened on both pc emulators and real devices.

hi-yeonju, 2024-01-08:
try use `onInitialized` onInitialized={handleInit}; handleInit = () => {  settimeout(() => { setIsActive(true) }, [500]) }

mrousavy, 2024-01-08:
isActive should not be false initially - I fixed that race condition in this PR: https://github.com/mrousavy/react-native-vision-camera/pull/2339

I'll release an update soon, then you won't need that 500ms delay workaround anymore :)

The QR code issue is a different one, not sure what this is about but maybe this fixes it as well? https://github.com/mrousavy/react-native-vision-camera/pull/2355

Wael-Zoaiter, 2024-01-09:
@mrousavy I'm still facing the same issue in v3.7.0

mrousavy, 2024-01-09:
Hm weird. There's nothing I can do without having adb logcat logs though. The logs shared here look like a tombstone, but there's no _symbols_ in here (aka no filenames or stack traces).

I either need a crash with symbols ("symbolicated stacktrace"/tombstone), or the full adb logcat logs.

Alejandro095, 2024-01-09:
@mrousavy I share you the full adb logcat logs (I imported from Android Studio) and a video of what I was doing. If you need something else let me know.

The code I was using is this

```js
import {useFocusEffect} from '@react-navigation/native';
import React, {useCallback, useEffect} from 'react';
import {StyleSheet, Text} from 'react-native';

import {
  Camera,
  useCameraDevice,
  useCameraPermission,
  useCodeScanner,
} from 'react-native-vision-camera';

export default function CamaraScreen(props: any) {
  const [isActive, setIsActive] = React.useState(true);
  const [isMount, setIsMount] = React.useState(true);

  const {hasPermission, requestPermission} = useCameraPermission();
  const device = useCameraDevice('back');

  useFocusEffect(
    useCallback(() => {
      console.log('Camera is mounted');
      setIsMount(true);
      return () => {
        console.log('Camera is unmounted');
        setIsMount(false);
      };
    }, []),
  );

  useEffect(() => {
    if (!hasPermission) {
      console.log(requestPermission());
    }
  }, [hasPermission, requestPermission]);

  const codeScanner = useCodeScanner({
    codeTypes: ['qr', 'ean-13'],
    onCodeScanned: codes => {
      //setIsActive(false);
      console.log(codes[0].value);
      props.navigation.navigate('ANOTHER_SCREEN'); // <-- crash
    },
  });

  if (device == null || !hasPermission) {
    return <Text>HasPermission False</Text>;
  }

  return isMount ? (
    <Camera
      style={styles.camera}
      device={device}
      isActive={isActive}
      audio={false}
      photo={false}
      video={false}
      codeScanner={codeScanner}
    />
  ) : null;
}
const styles = StyleSheet.create({
  camera: {
    width: 680,
    height: 680,
  },
});
```

https://github.com/mrousavy/react-native-vision-camera/assets/50101944/b4c1ec60-a799-4463-8a52-4b2672947675

[samsung-SM-X205-Android-13_2024-01-09_123256.logcat.json](https://github.com/mrousavy/react-native-vision-camera/files/13877808/samsung-SM-X205-Android-13_2024-01-09_123256.logcat.json)

klabusta, 2024-01-11:
Still crashing with 3.7.0, initialization works much better though! My adb logcat looks pretty similar to the one from above :(

mrousavy, 2024-01-11:
Thanks for the detailed info! The log format is a bit weird (json instead of just logs), but I can work with that.

I just wonder if the crash is because of some call in the `CodeScanner`: https://github.com/mrousavy/react-native-vision-camera/blob/2b106225590e2694f25a083003bc983d2cb490bb/package/android/src/main/java/com/mrousavy/camera/core/CodeScannerPipeline.kt#L42-L66

..or because of something in the CameraSession itself. One thing you could try is to wrap the method linked above in try catch statements:

```kt
    imageReader.setOnImageAvailableListener({ reader ->
      try {
        val image = reader.acquireNextImage() ?: return@setOnImageAvailableListener

        if (isBusy) {
          // We're currently executing on a previous Frame, so we skip this one.
          // Close it and free it again, so that the Camera does not stall.
          image.close()
          return@setOnImageAvailableListener
        }

        isBusy = true
        // TODO: Get correct orientation
        val inputImage = InputImage.fromMediaImage(image, Orientation.PORTRAIT.toDegrees())
        scanner.process(inputImage)
          .addOnSuccessListener { barcodes ->
            try {
              if (barcodes.isNotEmpty()) {
                callback.onCodeScanned(barcodes, CodeScannerFrame(inputImage.width, inputImage.height))
              }
            } catch (e: Throwable) {
              Log.e("VisionCamera", "Success callback threw an error!", e)
            }
          }
          .addOnFailureListener { error ->
            try {
              callback.onError(error)
            } catch (e: Throwable) {
              Log.e("VisionCamera", "Failure callback threw an error!", e)
            }
          }
          .addOnCompleteListener {
            try {
              image.close()
              isBusy = false
            } catch (e: Throwable) {
              Log.e("VisionCamera", "Completion callback threw an error!", e)
            }
          }
      } catch (e: Throwable) {
        Log.e("VisionCamera", "CodeScanner threw an error!", e)
      }
    }, CameraQueues.videoQueue.handler)
```

And then see what happens. It seems like this is a SIGSEV tho, so a C++ error that cannot really be caught. Might even be an issue in the Google MLKit barcode scanner library 🤔

Alper-Demir-Eng, 2024-01-11:
@mrousavy 
For me it only crashes if the View containing the Camera component is conditionally rendered, i.e. programmatically hidden or set to be visible, if the said container is always visible and never gets hidden it won't crash.

`{isCamVisible ?  (
        <View>
          <Camera
            style={StyleSheet.absoluteFill}
            device={device}
            isActive={isCamActive}
            codeScanner={codeScanner}
          /> 
         </View> ) : null}`

Alejandro095, 2024-01-11:
@mrousavy  sorry for the format haha. Well now I copied the logs using Flipper and this is the output, but I didn't see any error log of the try catch statements in the output :(

[log 11 01 23.txt](https://github.com/mrousavy/react-native-vision-camera/files/13907579/log.11.01.23.txt)

In other hand, in the section of Crash Reporter of Flipper I saw reports related SIGSEGV
[image]

First report
Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x7f045093c660 in tid 29274 (pool-7-thread-4), pid 29175 (com.test)

Second report
[second report.txt](https://github.com/mrousavy/react-native-vision-camera/files/13907625/second.report.txt)

Third report
[third report.txt](https://github.com/mrousavy/react-native-vision-camera/files/13907643/third.report.txt)

**EDIT**: This time I use the emulator for the test.

dalisalvador, 2024-01-12:
> @mrousavy For me it only crashes if the View containing the Camera component is conditionally rendered, i.e. programmatically hidden or set to be visible, if the said container is always visible and [...]

Same problem.
Quick dirty workaround `opacity: 0`

dFelinger, 2024-01-12:
Same issue just when screen with camera has button to navigate to another screen pressed and camera has codeScanner prop (without codeScanner prop app doesn't crashes).

Using @react-navigation/native-stack

My logcat:

```
01-12 21:42:13.235   361   361 E Layer   : [Surface(name=Task=285)/@0x21b39c8 - animation-leash#0] No local sync point found
01-12 21:42:13.235   361   361 E Layer   : [Surface(name=Task=285)/@0x21b39c8 - animation-leash#0] No local sync point found
01-12 21:42:13.235   361   361 E Layer   : [Surface(name=Task=1)/@0xf63d1b1 - animation-leash#0] No local sync point found
01-12 21:42:13.235   361   361 E Layer   : [Surface(name=Task=1)/@0xf63d1b1 - animation-leash#0] No local sync point found
01-12 21:42:14.515 21363 21444 E chromium: [0112/214214.515218:ERROR:variations_seed_loader.cc(69)] Failed to open file for reading. Errno: 2
01-12 21:42:19.746 21363 21437 E CameraManagerGlobal: cameraIdHasConcurrentStreamsLocked called on non existing camera id
01-12 21:42:19.754 21363 21437 E CameraManagerGlobal: cameraIdHasConcurrentStreamsLocked called on non existing camera id
01-12 21:42:20.166   297 12150 E libc    : Access denied finding property "ro.camera.req.fmq.size"
01-12 21:42:20.166   297 12150 E FMQ     : grantorIdx must be less than 3
01-12 21:42:20.166   297 12150 E libc    : Access denied finding property "ro.camera.res.fmq.size"
01-12 21:42:20.167   297 12150 E FMQ     : grantorIdx must be less than 3
01-12 21:42:20.167   395   395 E FMQ     : grantorIdx must be less than 3
01-12 21:42:20.167   395   395 E FMQ     : grantorIdx must be less than 3
01-12 21:42:20.313 10870 10870 E AbstractGmsTracer: Traced class shouldn't be obfuscated: ModuleInstallService [CONTEXT service_id=259 ]
01-12 21:42:20.429   395   395 E Camera3-Device: configureStreams: Stream 0: DataSpace override not allowed for format 0x23
01-12 21:42:20.455   395   395 E Camera3-Device: configureStreams: Stream 2: DataSpace override not allowed for format 0x23
01-12 21:42:20.943   361   380 E SurfaceFlinger: setDesiredDisplayConfigSpecs: failed to call setDesiredDisplayConfigSpecs: -22
01-12 21:42:20.943   517   542 E SurfaceFlinger: setDesiredDisplayConfigSpecs failed to transact: -22
01-12 21:42:24.920   475   475 E netmgr  : qemu_pipe_open_ns:62: Could not connect to the 'pipe:qemud:network' service: Invalid argument
01-12 21:42:24.920   475   475 E netmgr  : Failed to open QEMU pipe 'qemud:network': Invalid argument
01-12 21:42:24.992   478   478 E wifi_forwarder: qemu_pipe_open_ns:62: Could not connect to the 'pipe:qemud:wififorward' service: Invalid argument
01-12 21:42:24.992   478   478 E wifi_forwarder: RemoteConnection failed to initialize: RemoteConnection failed to open pipe
01-12 21:42:28.706 21363 21436 E BufferQueueProducer: [ImageReader-1280x720f23m2-21363-1](id:537300000001,api:4,p:395,c:21363) queueBuffer: BufferQueue has been abandoned
01-12 21:42:28.706   395  7884 E Surface : queueBuffer: error queuing buffer to SurfaceTexture, -19
01-12 21:42:28.706   395  7884 E Camera3-OutputStream: returnBufferCheckedLocked: Stream 2: Error queueing buffer to native window: No such device (-19)
01-12 21:42:28.710 21363 21436 E BufferQueueProducer: [ImageReader-1280x720f23m2-21363-1](id:537300000001,api:4,p:395,c:21363) queueBuffer: BufferQueue has been abandoned
01-12 21:42:28.710   395  7884 E Surface : queueBuffer: error queuing buffer to SurfaceTexture, -19
01-12 21:42:28.710   395  7884 E Camera3-OutputStream: returnBufferCheckedLocked: Stream 2: Error queueing buffer to native window: No such device (-19)
01-12 21:42:28.734 21363 21485 F libc    : Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0xb2484442 in tid 21485 (pool-5-thread-4), pid 21363 (com.tools)
01-12 21:42:28.764   395 15386 E Camera3-Device: configureStreams: Stream 3: DataSpace override not allowed for format 0x23
01-12 21:42:28.840 21498 21498 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
01-12 21:42:28.840 21498 21498 F DEBUG   : Build fingerprint: 'google/sdk_gphone_x86/generic_x86_arm:11/RSR1.201013.001/6903271:user/release-keys'
01-12 21:42:28.841 21498 21498 F DEBUG   : Revision: '0'
01-12 21:42:28.841 21498 21498 F DEBUG   : ABI: 'x86'
01-12 21:42:28.844 21498 21498 F DEBUG   : Timestamp: 2024-01-12 21:42:28+0700
01-12 21:42:28.844 21498 21498 F DEBUG   : pid: 21363, tid: 21485, name: pool-5-thread-4  >>> com.tools <<<
01-12 21:42:28.848 21498 21498 F DEBUG   : uid: 10156
01-12 21:42:28.848 21498 21498 F DEBUG   : signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0xb2484442
01-12 21:42:28.849 21498 21498 F DEBUG   :     eax 00007442  ebx b247d000  ecx 00000000  edx b0c1b2b0
01-12 21:42:28.849 21498 21498 F DEBUG   :     edi 00000500  esi b0c1b2b0
01-12 21:42:28.849 21498 21498 F DEBUG   :     ebp b0c1b098  esp b0c1b030  eip b23a25e8
01-12 21:42:28.973 21498 21498 F DEBUG   : backtrace:
01-12 21:42:28.973 21498 21498 F DEBUG   :     NOTE: Function names and BuildId information is missing for some frames due
01-12 21:42:28.973 21498 21498 F DEBUG   :     NOTE: to unreadable libraries. For unwinds of apps, only shared libraries
01-12 21:42:28.973 21498 21498 F DEBUG   :     NOTE: found under the lib/ directory are readable.
01-12 21:42:28.973 21498 21498 F DEBUG   :       #00 pc 001575e8  /data/user_de/0/com.google.android.gms/app_chimera/m/00000070/dl-VisionBarcode.optional_234414100700.apk
[... cut ...]
01-12 21:42:29.633 10870 21506 E AbstractGmsTracer: Traced class shouldn't be obfuscated: SnetDropBoxIntentOperation [CONTEXT service_id=259 ]
01-12 21:42:29.634 10847 10847 E AbstractGmsTracer: Traced class shouldn't be obfuscated: PersistentTrustedChimeraReceiver [CONTEXT service_id=259 ]
01-12 21:42:29.653 10847 21503 E AbstractGmsTracer: Traced class shouldn't be obfuscated: DropboxEntryAddedOperation [CONTEXT service_id=259 ]
01-12 21:42:29.780   395   395 E CameraService: binderDied: Java client's binder died, removing it from the list of active clients
```

dFelinger, 2024-01-14:
Camera version 3.7.1
React navigation @react-navigation/native-stack
Tested on emulator Pixel 4 Api level 30

Camera screen has button with navigation to another screen pressed causes app crashes.
Without codeScanner prop app doesn't crashes.

<details>
<summary>logcat</summary>
<code>
01-14 16:41:26.259 14621 14640 W et.felinger.tm: JNI critical lock held for 30.703ms on Thread[15,tid=14640,Runnable,Thread*=0xdde13410,peer=0x13a00538,"Binder:14621_2"]
01-14 16:41:28.046   540  1163 I system_server: oneway function results will be dropped but finished with status OK and parcel size 4
01-14 16:41:28.046   540  1163 I chatty  : uid=1000(system) Binder:540_A identical 1 line
01-14 16:41:28.046   540  1163 I system_server: oneway function results will be dropped but finished with status OK and parcel size 4
01-14 16:41:28.070 14621 14682 W unknown:ReactNative: Attempt to set local data for view with unknown tag: -1
01-14 16:41:28.071 14621 14682 W unknown:ReactNative: Attempt to set local data for view with unknown tag: -1
01-14 16:41:28.471 14621 14621 I CameraSession: PreviewView Surface destroyed! Surface(name=null)/@0x81ab69a
01-14 16:41:28.471 14621 14621 I CameraSession: Destroying Preview Output...
01-14 16:41:28.515 14621 14621 I CameraSession: configure { ... }: Waiting for lock...
01-14 16:41:28.515 14621 14621 I CameraSession: configure { ... }: Updating CameraSession Configuration... Difference(deviceChanged=false, outputsChanged=true, sidePropsChanged=true, isActiveChanged=false)
01-14 16:41:28.515 14621 14621 I CameraSession: Destroying previous outputs...
01-14 16:41:28.515 14621 14621 I SurfaceOutput: Closing BarcodeScanner..
01-14 16:41:28.520 14621 14621 I CameraView: invokeOnStopped()
01-14 16:41:28.525 14621 14621 I CameraSession: Creating outputs for Camera #0...
01-14 16:41:28.526 14621 14640 E BufferQueueProducer: [ImageReader-1280x720f23m2-14621-0](id:391d00000000,api:4,p:394,c:14621) queueBuffer: BufferQueue has been abandoned
01-14 16:41:28.526 14621 14621 I CameraSession: Adding 1280x720 CodeScanner Output in YUV_420_888...
01-14 16:41:28.527   394   495 E Surface : queueBuffer: error queuing buffer to SurfaceTexture, -19
01-14 16:41:28.527   394   495 E Camera3-OutputStream: returnBufferCheckedLocked: Stream 1: Error queueing buffer to native window: No such device (-19)
01-14 16:41:28.528 14621 14640 E BufferQueueProducer: [ImageReader-1280x720f23m2-14621-0](id:391d00000000,api:4,p:394,c:14621) dequeueBuffer: BufferQueue has been abandoned
01-14 16:41:28.529   394 14712 E Camera3-OutputStream: getBufferLockedCommon: Stream 1: Can't dequeue next output buffer: No such device (-19)
01-14 16:41:28.541 14621 14621 I CreateCaptureSession: Camera #0: Creating Capture Session #2... (Hardware Level: 0 | Outputs: [VIDEO (1280 x 720 CodeScanner for [QR, EAN_13] (35))])
01-14 16:41:28.541 14621 14621 I CreateCaptureSession: Using new API (>=28)
01-14 16:41:28.542   394  5827 E CameraDeviceClient: cancelRequest: Camera 0: Canceling request ID 0 doesn't match current request ID -1
01-14 16:41:28.548 14621 14714 D TransportRuntime.SQLiteEventStore: Storing event with priority=DEFAULT, name=FIREBASE_ML_SDK for destination cct
01-14 16:41:28.559 14621 14714 D TransportRuntime.JobInfoScheduler: Upload for context TransportContext(cct, DEFAULT, MSRodHRwczovL2ZpcmViYXNlbG9nZ2luZy5nb29nbGVhcGlzLmNvbS92MGNjL2xvZy9iYXRjaD9mb3JtYXQ9anNvbl9wcm90bzNc) is already scheduled. Returning...
01-14 16:41:28.585   394  1483 I CameraLatencyHistogram: Stream 0 dequeueBuffer latency histogram (71) samples:
01-14 16:41:28.585   394  1483 I CameraLatencyHistogram:         5     10     15     20     25     30     35     40     45    inf (max ms)
01-14 16:41:28.585   394  1483 I CameraLatencyHistogram:      29.58   2.82   1.41   1.41   1.41   2.82   1.41   5.63   9.86   43.66 (%)
01-14 16:41:28.585   394  1483 I CameraLatencyHistogram: Stream 0 latency histogram for wait on max_buffers (1) samples:
01-14 16:41:28.585   394  1483 I CameraLatencyHistogram:        33     66     99    132    165    198    231    264    297    inf (max ms)
01-14 16:41:28.585   394  1483 I CameraLatencyHistogram:      100.00   0.00   0.00   0.00   0.00   0.00   0.00   0.00   0.00   0.00 (%)
01-14 16:41:28.587   394   735 I CameraLatencyHistogram: Stream 1 dequeueBuffer latency histogram (71) samples:
01-14 16:41:28.587   394   735 I CameraLatencyHistogram:         5     10     15     20     25     30     35     40     45    inf (max ms)
01-14 16:41:28.587   394   735 I CameraLatencyHistogram:      94.37   4.23   1.41   0.00   0.00   0.00   0.00   0.00   0.00   0.00 (%)
01-14 16:41:28.598 14621 14717 F libc    : Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0xae311e42 in tid 14717 (pool-6-thread-1), pid 14621 (et.felinger.tmc)
01-14 16:41:28.604   394   735 E Camera3-Device: configureStreams: Stream 2: DataSpace override not allowed for format 0x23
01-14 16:41:28.624   394   735 D Camera3-Device: Set real time priority for request queue thread (tid 14712)
01-14 16:41:28.626 14621 14703 I CreateCaptureSession: Camera #0: Successfully created CameraCaptureSession #2!
01-14 16:41:28.627 14621 14621 I CameraSession: Successfully configured Session with 1 outputs for Camera #0!
01-14 16:41:28.685 14725 14725 I crash_dump32: obtaining output fd from tombstoned, type: kDebuggerdTombstone
01-14 16:41:28.688   283   283 I tombstoned: received crash request for pid 14717
[... cut ...]
01-14 16:41:30.037  1245  1245 I GoogleInputMethodService: GoogleInputMethodService.onStartInput():1894 onStartInput(EditorInfo{EditorInfo{packageName=com.google.android.apps.nexuslauncher, inputType=0, inputTypeString=NULL, enableLearning=false, autoCorrection=false, autoComplete=false, imeOptions=0, privateImeOptions=null, actionName=UNSPECIFIED, actionLabel=null, initialSelStart=-1, initialSelEnd=-1, initialCapsMode=0, label=null, fieldId=-1, fieldName=null, extras=null, hintText=null, hintLocales=[]}}, false)
01-14 16:41:30.038  1245  1245 I GoogleInputMethodService: GoogleInputMethodService.updateDeviceLockedStatus():2147 repeatCheckTimes = 1, locked = false
0
</code>
</details>

SveBo, 2024-01-19:
> @mrousavy For me it only crashes if the View containing the Camera component is conditionally rendered, i.e. programmatically hidden or set to be visible, if the said container is always visible and never gets hidden it won't crash.
> 

For me the following workaround seems to work out:

I set isActive to false and wait a few moments before hiding the container

marcshilling, 2024-01-22:
Same stack trace as https://github.com/mrousavy/react-native-vision-camera/issues/2116

klabusta, 2024-01-25:
> I set Camera.isActive to false and wait a few moments before hiding the container

Although not a great workaround, this works for me.

closed 2024-02-06 by mrousavy

mrousavy, 2024-02-06:
Just released [VisionCamera 3.9.0-beta.1](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v3.9.0-beta.1), which includes the [big `PersistentCaptureSession` PR](https://github.com/mrousavy/react-native-vision-camera/pull/2494) that aims to fix all the blackscreen issues.... :eyes::tada: 

You can try that new beta out yourself and let me know if you still encounter blackscreen issues. I'll continue to test on more devices in the meantime, organize the code a bit better, and then I'll tackle the preview stretching, focus, and flash issues!

If that fixes the blackscreen issues for you, please consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks! ❤️

mgcrea, 2024-02-07:
@mrousavy I am hit by this `signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x6f9005c549` fault using either [my frame processor](https://github.com/mgcrea/vision-camera-barcode-scanner) or your `codeScanner` API while using `3.9.0-beta.1`.

In the case of my frame processor the crash always happen when the unmount slips during a scan (which when using a 5fps processor happens pretty quickly!).

I tried to track the unmount using a parent component with:

```tsx
  const isMountedRef = useSharedValue<boolean>(true);
  useEffect(() => {
    return () => {
      console.log("unmont!");
      isMountedRef.value = false;
    };
  }, [isMountedRef]);
  
    const frameProcessor = useFrameProcessor(
    (frame) => {
      "worklet";
      if (isMountedRef && isMountedRef.value === false) {
        console.log("abort!");
        return;
      }
      runAsync(frame, () => {
        "worklet";
        runAtTargetFps(fps, () => {
        ...
```

to abort the scan if unmount but alas it's not enough. Any ideas on how you could detect unmount and either skip calling frame processors soon enough or keep around the frame as I guess the segmentation fault is the frame that has gone away.

mrousavy, 2024-02-08:
@mgcrea I need the full logs

mgcrea, 2024-02-08:
> @mgcrea I need the full logs

@mrousavy crash with codeScanner (full adb dump), added newlines on mount toggle and kept a working unmount and the last failed one:

https://gist.github.com/mgcrea/8776644d047af93f4f73fdb3f0b05914

Related crash:

```
02-08 09:31:51.454 27273  7299 F libc    : Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x70c91f60fa in tid 7299 (pool-5-thread-4), pid 27273 (escannerexample)
02-08 09:31:51.628  7877  7877 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
02-08 09:31:51.628  7877  7877 F DEBUG   : Build fingerprint: 'OnePlus/OnePlus6/OnePlus6:11/RKQ1.201217.002/2111252325:user/release-keys'
02-08 09:31:51.628  7877  7877 F DEBUG   : Revision: '0'
02-08 09:31:51.628  7877  7877 F DEBUG   : ABI: 'arm64'
02-08 09:31:51.629  7877  7877 F DEBUG   : Timestamp: 2024-02-08 09:31:51+0100
02-08 09:31:51.629  7877  7877 F DEBUG   : pid: 27273, tid: 7299, name: pool-5-thread-4  >>> com.visioncameracodescannerexample <<<
02-08 09:31:51.629  7877  7877 F DEBUG   : uid: 10337
02-08 09:31:51.629  7877  7877 F DEBUG   : signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x70c91f60fa
02-08 09:31:51.629  7877  7877 F DEBUG   :     x0  0000006e00a61508  x1  0000000000000206  x2  0000000000000000  x3  0000000000000010
02-08 09:31:51.629  7877  7877 F DEBUG   :     x4  0000000000000000  x5  0000006e00a618c8  x6  0000000000000120  x7  0000006e00a62100
02-08 09:31:51.629  7877  7877 F DEBUG   :     x8  0000000000000000  x9  0000000000000103  x10 00000000000000b9  x11 00000072dafa04c0
02-08 09:31:51.629  7877  7877 F DEBUG   :     x12 0000000000000102  x13 000000000008c0fa  x14 00000070c91f60fa  x15 00000070c91f5bfa
02-08 09:31:51.629  7877  7877 F DEBUG   :     x16 00000070c916a000  x17 00000073cd4cbb74  x18 00000070c924a5f4  x19 0000006e00a61628
02-08 09:31:51.629  7877  7877 F DEBUG   :     x20 0000006e00a615c8  x21 0000006e00a61760  x22 0000000000000103  x23 00000000000000b9
02-08 09:31:51.629  7877  7877 F DEBUG   :     x24 0000000000000000  x25 0000000000000100  x26 0000006e00a62198  x27 0000006e00a61e18
02-08 09:31:51.629  7877  7877 F DEBUG   :     x28 0000007138b23970  x29 0000006e00a622e0
02-08 09:31:51.629  7877  7877 F DEBUG   :     lr  0000006e005b3598  sp  0000006e00a61500  pc  0000006e005b3660  pst 0000000080000000
02-08 09:31:51.759  7877  7877 F DEBUG   : backtrace:
02-08 09:31:51.759  7877  7877 F DEBUG   :       #00 pc 00000000000ed660  /data/app/~~EoiCw6qP9q17YRrPLW2cMA==/com.visioncameracodescannerexample-7TaaHtsWkYBYbDxlfdFEPg==/lib/arm64/libbarhopper_v3.so (BuildId: cf64ac3d5cfd71b04d3bc4112d678e48)
02-08 09:31:51.759  7877  7877 F DEBUG   :       #01 pc 00000000000d03c8  /data/app/~~EoiCw6qP9q17YRrPLW2cMA==/com.visioncameracodescannerexample-7TaaHtsWkYBYbDxlfdFEPg==/lib/arm64/libbarhopper_v3.so (BuildId: cf64ac3d5cfd71b04d3bc4112d678e48)
02-08 09:31:51.759  7877  7877 F DEBUG   :       #02 pc 00000000000ca240  /data/app/~~EoiCw6qP9q17YRrPLW2cMA==/com.visioncameracodescannerexample-7TaaHtsWkYBYbDxlfdFEPg==/lib/arm64/libbarhopper_v3.so (BuildId: cf64ac3d5cfd71b04d3bc4112d678e48)
02-08 09:31:51.759  7877  7877 F DEBUG   :       #03 pc 00000000000c9f64  /data/app/~~EoiCw6qP9q17YRrPLW2cMA==/com.visioncameracodescannerexample-7TaaHtsWkYBYbDxlfdFEPg==/lib/arm64/libbarhopper_v3.so (BuildId: cf64ac3d5cfd71b04d3bc4112d678e48)
02-08 09:31:51.759  7877  7877 F DEBUG   :       #04 pc 00000000000ca4f0  /data/app/~~EoiCw6qP9q17YRrPLW2cMA==/com.visioncameracodescannerexample-7TaaHtsWkYBYbDxlfdFEPg==/lib/arm64/libbarhopper_v3.so (BuildId: cf64ac3d5cfd71b04d3bc4112d678e48)
02-08 09:31:51.759  7877  7877 F DEBUG   :       #05 pc 00000000000c8aa4  /data/app/~~EoiCw6qP9q17YRrPLW2cMA==/com.visioncameracodescannerexample-7TaaHtsWkYBYbDxlfdFEPg==/lib/arm64/libbarhopper_v3.so (BuildId: cf64ac3d5cfd71b04d3bc4112d678e48)
02-08 09:31:51.759  7877  7877 F DEBUG   :       #06 pc 00000000000c8b44  /data/app/~~EoiCw6qP9q17YRrPLW2cMA==/com.visioncameracodescannerexample-7TaaHtsWkYBYbDxlfdFEPg==/lib/arm64/libbarhopper_v3.so (Java_com_google_android_libraries_barhopper_BarhopperV3_recognizeBufferNative+84) (BuildId: cf64ac3d5cfd71b04d3bc4112d678e48)
02-08 09:31:51.759  7877  7877 F DEBUG   :       #07 pc 0000000000050c7c  /data/app/~~EoiCw6qP9q17YRrPLW2cMA==/com.visioncameracodescannerexample-7TaaHtsWkYBYbDxlfdFEPg==/oat/arm64/base.odex (art_jni_trampoline+204)
02-08 09:31:51.759  7877  7877 F DEBUG   :       #08 pc 0000000000133564  /apex/com.android.art/lib64/libart.so (art_quick_invoke_stub+548) (BuildId: 32cda5ceec1e68666841a6ebde60d2e1)
02-08 09:31:51.759  7877  7877 F DEBUG   :       #09 pc 00000000001a8a78  /apex/com.android.art/lib64/libart.so (art::ArtMethod::Invoke(art::Thread*, unsigned int*, unsigned int, art::JValue*, char const*)+200) (BuildId: 32cda5ceec1e68666841a6ebde60d2e1)
02-08 09:31:51.759  7877  7877 F DEBUG   :       #10 pc 00000000003185d8  /apex/com.android.art/lib64/libart.so (art::interpreter::ArtInterpreterToCompiledCodeBridge(art::Thread*, art::ArtMethod*, art::ShadowFrame*, unsigned short, art::JValue*)+376) (BuildId: 32cda5ceec1e68666841a6ebde60d2e1)
02-08 09:31:51.759  7877  7877 F DEBUG   :       #11 pc 000000000030f9c4  /apex/com.android.art/lib64/libart.so (bool art::interpreter::DoCall<true, false>(art::ArtMethod*, art::Thread*, art::ShadowFrame&, art::Instruction const*, unsigned short, art::JValue*)+696) (BuildId: 32cda5ceec1e68666841a6ebde60d2e1)
[... cut ...]
02-08 09:31:51.760  7877  7877 F DEBUG   :       #100 pc 00000000005a4788  /apex/com.android.art/lib64/libart.so (art::Thread::CreateCallback(void*)+1308) (BuildId: 32cda5ceec1e68666841a6ebde60d2e1)
02-08 09:31:51.760  7877  7877 F DEBUG   :       #101 pc 00000000000b0048  /apex/com.android.runtime/lib64/bionic/libc.so (__pthread_start(void*)+64) (BuildId: 07fbaeed7b7a19203975f06be6f1d5ef)
02-08 09:31:51.760  7877  7877 F DEBUG   :       #102 pc 00000000000503c8  /apex/com.android.runtime/lib64/bionic/libc.so (__start_thread+64) (BuildId: 07fbaeed7b7a19203975f06be6f1d5ef)
```

huanguolin, 2024-02-26:
Same issue here.
```
"react-native-vision-camera": "3.9.0", 
"react-native": "^0.72.7",
```

```
*** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***

Build fingerprint: 'Sony/A202SO/A202SO:13/65.1.D.3.9/065001D003000901165108286:user/release-keys'
Revision: '0'
ABI: 'arm64'
Timestamp: 2024-02-26 14:56:17.717281136+0900
Process uptime: 7694s
Cmdline: com.mobilepart.teacher
pid: 28646, tid: 9262, name: pool-18-thread-  >>> com.mobilepart.teacher <<<
uid: 10687
signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x000000721f0b01c2
    x0  000000721ef77a28  x1  0000000000000164  x2  000000755a81d2cc  x3  0000000000000010
    x4  0000000000000000  x5  000000721ef77de8  x6  b400007300000100  x7  000000721ef78620
    x8  0000000000000000  x9  00000000000000b2  x10 00000000000000a4  x11 b40000741ed9b450
    x12 00000000000000b1  x13 00000000000301c2  x14 000000721f0b01c2  x15 000000721f0afbc2
    x16 000000721f080000  x17 000000755a7820e4  x18 000000721f18cdf1  x19 000000721ef77b48
    x20 000000721ef77ae8  x21 000000721ef77c80  x22 00000000000000b2  x23 00000000000000a4
    x24 0000000000000000  x25 0000000000000100  x26 000000721ef786b8  x27 000000721ef78338
    x28 b40000729e3cb340  x29 000000721ef78820
    lr  00000070a735d598  sp  000000721ef77a20  pc  00000070a735d660  pst 0000000080001000
backtrace:
      #00 pc 00000000000ed660  /data/app/~~o0DJueGdxsdbGpzbRNJrOg==/com.mobilepart.teacher-L6aGGx-wMPf4--6FR06KRA==/base.apk!libbarhopper_v3.so (BuildId: cf64ac3d5cfd71b04d3bc4112d678e48)
      #01 pc 00000000000d03c8  /data/app/~~o0DJueGdxsdbGpzbRNJrOg==/com.mobilepart.teacher-L6aGGx-wMPf4--6FR06KRA==/base.apk!libbarhopper_v3.so (BuildId: cf64ac3d5cfd71b04d3bc4112d678e48)
      #02 pc 00000000000ca240  /data/app/~~o0DJueGdxsdbGpzbRNJrOg==/com.mobilepart.teacher-L6aGGx-wMPf4--6FR06KRA==/base.apk!libbarhopper_v3.so (BuildId: cf64ac3d5cfd71b04d3bc4112d678e48)
      #03 pc 00000000000c9f64  /data/app/~~o0DJueGdxsdbGpzbRNJrOg==/com.mobilepart.teacher-L6aGGx-wMPf4--6FR06KRA==/base.apk!libbarhopper_v3.so (BuildId: cf64ac3d5cfd71b04d3bc4112d678e48)
      #04 pc 00000000000ca4f0  /data/app/~~o0DJueGdxsdbGpzbRNJrOg==/com.mobilepart.teacher-L6aGGx-wMPf4--6FR06KRA==/base.apk!libbarhopper_v3.so (BuildId: cf64ac3d5cfd71b04d3bc4112d678e48)
      #05 pc 00000000000c8aa4  /data/app/~~o0DJueGdxsdbGpzbRNJrOg==/com.mobilepart.teacher-L6aGGx-wMPf4--6FR06KRA==/base.apk!libbarhopper_v3.so (BuildId: cf64ac3d5cfd71b04d3bc4112d678e48)
      #06 pc 00000000000c8b44  /data/app/~~o0DJueGdxsdbGpzbRNJrOg==/com.mobilepart.teacher-L6aGGx-wMPf4--6FR06KRA==/base.apk!libbarhopper_v3.so (Java_com_google_android_libraries_barhopper_BarhopperV3_recognizeBufferNative+84) (BuildId: cf64ac3d5cfd71b04d3bc4112d678e48)
      #07 pc 0000000000355630  /apex/com.android.art/lib64/libart.so (art_quick_generic_jni_trampoline+144) (BuildId: 02bec5940be704b863f6514fc7d81c41)
      #08 pc 000000000033eba4  /apex/com.android.art/lib64/libart.so (art_quick_invoke_stub+612) (BuildId: 02bec5940be704b863f6514fc7d81c41)
      #09 pc 000000000037c03c  /apex/com.android.art/lib64/libart.so (art::interpreter::ArtInterpreterToCompiledCodeBridge(art::Thread*, art::ArtMethod*, art::ShadowFrame*, unsigned short, art::JValue*)+300) (BuildId: 02bec5940be704b863f6514fc7d81c41)
      #10 pc 000000000037b860  /apex/com.android.art/lib64/libart.so (bool art::interpreter::DoCall<true>(art::ArtMethod*, art::Thread*, art::ShadowFrame&, art::Instruction const*, unsigned short, bool, art::JValue*)+1960) (BuildId: 02bec5940be704b863f6514fc7d81c41)
      #11 pc 00000000004985e4  /apex/com.android.art/lib64/libart.so (void art::interpreter::ExecuteSwitchImplCpp<false>(art::interpreter::SwitchImplContext*)+11116) (BuildId: 02bec5940be704b863f6514fc7d81c41)
      #12 pc 0000000000357dd8  /apex/com.android.art/lib64/libart.so (ExecuteSwitchImplAsm+8) (BuildId: 02bec5940be704b863f6514fc7d81c41)
      #13 pc 000000000027cac0  [anon:dalvik-classes19.dex extracted in memory from /data/app/~~o0DJueGdxsdbGpzbRNJrOg==/com.mobilepart.teacher-L6aGGx-wMPf4--6FR06KRA==/base.apk!classes19.dex] (com.google.android.libraries.barhopper.BarhopperV3.recognize+0)
      #14 pc 0000000000371b14  /apex/com.android.art/lib64/libart.so (art::interpreter::Execute(art::Thread*, art::CodeItemDataAccessor const&, art::ShadowFrame&, art::JValue, bool, bool) (.__uniq.112435418011751916792819755956732575238.llvm.8722505846101882172)+232) (BuildId: 02bec5940be704b863f6514fc7d81c41)
      #15 pc 00000000005136f0  /apex/com.android.art/lib64/libart.so (bool art::interpreter::DoCall<false>(art::ArtMethod*, art::Thread*, art::ShadowFrame&, art::Instruction const*, unsigned short, bool, art::JValue*)+5252) (BuildId: 02bec5940be704b863f6514fc7d81c41)
      #16 pc 0000000000495e6c  /apex/com.android.art/lib64/libart.so (void art::interpreter::ExecuteSwitchImplCpp<false>(art::interpreter::SwitchImplContext*)+1012) (BuildId: 02bec5940be704b863f6514fc7d81c41)
      #17 pc 0000000000357dd8  /apex/com.android.art/lib64/libart.so (ExecuteSwitchImplAsm+8) (BuildId: 02bec5940be704b863f6514fc7d81c41)
      #18 pc 000000000034b900  [anon:dalvik-classes19.dex extracted in memory from /data/app/~~o0DJueGdxsdbGpzbRNJrOg==/com.mobilepart.teacher-L6aGGx-wMPf4--6FR06KRA==/base.apk!classes19.dex] (com.google.mlkit.vision.barcode.bundled.internal.zzb.zzf+0)
[... cut ...]
      #133 pc 000000000023a9ac  /apex/com.android.art/lib64/libart.so (art::ArtMethod::Invoke(art::Thread*, unsigned int*, unsigned int, art::JValue*, char const*)+144) (BuildId: 02bec5940be704b863f6514fc7d81c41)
      #134 pc 000000000053b96c  /apex/com.android.art/lib64/libart.so (art::Thread::CreateCallback(void*)+1600) (BuildId: 02bec5940be704b863f6514fc7d81c41)
      #135 pc 00000000000ba598  /apex/com.android.runtime/lib64/bionic/libc.so (__pthread_start(void*)+208) (BuildId: 2938f6235116cbc48464ee0f7622625e)
      #136 pc 0000000000053f3c  /apex/com.android.runtime/lib64/bionic/libc.so (__start_thread+68) (BuildId: 2938f6235116cbc48464ee0f7622625e)
```

mrousavy, 2024-03-25:
I just released a V4 beta, can you try if that fixes the bugs?

nassimmiled, 2024-04-19:
This bug was fixed in the new version 3.9.2 or not yet?

pke, 2024-04-26:
not fixed in any 3.x version @nassimmiled