margelo/react-native-vision-camera #2343: 🐛

juliacremus, 2024-01-03:
### What's happening?

even though I set 

`photo = true`, 

 [capture/photo-not-enabled: [capture/photo-not-enabled] Photo capture is disabled! Pass `photo={true}` to enable photo capture.]

I got this error, and as others issues, my camera gets black and doesn't return to normal state.

### Reproduceable Code

```tsx
<Camera
                        ref={camera}
                        style={StyleSheet.absoluteFill}
                        device={device}
                        isActive={showCamera}
                        photo={true}
                        onError={(error) => {cameraOnError}}
                        onInitialized={() => {setShowCamera(true)}}

                    />
```

### Relevant log output

```shell
[capture/photo-not-enabled: [capture/photo-not-enabled] Photo capture is disabled! Pass `photo={true}` to enable photo capture.]
```

### Camera Device

```json
{
  "sensorOrientation": "landscape-right",
  "hardwareLevel": "limited",
  "maxZoom": 4,
  "minZoom": 1,
  "maxExposure": 200,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": false,
  "isMultiCam": false,
  "minExposure": -200,
  "name": "BACK (0)",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

samsung galaxy 

### VisionCamera Version

3.6.17

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

juliacremus, 2024-01-03:
the error seems to be on the device definition, since I changed back to front and it opens the camera, but when I take the photo it gets black again. When I change again to back device it opens just for one photo and then blacks out.

LinaEfer, 2024-01-05:
I have the same issue here on the OnePlus.

LIbrary version `3.6.17 `
Android version `13`

```javascript
 <PinchGestureHandler enabled={isActive} onGestureEvent={onPinchGesture}>
      <Reanimated.View style={theme.styles.flex1}>
        <TapGestureHandler numberOfTaps={2}>
          <ReanimatedCamera
            animatedProps={cameraAnimatedProps}
            device={device!}
            isActive={true}
            onError={(error) => logger.error({ tag: TAG, error, message: error?.message })}
            onInitialized={() => setIsCameraReady(true)}
            orientation="portrait"
            ref={ref}
            video={false}
            enableHighQualityPhotos
            photo={true}
          />
        </TapGestureHandler>
      </Reanimated.View>
    </PinchGestureHandler>
```

leppid, 2024-01-05:
Same thing!

mrousavy, 2024-01-05:
Hey - weird issue!

Can you try if my PR fixes this issue? https://github.com/mrousavy/react-native-vision-camera/pull/2339

juliacremus, 2024-01-05:
Hey Marc, this problem is solved, but the blackout in the camera, for me, didn't change...... still has this strange behavior working only once when I change the device to back to front......
I downloaded the commited code inside my project, but still returns this error.

mrousavy, 2024-01-05:
Someone else said that the blackscreen issue is fixed: https://github.com/mrousavy/react-native-vision-camera/issues/2340#issuecomment-1878931287

Can you share a video and logs of the issue? @juliacremus

NoatToan, 2024-01-07:
[image]
"react-native-vision-camera": "^3.6.16",
"react-native": "0.72.4",
```
  const renderCamera = useCallback(() => {
    return (
      <Camera
        onInitialized={onInitialized}
        onError={_ => {
          console.log('Camera', _);
        }}
        ref={camera}
        enableHighQualityPhotos={true}
        style={[
          {
            flex: 1,

            borderWidth: 0,
            zIndex: 0,
            width: width,
            height: height,
          },
        ]}
        photo={true}
        device={device}
        isActive={isActiveCamera && isCameraInitialized && isFocused}
      />
    );
  }, [isActiveCamera, isCameraInitialized, isFocused, device]);
```

> Someone else said that the blackscreen issue is fixed: [#2340 (comment)](https://github.com/mrousavy/react-native-vision-camera/issues/2340#issuecomment-1878931287)
> 
> Can you share a video and logs of the issue? @juliacremus

mrousavy, 2024-01-07:
@NoatToan please share the full logs. Not a screenshot showing three lines of logs.

NoatToan, 2024-01-07:
@mrousavy Sorry my mistake.

Here is for you. Hope that you will know and help me fix the issue.

[photo_true_issue.md](https://github.com/mrousavy/react-native-vision-camera/files/13853560/photo_true_issue.md)

closed 2024-01-08 by mrousavy

aglmnzr16, 2024-05-15:
it still happened in version 4.0.3

> @NoatToan please share the full logs. Not a screenshot showing three lines of logs.