margelo/react-native-vision-camera #2344: 🐛 [capture/unknown: [capture/unknown] An unknown error occurred while trying to capture an Image! Was Image captured: false]

gosbhs12, 2024-01-04:
### What's happening?

I want to use a feature of takePhoto in React Native. It worked in iOS. Also when I take a video it worked well, but not takePhoto feature. The error shows one of the below errors. It sometimes shows the first or second one. 
Error - 
[capture/unknown: [capture/unknown] An unknown error occurred while trying to capture an Image! Was Image captured: false]
[capture/aborted: [capture/aborted] The image capture was aborted! Was Image captured: false]

I have set the version below.
ext {
        buildToolsVersion = "33.0.0"
        minSdkVersion = 26
        compileSdkVersion = 33
        targetSdkVersion = 33
        kotlinVersion = "1.8.10"

        ndkVersion = "23.1.7779620"
    }
I have tested it in Pixel 5 emulator, and never tried with different actual device. Is it possible to happen to only emulator? 
(will get actual android device soon)

### Reproduceable Code

```tsx
const [cameraPosition, setCameraPosition] = useState<'front' | 'back'>(
    'back',
  );
const device = useCameraDevice(cameraPosition);
<Camera
                  ref={currCamera}
                  torch={flash === 'on' ? 'on' : 'off'}
                  photo={true}
                  style={flexOne}
                  onInitialized={initialize}
                  device={device}
                  format={format}
                  videoHdr={enableHdr}
                  video={true}
                  fps={fps}
                  isActive={isCamActive}
                />
```

### Relevant log output

```shell
2024-01-04 01:43:58.903  7417-7608  BufferQueueProducer     com.realplayground                   E  [ImageReader-640x480f21m3-7417-1](id:1cf900000005,api:4,p:476,c:7417) dequeueBuffer: BufferQueue has been abandoned
2024-01-04 01:43:58.903   476-7593  Camera3-OutputStream    cameraserver                         E  getBufferLockedCommon: Stream 2: Can't dequeue next output buffer: No such device (-19)
2024-01-04 01:43:58.905  7417-7612  System.err              com.realplayground                   W  com.mrousavy.camera.core.UnknownCaptureError: [capture/unknown] An unknown error occurred while trying to capture an Image! Was Image captured: false
2024-01-04 01:43:58.906  7417-7417  libEGL                  com.realplayground                   E  call to OpenGL ES API with no current context (logged once per thread)
2024-01-04 01:43:58.906  7417-7417  EGL_emulation           com.realplayground                   E  eglMakeCurrent: error: EGL_BAD_ACCESS: context 0xb400006f9b4ce010 current to another thread!
2024-01-04 01:43:58.906  7417-7417  EGL_emulation           com.realplayground                   E  tid 7417: eglMakeCurrent(1933): error 0x3002 (EGL_BAD_ACCESS)
2024-01-04 01:43:58.906  7417-7417  libEGL                  com.realplayground                   E  eglMakeCurrentImpl:1043 error 3002 (EGL_BAD_ACCESS)
2024-01-04 01:43:58.906  7417-7417  OpenGLContext           com.realplayground                   I  Destroying OpenGL Surface...
2024-01-04 01:43:58.906  7417-7417  OpenGLContext           com.realplayground                   I  Destroying OpenGL Context...
2024-01-04 01:43:58.906  7417-7417  OpenGLContext           com.realplayground                   I  Destroying OpenGL Display...
2024-01-04 01:43:58.907  7417-7603  BufferQueueProducer     com.realplayground                   E  [SurfaceTexture-0-7417-1](id:1cf900000006,api:4,p:476,c:7417) queueBuffer: BufferQueue has been abandoned
2024-01-04 01:43:58.907   476-7598  Surface                 cameraserver                         E  queueBuffer: error queuing buffer, -19
2024-01-04 01:43:58.907   476-7598  Camera3-Pr...rameSpacer cameraserver                         E  queueBufferToClientLocked: Failed to queue buffer to client: No such device(-19)
2024-01-04 01:43:58.907  7417-7612  System.err              com.realplayground                   W  	at com.mrousavy.camera.extensions.CameraCaptureSession_captureKt$capture$2$1.onCaptureFailed(CameraCaptureSession+capture.kt:39)
2024-01-04 01:43:58.908  7417-7612  System.err              com.realplayground                   W  	at android.hardware.camera2.impl.CameraCaptureSessionImpl$1.lambda$onCaptureFailed$5$android-hardware-camera2-impl-CameraCaptureSessionImpl$1(CameraCaptureSessionImpl.java:722)
2024-01-04 01:43:58.908  7417-7612  System.err              com.realplayground                   W  	at android.hardware.camera2.impl.CameraCaptureSessionImpl$1$$ExternalSyntheticLambda7.run(Unknown Source:8)
2024-01-04 01:43:58.908  7417-7612  System.err              com.realplayground                   W  	at android.os.Handler.handleCallback(Handler.java:942)
2024-01-04 01:43:58.908  7417-7612  System.err              com.realplayground                   W  	at android.os.Handler.dispatchMessage(Handler.java:99)
2024-01-04 01:43:58.908  7417-7612  System.err              com.realplayground                   W  	at android.os.Looper.loopOnce(Looper.java:201)
2024-01-04 01:43:58.908  7417-7612  System.err              com.realplayground                   W  	at android.os.Looper.loop(Looper.java:288)
2024-01-04 01:43:58.908  7417-7612  System.err              com.realplayground                   W  	at android.os.HandlerThread.run(HandlerThread.java:67)
2024-01-04 01:43:58.908  7417-7417  CameraSession           com.realplayground                   I  Adding 640 x 480 Photo Output in Format #256...
2024-01-04 01:43:58.911  7417-7417  CameraSession           com.realplayground                   I  Adding 640 x 480 Video Output in Format #34...
2024-01-04 01:43:58.911  7417-7417  VideoPipeline           com.realplayground                   I  Initializing 640 x 480 Video Pipeline (format: NATIVE)
2024-01-04 01:43:58.912  7417-7417  CreateCaptureSession    com.realplayground                   I  Camera 0: Creating Capture Session #1002... Hardware Level: 0} | Outputs: [android.hardware.camera2.params.OutputConfiguration@c4ef4e8, android.hardware.camera2.params.OutputConfiguration@4c44c9b4]
2024-01-04 01:43:58.912  7417-7417  CreateCaptureSession    com.realplayground                   I  Using new API (>=28)
2024-01-04 01:43:58.916  7417-7476  ReactNativeJS           com.realplayground                   E  'Failed to take photo!', { [capture/unknown: [capture/unknown] An unknown error occurred while trying to capture an Image! Was Image captured: false]
                                                                                                      name: 'capture/unknown',
                                                                                                      _code: 'capture/unknown',
                                                                                                      _message: '[capture/unknown] An unknown error occurred while trying to capture an Image! Was Image captured: false',
                                                                                                      _cause: undefined }
2024-01-04 01:43:58.939  7417-7603  BufferQueueProducer     com.realplayground                   E  [SurfaceTexture-0-7417-1](id:1cf900000006,api:4,p:476,c:7417) queueBuffer: BufferQueue has been abandoned
2024-01-04 01:43:58.939   476-7598  Surface                 cameraserver                         E  queueBuffer: error queuing buffer, -19
2024-01-04 01:43:58.939   476-7598  Camera3-Pr...rameSpacer cameraserver                         E  queueBufferToClientLocked: Failed to queue buffer to client: No such device(-19)
2024-01-04 01:43:58.963   476-476   CameraLatencyHistogram  cameraserver
```

### Camera Device

```json
{"hardwareLevel": "limited", "hasFlash": false, "hasTorch": false, "id": "0", "isMultiCam": false, "maxExposure": 27, "maxZoom": 10, "minExposure": -27, "minZoom": 1, "name": "BACK (0)", "neutralZoom": 1, "physicalDevices": ["telephoto-camera"], "position": "back", "sensorOrientation": "landscape-right", "supportsFocus": true, "supportsLowLightBoost": false, "supportsRawCapture": false}
```

### Device

Pixel 5 API 33

### VisionCamera Version

3.6.16

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-01-05:
```
2024-01-04 01:53:46.838  7417-7612  CameraSession           com.realplayground                   I  Photo capture 0/3 - preparing capture request (640x480)...
2024-01-04 01:53:46.855  7417-7612  CameraSession           com.realplayground                   I  Photo capture 1/3 - starting capture...
2024-01-04 01:53:46.861   386-435   goldfish-address-space  and...raphics.allocator@3.0-service  D  claimShared: Ask to claim region [0x1f3520000 0x1f3594000]
2024-01-04 01:53:46.889   404-547   AudioFlinger            audioserver                          D  mixer(0xb4000079a909a9a0) throttle end: throttle time(44)
2024-01-04 01:53:46.891  7417-7417  CameraSession           com.realplayground                   I  PreviewView Surface destroyed! Surface(name=null)/@0x53529b5
2024-01-04 01:53:46.891  7417-7417  CameraSession           com.realplayground                   I  Destroying Preview Output...
```

those logs showcase that the PreviewView gets destroyed as soon as you capture an image. Is it possible that you navigate away from the screen once you capture?

[image]

Can you upload a screen recording of that happening? It doesn't happen for me.

gosbhs12, 2024-01-06:
I've found a solution! The issue was related to how the camera button's events were handled. I had assigned multiple events to the button, which caused the events to trigger multiple times on Android, leading to errors.

closed 2024-01-06 by gosbhs12