margelo/react-native-vision-camera #2345: capture/no-recording-in-progress

SUBTAIN-ALI, 2024-01-04:
### What's happening?

When I start video recording in some android devices it give me this error [capture/no-recording-in-progress] There was no active video recording in progress! Did you call stopRecording() twice?

### Reproduceable Code

```tsx
const [selectedCamera, setSelectedCamera] = useState('front');
  const device = useCameraDevice(selectedCamera, {
    physicalDevices: [
      'ultra-wide-angle-camera',
      'wide-angle-camera',
      'telephoto-camera',
    ],
  });
  const camera = useRef();
  const isFocused = useIsFocused();
  const [isRecording, setIsRecording] = useState(false);

{device && (
        <Camera
          ref={camera}
          style={[StyleSheet.absoluteFill, {borderRadius: 25}]}
          device={device}
          isActive={isFocused}
          video={true}
          audio={true}
          photo={false}
          format={format}
        />
      )}
```

### Relevant log output

```shell
[capture/no-recording-in-progress] There was no active video recording in progress! Did you call stopRecording() twice?
```

### Camera Device

```json
hardwareLevel: "full"
hasFlash: false
hasTorch: false
id: "1"
isMultiCam: false
maxExposure: 400
maxZoom: 10
minExposure: -400
minZoom: 1
name: "FRONT (1)"
neutralZoom: 1
physicalDevices: ['wide-angle-camera']
position: "front"
sensorOrientation: 
"landscape-left"
supportsFocus: true
supportsLowLightBoost: false
supportsRawCapture: true
```

### Device

Samsung SM-N981N,SM-G990E, Realme  SM-N981N (TP1A.220624.014), M2003J15SC

### VisionCamera Version

3.6.4

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-01-29:
Hey - sorry I cannot help you without any logs (you only posted the error, not the logs).

Also, that error is only thrown in 3 locations: `stopRecording`, `pauseRecording` and `resumeRecording`. I'm guessing you just called one of those methods, then the error is intentional.

closed 2024-01-29 by mrousavy

aymenboudissa, 2025-01-14:
I got the same error could you please provide me the solution