margelo/react-native-vision-camera #2346: Barcode scaning from captured images❓

786Muhammadshams, 2024-01-04:
### Question

I am interested in using React Native Vision Camera for a specific use case. My goal is to first capture a photo of a barcode and then scan the barcode from this captured image. I understand that React Native Vision Camera is primarily designed for real-time scanning from the camera's live feed. However, I would like to know if it also supports scanning from a photo that has already been taken. Is this functionality possible with the current version of React Native Vision Camera?

### VisionCamera Version

latest

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

aguinaldotupy, 2024-01-04:
I have the same use case and I'm not sure how to implement this flow

mrousavy, 2024-01-05:
Hey - no, this is not supported. VisionCamera is a Camera, not a post-processing toolkit.

closed 2024-01-05 by mrousavy