margelo/react-native-vision-camera #2351: 🐛  The given value for videoStabilizationMode could not be parsed! (Received: auto)

BLOCKMATERIAL, 2024-01-05:
### What's happening?

Crash on Android Phone 

InvalidTypeScriptUnionError: [parameter/invalid-parameter] The given value for videoStabilizationMode could not be parsed! (Received: auto)
[image]

### Reproduceable Code

```tsx
<Camera
        ref={cameraRef}
        isActive={isActive}
        photo={true}
        video={true}
        audio={true}
        style={styles.preview}
        device={device}
        videoStabilizationMode="auto"
        onInitialized={onInitialized}
        exposure={0}
        orientation="portrait"
        focusable
        fps={fps}
        format={format}
        enableZoomGesture
        onError={onError}
      />
```

### Relevant log output

```shell
InvalidTypeScriptUnionError: [parameter/invalid-parameter] The given value for videoStabilizationMode could not be parsed! (Received: auto)
```

### Camera Device

```json
{
  "sensorOrientation": "landscape-right",
  "hardwareLevel": "full",
  "maxZoom": 7,
  "minZoom": 0.6704425811767578,
  "maxExposure": 144,
  "supportsLowLightBoost": true,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera",
    "ultra-wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": true,
  "isMultiCam": true,
  "minExposure": -144,
  "name": "BACK (0)",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Xiaomi Redmi 2303ERA42L

### VisionCamera Version

3.6.16

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

vijaykhandal1290, 2024-01-05:
Please give me sample code for video record in frame and check face detect in frame or not... my question is irt is possible with react native vision camera.

mrousavy, 2024-01-05:
That's because `auto` video stabilization is not supported on that phone. You can't just pass it, you need to check if that is supported first by doing
```ts
const videoStabilizationMode = format.videoStabilizationModes.includes("auto") ? "auto" : undefined
```

Either way, I'll add a patch to disable video stabilization on Android so this runs as I agree it is a bit confusing. In future versions, `auto` will resolve to `off` on Android.

closed 2024-01-05 by mrousavy