margelo/react-native-vision-camera #2357: ❓ Does video quality affects photo preview?

rodrigobertolottizinsp, 2024-01-08:
### Question

Hi, I have an app where users can take pictures and videos with different resolutions. Im selecting a format using both picture and video resolutions values set by the users, they are the default ones 2048, 1024, 720 for pictures and 720p, 480p and 288p for videos. 
When I set a lower video resolution parameter for selecting a format, for example  the pictures preview gets really blurry, this is the intended behavior? Can I decouple preview from video resolution in any way? 
Thanks!

### What I tried

Example of video resolution parameter :
{"videoResolution": {"height": 480, "width": 640}

format got: 

{"autoFocusSystem": "phase-detection", "fieldOfView": 101.02159118652344, "maxExposure": 8, "maxFps": 60, "maxISO": 5280, "maxZoom": 94.5, "minExposure": -8, "minFps": 60, "minISO": 55, "photoHeight": 1512, "photoWidth": 2016, "pixelFormats": ["yuv", "yuv", "rgb", "yuv", "unknown", "yuv", "unknown", "rgb", "unknown"], "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 480, "videoStabilizationModes": ["auto", "off"], "videoWidth": 640}

Device: iPhone 15 Pro with iOS 17.3

### VisionCamera Version

3.6.8

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2024-01-09:
Yes, as a technical explanation:

- Photo: A _single_ capture request that can be of high quality. The output is a special format, will be encoded to JPEG. It's quality can be high even if video quality is low.
- Video: A _repeating_ capture request that can also be of high quality. The output is a streaming format such as YUV or PRIVATE and will be used for multiple outputs:
  - Video Recordings (pretty obvious)
  - Frame Processing (raw frame data access for ML stuff)
  - Code Scanning (the same as Frame Processing just wrapped nicely)
  - Preview (lower resolution to match the screen size)

mrousavy, 2024-01-09:
So in your case, you need to select a video resolution that is at least screen size to make the preview look a bit better, and then downsize the video later maybe. But preview and video are tied together. I don't see any reason why modern apps should choose a video resolution of less than 720p.

closed 2024-01-09 by mrousavy

rodrigobertolottizinsp, 2024-04-16:
The issue is that I need to switch between different photo resolutions, but i need the preview to always look fine. So I dont have any format with high video resolution to make the screen look good, and smaller image size that allows me to take lighter pictures.

mrousavy, 2024-04-17:
yea still, just pass videoResolution of at least screen size. Even if you only use photo