margelo/react-native-vision-camera #2358: 🐛 [Android] Camera is not switching when isActive is turned to false

GolliAch, 2024-01-08:
### What's happening?

Camera is not switching when isActive is turned to false with V3.7.0
Same code was working with V2 and V3  < 3.7.0

In logcat the we see that  isActive is set to false, but the camera still does not pause. 

### Reproduceable Code

```tsx
<Camera
                style={[StyleSheet.absoluteFill, styles.cameraSytle]}
                isActive={isCameraActive && isFocused}
                photo={true}
                ref={camera}
                device={device}
            />
```

### Relevant log output

```shell
2024-01-08 22:15:48.484 19951-20281 CameraSession           anonymous                       I  Camera Session android.hardware.camera2.impl.CameraCaptureSessionImpl@3e5f163 has been closed.
2024-01-08 22:15:48.484 19951-20281 CameraSession           anonymous                       I  Successfully configured Session with 2 outputs for Camera #0!
2024-01-08 22:15:48.498 19951-20281 CameraView              anonymous                       I  invokeOnStarted()
2024-01-08 22:15:48.499 19951-19984 CameraManagerGlobal     anonymous                       I  Camera 0 facing CAMERA_FACING_BACK state now CAMERA_STATE_ACTIVE for client anonymous API Level 2User Id 0
2024-01-08 22:15:48.501 19951-20281 CameraSession           anonymous                       I  Successfully updated CameraSession Configuration! isActive: true
2024-01-08 22:15:57.214 19951-19951 CameraView              anonymous                       D  Finding view 95...
2024-01-08 22:15:57.214 19951-19951 CameraView              anonymous                       D  Found view 95!
2024-01-08 22:15:57.216 19951-20109 CameraView.takePhoto    anonymous                       I  Taking photo... Options: {}
2024-01-08 22:15:57.216 19951-20109 CameraSession           anonymous                       I  Photo capture 0/3 - preparing capture request (4624x3468)...
2024-01-08 22:15:57.225 19951-20109 CameraSession           anonymous                       I  Photo capture 1/3 - starting capture...
2024-01-08 22:15:57.467 19951-20281 SoundPool               anonymous                       E  error loading /product/media/audio/ui/camera_click.ogg
2024-01-08 22:15:57.469 19951-20281 SoundPool               anonymous                       E  error loading /product/media/audio/ui/camera_click.ogg
2024-01-08 22:15:57.637 19951-20109 CameraSession           anonymous                       I  Photo capture 2/3 complete - received metadata with timestamp 4678547033806619
2024-01-08 22:15:57.640 19951-20281 CameraSession           anonymous                       I  Photo Captured!
2024-01-08 22:15:57.642 19951-20281 CameraSession           anonymous                       I  Photo captured! 4624 x 3468
2024-01-08 22:15:57.644 19951-20109 CameraSession           anonymous                       I  Photo capture 3/3 complete - received 4624 x 3468 image.
2024-01-08 22:15:57.645 19951-20109 CameraView.takePhoto    anonymous                       I  Successfully captured 4624 x 3468 photo!
2024-01-08 22:15:57.667 19951-20109 CameraView.takePhoto    anonymous                       I  Successfully saved photo to file! /data/user/0/anonymous/cache/mrousavy4337852227698016070.jpg
2024-01-08 22:15:57.695 19951-21039 ReactNativeJS           anonymous                       I  {
                                                                                                      "sensorOrientation": "landscape-right",
                                                                                                      "hardwareLevel": "full",
                                                                                                      "maxZoom": 8,
                                                                                                      "minZoom": 1,
                                                                                                      "maxExposure": 20,
                                                                                                      "supportsLowLightBoost": false,
                                                                                                      "neutralZoom": 1,
                                                                                                      "physicalDevices": [
                                                                                                        "wide-angle-camera"
                                                                                                      ],
                                                                                                      "supportsFocus": true,
                                                                                                      "supportsRawCapture": true,
                                                                                                      "isMultiCam": false,
                                                                                                      "minExposure": -20,
                                                                                                      "name": "BACK (0)",
                                                                                                      "hasFlash": true,
                                                                                                      "hasTorch": true,
                                                                                                      "position": "back",
                                                                                                      "id": "0"
                                                                                                    }
2024-01-08 22:15:57.716 19951-21039 ReactNativeJS           anonymous                       I  {
                                                                                                      "sensorOrientation": "landscape-right",
                                                                                                      "hardwareLevel": "full",
                                                                                                      "maxZoom": 8,
                                                                                                      "minZoom": 1,
                                                                                                      "maxExposure": 20,
                                                                                                      "supportsLowLightBoost": false,
                                                                                                      "neutralZoom": 1,
                                                                                                      "physicalDevices": [
[... 62 lines cut ...]
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Android 

### VisionCamera Version

3.7.0

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

tytom2003, 2024-01-09:
const [isActive, setActive] = useState(false);

  useEffect(() => {
    if (hasPermission) {
      setTimeout(() => setActive(true), 500); // I'm not sure "setTimeout" is needed
    }
  }, [hasPermission]);

And

<Camera
        style={StyleSheet.absoluteFill}
        ref={camera}
        device={device}
        isActive={isActive}
      />

tytom2003, 2024-01-09:
tried the method to solve black screen issue.
[image]

tytom2003, 2024-01-09:
reference link. i tried react native vision lifecycle still can't solve black screen problem, but setTimeout can solve the black screen problem.

https://discord.com/channels/1147256486762913884/1185605026907967568/1186034102248419559

hasaneks, 2024-01-09:
Hello,

I'm having the same problem too. When I want to turn off the camera after scanning the QR code, I encounter a black screen error.Hello,

mrousavy, 2024-01-09:
oh I think I know why this is

R4tch3t, 2024-01-10:
> Done! Fix: #2361

works

GolliAch, 2024-01-10:
@mrousavy This fixed it!

closed 2024-01-10 by GolliAch

mrousavy, 2024-01-10:
Awesome!