margelo/react-native-vision-camera #2360: ERROR  Camera.onError(unknown/unknown): [unknown/unknown] Only the original thread that created a view hierarchy can touch its views. [unknown/unknown: [unknown/unknown] Only the original thread that created a view hierarchy can touch its views.]🐛

favazpa, 2024-01-09:
### What's happening?

In some of the devices like samsung note 5 , realme 11 pro + 5g etc.. iam getting blanck screen instead of camera view in the screen iam getting blacnk screen sometimes green screen. In version 2 vision camera this works fine but in vision camera 3 camera is not working. 

### Reproduceable Code

```tsx
const camera = useRef(null);
const [cameraPermission, setCameraPermission] = useState();
useEffect(() => {
    requestCameraPermissions().then(hasPermission => {
      if (hasPermission) {
        setCameraPermission(true);
      } else {
        console.error('Permissions not granted');
        // Handle lack of permissions appropriately
      }
    });
  }, []);
async function requestCameraPermissions() {
    if (Platform.OS === 'android') {
      const cameraPermission = await PermissionsAndroid.request(
        PermissionsAndroid.PERMISSIONS.CAMERA,
      );
      const microphonePermission = await PermissionsAndroid.request(
        PermissionsAndroid.PERMISSIONS.RECORD_AUDIO,
      );
      return (
        cameraPermission === PermissionsAndroid.RESULTS.GRANTED &&
        microphonePermission === PermissionsAndroid.RESULTS.GRANTED
      );
    } else {
      // For iOS, use react-native-vision-camera's requestCameraPermission method
      const cameraPermissionStatus = await Camera.requestCameraPermission();
      const microphonePermissionStatus =
        await Camera.requestMicrophonePermission();
      return (
        cameraPermissionStatus === 'authorized' &&
        microphonePermissionStatus === 'authorized'
      );
    }
  }
 useEffect(() => {
    async function setupSing() {
      let isSetup = await setupPlayer();
      setIsPlayerReady(isSetup);
    }

    if (cameraPermission) {
      setupSing();
    }
  }, [cameraPermission]);
 const cameraDevice = useCameraDevice('front');
 const renderRecordingVideo = () => {
    return (
      <View>
        {
          // videoPath ? (
          //   <Video source={{uri: videoPath}} style={styles.photoAndVideoCamera} />
          // ) : (
          cameraDevice && (
            <Camera
              ref={camera}
              style={[styles.camera, styles.photoAndVideoCamera]}
              device={cameraDevice}
              isActive
              video
              audio
            />
          )
[... 208 lines cut ...]
  "hasTorch": false,
  "position": "front",
  "id": "1"
}
```

### Device

Samsung galaxy note 5 , realme 11 pro + 5g

### VisionCamera Version

3.6.10

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

francesco-clementi-92, 2024-01-09:
Latest version should fix your error

mrousavy, 2024-01-11:
Hey! this is fixed in the last version :)

closed 2024-01-11 by mrousavy

favazpa, 2024-01-12:
> Hey! this is fixed in the last version :)

i tried and working for back cam but when i use front cam I am getting a green colour screen nothing else. also when i log i got like below.

{
  "sensorOrientation": "landscape-left",
  "hardwareLevel": "limited",
  "maxZoom": 4,
  "minZoom": 1,
  "maxExposure": 4,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": false,
  "isMultiCam": false,
  "minExposure": -4,
  "name": "FRONT (1)",
  "hasFlash": false,
  "hasTorch": false,
  "position": "front",
  "id": "1"
}

favazpa, 2024-01-12:
> Latest version should fix your error

not working for front cam iam getting a green screen while using front camera. but if i just change the position from back to front while app is running front camera will turn on. but if we directly use front camera its green color screen

favazpa, 2024-01-16:
@mrousavy  could you please help me with this?

mrousavy, 2024-01-16:
I'm sorry I don't have the free time to look into this.

Also I cannot reproduce it.

francesco-clementi-92, 2024-01-16:
@favazpa I had the same problem on my Redmi note, tested now on 3.8.0 and it's working. Try to update again to the latest version and the troubleshoot guide https://react-native-vision-camera.com/docs/guides/troubleshooting