margelo/react-native-vision-camera #2362: 🐛Camera showing black screen on first render

anuragkaushik10, 2024-01-09:
### What's happening?

When i am trying to open camera for the first time its, showing black screen.

### Reproduceable Code

```tsx
<Camera
        style={stylesWithTheme.camera}
        device={device}
        isActive={isFocused}
        audio={false}
        codeScanner={codeScanner}
      />
```

### Relevant log output

```shell
2024-01-09T08:51:50.679Z
419
Camera3-Device
Set real time priority for request queue thread (tid 11343)
2024-01-09T08:51:50.681Z
11072
CreateCaptureSession
Camera 10: Capture Session #1005 closed!
2024-01-09T08:51:50.681Z
11072
CreateCaptureSession
Camera 10: Capture Session #1006 configured!
2024-01-09T08:51:50.682Z
11072
CameraSession
Successfully configured Session with 1 outputs for Camera #10!
2024-01-09T08:51:50.682Z
11072
CameraSession
Preview Output is null, aborting...
2024-01-09T08:51:50.682Z
11072
CameraSession
Successfully updated CameraSession Configuration! isActive: true
2024-01-09T08:51:50.682Z
11072
CameraView
invokeOnInitialized()
2024-01-09T08:51:50.684Z
11072
CameraView
invokeOnStarted()
2024-01-09T08:51:50.690Z
11072
CameraSession
Configuring Session for Camera #10...
2024-01-09T08:51:50.690Z
11072
SurfaceOutput
Closing BarcodeScanner..
2024-01-09T08:51:50.692Z
11072
CameraSession
Adding 1280 x 960 Preview Output...
2024-01-09T08:51:50.692Z
11072
PreviewView
Resizing PreviewView to 1280 x 960...
2024-01-09T08:51:50.693Z
11072
CameraSession
[... 1831 lines cut ...]
  "hasTorch": false,
  "position": "back",
  "id": "10"
}
```

### Device

Pixel_5_API34

### VisionCamera Version

^3.6.4

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

maksimLobachevskiy, 2024-01-09:
According to the last release notes Black screen should be already resolved in v 3.7.0.

amiransari8921, 2024-01-09:
getting the same black screen not rendering the camera in first render !
on version react-native-vision-camera 
3.6.17

GolliAch, 2024-01-09:
@maksimLobachevskiy  v 3.7.0. Didn't solve it for us

anuragkaushik10, 2024-01-10:
> According to the last release notes Black screen should be already resolved in v 3.7.0.

Working fine for Android, but not fixed for iOS.

alexandre-ducarne, 2024-01-11:
For me it's the opposite

harshbytes9026, 2024-01-11:
> For me it's the opposite

+1

GolliAch, 2024-01-11:
@alexandre-ducarne  opposite as working on iOS not Android?

mrousavy, 2024-01-11:
Sorry there's so much conflicting information here. My thoughts:

1. iOS never had those issues, and currently still doesn't. @anuragkaushik10 you're saying you have a blackscreen on iOS - which device did you test on? Please test the example app here.
2. @alexandre-ducarne @harshbytes9026 you're saying you are still experiencing blackscreens on Android even after upgrading to 3.7.0. - can you reproduce those in the example app? Please create a reproduction and share your `adb logcat` logs with me.
3. @maksimLobachevskiy you (+2 upvoters) are saying the issue is fixed in 3.7.0 - that's great to hear. I also found the issue to be fixed on my end.
4. @anuragkaushik10 the "Relevant Logs" are wrong. These are not adb logcat logs. Please share the full ADB logcat logs (on a gist), I can't do anything with the logs you gave me.

Amberevol, 2024-01-12:
[image]
I'm getting this black screen on few devices android using v3.7.0

alexandre-ducarne, 2024-01-12:
@mrousavy Actually I got both working, the black screen was caused because the Android Camera permission wasn't requested beforehand. Before 3.7.0 I had issues on IOS but since that version everything is fine on both Android & IOS.

mrousavy, 2024-01-30:
> I'm getting this black screen on few devices android using v3.7.0

What? I can literally see the Camera

mrousavy, 2024-01-30:
> @mrousavy Actually I got both working, the black screen was caused because the Android Camera permission wasn't requested beforehand. Before 3.7.0 I had issues on IOS but since that version everything is fine on both Android & IOS.

Great to hear! Let's close this issue then.

closed 2024-01-30 by mrousavy

Amberevol, 2024-01-31:
@mrousavy I can also see the camera but as you can see it is very dark and i can confirm by default camera app of my phone that it's clear